openwrt/openwrt #16341: Rock64 EMMC not working

imtoanle, 2024-09-06:
### Describe the bug

I have flash latest snapshot to my Rock64 SBC.
SD: working as expected
EMMC: not working

I attached log below:
```
U-Boot 2024.07-OpenWrt-r27300-e7ea93e1e3 (Sep 05 2024 - 08:16:19 +0000)

Model: Pine64 Rock64
DRAM:  4 GiB
PMIC:  RK8050 (on=0x40, off=0x00)
Core:  239 devices, 29 uclasses, devicetree: separate
MMC:   mmc@ff500000: 1, mmc@ff520000: 0
Loading Environment from MMC... Reading from MMC(0)... *** Warning - bad CRC, using default environment

In:    serial@ff130000
Out:   serial@ff130000
Err:   serial@ff130000
Model: Pine64 Rock64
Net:   eth0: ethernet@ff540000
Hit any key to stop autoboot:  0
Card did not respond to voltage select! : -110
No EFI system partition
No EFI system partition
Failed to persist EFI variables
No EFI system partition
Failed to persist EFI variables
No EFI system partition
Failed to persist EFI variables
** Booting bootflow '<NULL>' with efi_mgr
Loading Boot0000 'mmc 0' failed
EFI boot manager: Cannot load any image
Boot failed (err=-14)
Card did not respond to voltage select! : -110
** Booting bootflow 'mmc@ff520000.bootdev.part_1' with script
4786048 bytes read in 109 ms (41.9 MiB/s)
## Loading kernel from FIT Image at 02080000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  ARM64 OpenWrt Linux-6.6.48
     Type:         Kernel Image
     Compression:  lzma compressed
     Data Start:   0x020800e8
     Data Size:    4747950 Bytes = 4.5 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x03200000
     Entry Point:  0x03200000
     Hash algo:    crc32
     Hash value:   8b3f076a
     Hash algo:    sha1
     Hash value:   315827f0e1b74f379de656a8fcb32d90f53d763f
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 02080000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'fdt-1' fdt subimage
     Description:  ARM64 OpenWrt pine64_rock64 device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x025074d8
     Data Size:    36193 Bytes = 35.3 KiB
     Architecture: AArch64
     Hash algo:    crc32
     Hash value:   425b3acc
     Hash algo:    sha1
     Hash value:   430bec77d0fbec90237c5bb2d826ef3db4cbcb20
[... 248 lines cut ...]
```

### OpenWrt version

r27300-e7ea93e1e3

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

Pine64 Rock64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

antnyfls, 2024-09-07:
@imtoanle a fix for the rock64 not booting from a eMMC was backported to kernel 6.6.49 https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.6.49&id=d80bdfaa48c1aa0c5b3aeac7beffb0d8e6c4df18 the original bug fix https://lore.kernel.org/all/7a2641f7-c928-4554-a795-e14dd4f644fa@eh5.me/
Try building the firmware or wait for a new official build , update was applied yesterday on the main branch, I can confirm that booting from the eMMC is now working.

```
root@OpenWrt:~# dmesg | grep mmc
[    0.296231] dwmmc_rockchip ff500000.mmc: IDMAC supports 32-bit address mode.
[    0.296242] dwmmc_rockchip ff520000.mmc: IDMAC supports 32-bit address mode.
[    0.296262] dwmmc_rockchip ff520000.mmc: Using internal DMA controller.
[    0.296270] dwmmc_rockchip ff520000.mmc: Version ID is 270a
[    0.296308] dwmmc_rockchip ff520000.mmc: DW MMC controller at irq 43,32 bit host data width,256 deep fifo
[    0.297195] mmc_host mmc1: card is non-removable.
[    0.297623] dwmmc_rockchip ff500000.mmc: Using internal DMA controller.
[    0.299507] dwmmc_rockchip ff500000.mmc: Version ID is 270a
[    0.299552] dwmmc_rockchip ff500000.mmc: DW MMC controller at irq 42,32 bit host data width,256 deep fifo
[    0.311281] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    0.314214] mmc_host mmc0: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    0.423024] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 52000000Hz, actual 50000000HZ div = 0)
[    0.423947] mmc_host mmc1: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[    0.437954] dwmmc_rockchip ff520000.mmc: Successfully tuned phase to 225
[    0.438659] mmc1: new HS200 MMC card at address 0001
[    0.440853] mmcblk1: mmc1:0001 SOD16G 14.6 GiB
[    0.442725]  mmcblk1: p1 p2
[    0.443672] mmcblk1boot0: mmc1:0001 SOD16G 4.00 MiB
[    0.445664] mmcblk1boot1: mmc1:0001 SOD16G 4.00 MiB
[    0.447112] mmcblk1rpmb: mmc1:0001 SOD16G 4.00 MiB, chardev (245:0)
[    8.178589] EXT4-fs (mmcblk1p1): mounted filesystem 84173db5-fa99-e35a-95c6-28613cc79ea9 r/w without journal. Quota mode: none.
[    8.194746] EXT4-fs (mmcblk1p1): unmounting filesystem 84173db5-fa99-e35a-95c6-28613cc79ea9.
```

```
root@OpenWrt:~# ubus call system board
{
        "kernel": "6.6.49",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 4",
        "model": "Pine64 Rock64",
        "board_name": "pine64,rock64",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r27312-62d3773bf1",
                "target": "rockchip/armv8",
                "description": "OpenWrt SNAPSHOT r27312-62d3773bf1"
        }
}
```

imtoanle, 2024-09-08:
@antnyfls I confirmed it's working with version "OpenWrt SNAPSHOT, r27312-62d3773bf1"
We can close this issue.
Thank you!