openwrt/openwrt #16347: Issues with Xiaomi AX3200 after Installing OpenWRT

sajadtrb, 2024-09-07:
Hello, I’m facing a few critical issues.
Device: Xiaomi AX3200

After installing OpenWRT, the device randomly stops working, and the orange light turns on. When the device is powered off and on, either manually or due to a power outage, it enters this state where the system light blinks on and off. A simple reboot seems to cause issues, and I have to reinstall OpenWRT to get it working again, which requires me to seek help from a specialist and incurs extra costs each time!Another bug occurs when I make even the smallest configuration changes via Wi-Fi in the OpenWRT settings page. After rebooting, the device again enters the same state with the orange light.This situation is very frustrating and is costing me both time and money..

danpawlik, 2024-09-07:
It is because the partition size has changed to have enough space for new kernel.
Please check https://github.com/openwrt/openwrt/pull/14770 and openwrt forum.

AuthorReflex, 2024-09-08:
What device version and revision are you using?

danpawlik, 2024-09-08:
I think you already get an answer, am I right @sajadtrb?
For others, @sajadtrb create a thread on reddit https://www.reddit.com/r/openwrt/s/A2r50Qvb5x where might be an answer how to solve that issue.

Would be nice if you share that knowledge with others. Thanks

AuthorReflex, 2024-09-08:
May be need to include this workaround to bootcount script?

maurerle, 2024-09-13:
To also include this workaround here:

> There is an issue with OpenWrt 23.05.4 and Xiaomi AX3200 that is causing a soft brick after 6 reboots on some installations.
> 
> Here is a workaround: https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/1897
> 
> The softbrick is caused by the counting try_sys flag. After it reaches 6 then the bootloader switches slots. Therefore the next boot tries from booting from the overwritten second slot. So one solution is to reset those try_sys counters to 0.
> 
> Add to /etc/rc.local
> 
> ```
> fw_setenv flag_try_sys1_failed 0
> fw_setenv flag_try_sys2_failed 0
> ```
> Thanks goes to xabolcs (https://forum.openwrt.org/u/xabolcs) for the solution
> 
> That being said, if you're not comfortable tinkering and solving issues in OpenWrt, you might want to use the stock firmware instead

We might need to set the boot success in the same way for MT7622 as there is for ramips-mt76x8:

https://github.com/openwrt/openwrt/blob/47370eb9ba2cf7021ac2bb208136c52ee85b6ba1/target/linux/ramips/mt76x8/base-files/etc/init.d/bootcount#L13