openwrt/openwrt #16358: opkg sends GET to http proxy server when https_proxy is enabled

sculpordwarf9, 2024-09-09:
### Describe the bug

When https_proxy is configured in /etc/opkg.conf, opkg sends GET to http proxy server, instead of CONNECT. As a result, TLS connection is not established by openwrt but by the proxy server. This makes download process less secure and less private because connection between openwrt and proxy server is not encrypted. Also, this behavior differs from majority of softwares like curl and package managers in other linux distro.

### OpenWrt version

23.05.4

### OpenWrt release

stable

### OpenWrt target/subtarget

ramips

### Device

ASUS RT-AC85P

### Image kind

Official downloaded image

### Steps to reproduce

1. Append `option https_proxy http://proxy:port` to /etc/opkg.conf.
2. Start tcpdump on proxy server.
3. Update package list or download a package.
4. Stop tcpdump.

### Actual behaviour

Requests like 'GET https://downloads.openwrt.org/... HTTP/1.1' is sent to proxy server. And response contains the requested content in plaintext.

### Expected behaviour

'CONNECT downloads.openwrt.org:443 HTTP/1.1' is sent to proxy server.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

sculpordwarf9, 2024-09-09:
> Invalid Target/Subtarget reported. `ramips` Is this from a supported device?

Yes, it is the supported device. See https://openwrt.org/toh/asus/rt-ac85p.

brada4, 2024-09-09:
Machinery relies on accurate copying of info from a real openwrt system. Please edit your report.
wget-ssl will work better since you have flash space