openwrt/openwrt #16361: IPQ40XX: Kernel panic - not syncing: PSGMII work is unstable

Zivertsen, 2024-09-10:
### Describe the bug

The device stops in the bootloader after running the reboot command 10-20 times. 

### OpenWrt version

'23.05.0-rc3'

### OpenWrt release

'23.05.0-rc3' 

### OpenWrt target/subtarget

ipq40xx/generic

### Device

8devices Habanero DVK

### Image kind

Self-built image

### Steps to reproduce

keep rebooting the device. 

### Actual behaviour

Stops in the bootloader 

### Expected behaviour

Do a normal reboot.

### Additional info

I have add the "broken-flash-reset;" to the flash@0 in the qcom-ipq4019-habanero-dvk DeviceTree file. I have done this to able to use the reboot command in the first place. 

**Serial port log when error happens**

```
Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.BF.3.1.1-00123
S - IMAGE_VARIANT_STRING=DAABANAZA
S - OEM_IMAGE_VERSION_STRING=CRM
S - Boot Config, 0x00000021
S - Reset status Config, 0x00000010
S - Core 0 Frequency, 0 MHz
B -       261 - PBL, Start
B -      1339 - bootable_media_detect_entry, Start
B -      1679 - bootable_media_detect_success, Start
B -      1693 - elf_loader_entry, Start
B -      5071 - auth_hash_seg_entry, Start
B -      7216 - auth_hash_seg_exit, Start
B -    577844 - elf_segs_hash_verify_entry, Start
B -    695772 - PBL, End
B -    695796 - SBL1, Start
B -    786641 - pm_device_init, Start
D -         7 - pm_device_init, Delta
B -    788105 - boot_flash_init, Start
D -     52777 - boot_flash_init, Delta
B -    845023 - boot_config_data_table_init, Start
D -      3840 - boot_config_data_table_init, Delta - (419 Bytes)
B -    852238 - clock_init, Start
D -      7556 - clock_init, Delta
B -    864270 - CDT version:2,Platform ID:9,Major ID:0,Minor ID:0,Subtype:64
B -    867798 - sbl1_ddr_set_params, Start
[... 745 lines cut ...]
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_vnstat2=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_wireless-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-basic-wolfssl=y
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_STRACE_LIBUNWIND=y
# CONFIG_STRACE_NONE is not set
CONFIG_WOLFSSL_HAS_NO_HW=y
CONFIG_WPA_WOLFSSL=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-09-10:
Are you sure your board is not defective?

I have the same board and it works just fine, also why the random 23.05-rc3?

Zivertsen, 2024-09-11:
@robimarko 
We have 2 more devices which seems to be having the same issue where they do not boot, but i do not know if they give the same error. 

The version is because it was the new at that time. Would it make sense to try version 23.05.04?

robimarko, 2024-09-11:
Then it gets even weirder, as there is no need for the broken flash reset as W25Q256JV has 4B OPCODE-s.

Can you remove the SoM RF can and take a picture of the Winbond NOR, I am interested in the production date

Zivertsen, 2024-09-12:
[image]

I hope this  is good enough

robimarko, 2024-09-12:
Well, considering this one is from 2021 then it has SFDP support for sure, mine has an mid-2018 NOR and it works just fine.

Have you done any more modifications?

I would strongly advise just trying a clean snapshot image (Without adding Quectel drivers or stuff like that)

Zivertsen, 2024-09-12:
There are some modification to the devicetree inorder to make some modules work. 

I have just build version 23.05.04 without any modification. And here the reboot command does not work as i also saw in the 23.05-rc3 build. 

Now i will try and add the "broken flash reset" and do the reboot test again and see if i get the same error as before.

Zivertsen, 2024-09-12:
With the snapshots image, the reboot command does not work.

robimarko, 2024-09-15:
Can you go to:
`/sys/bus/spi/devices/spi*/spi-nor` and dump the SFDP tables?

Also, adding a print here to make sure it actually gets identified as the JV variant would be great:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/mtd/spi-nor/winbond.c?h=v6.6.51#n35

Zivertsen, 2024-09-17:
This is what inside the spi-nor:
[image]

Regarding the print in winbond.c. I have some problems including the stdio.h for a printf call. I guess i need add something to a makefile some there for it the compile corrently? or is there another way of print a message out?

robimarko, 2024-09-17:
In the kernel, just use `pr_info` to print messages to the kernel log with the informational log level, kernel implements its own `printk`, not `printf`.

`pr_info` behaves the same as regular userspace `printf` would.

This is with kernel 5.15 or 6.6?
Either way, the NOR that is on your board is not responding with the SFDP tables and that is the issue why reboot doesn't work.
But it makes no sense since all of Winbond NOR should have SFDP tables since mid 2012 or so, I have couple of these from 2018 and they have them, I also bought some fresh 2023 models and they have it as well, so I am not sure whether that batch is faulty or they are clones(Bad ones).

Zivertsen, 2024-09-17:
I will give it a go with the `pr_info`

It is the snapshot image, so 6.6. But the spi-nor informations are the same on 5.15. 

It is strange that mine is not working and yours has no problems. The Habanero modules comes directly from 8devices them self.

robimarko, 2024-09-17:
Yeah, I know they come directly but the flash is broken.

It's hard to dig through further without the broken HW itself but the only path would be to dig through the MTD code to see where SFDP gets probed since it is clearly not responding to the `0x5a` command to read SFDP tables

Zivertsen, 2024-09-17:
I will see if i can find anything in the MTD, although it is probably beyond my understanding. :D

But thanks for the help so after.

Zivertsen, 2024-10-28:
Hi again

Do you know if it is possible to see if the SFDP tables are correcly on version 19.07.2?

robimarko, 2024-10-31:
Only by printing the checks as there is no sysfs yet for it in that old of a kernel