openwrt/openwrt #16365: Issue Report: Intel 82576 Network Card Instability in OpenWrt/ImmortalWrt 23 Versions

MOZHEYU, 2024-09-11:
### Describe the bug

Issue Report: Intel 82576 Network Card Instability in OpenWrt/ImmortalWrt 23 Versions

Environment Information:

Hardware: Intel 82576 Gigabit Network Connection (4 ports)
Virtualization Environment: ESXi 6.7, network card passed through to VM
Affected Firmware:
ImmortalWrt 23 (Kernel version: 5.15.162)
OpenWrt 23 (Kernel version: 5.15.162)
Properly Working Firmware:
ImmortalWrt 21 (Kernel version: 5.4.255)
OpenWrt 21 (Kernel version: 5.4.255)
Ubuntu 22.04
Problem Description:
In ImmortalWrt 23 and OpenWrt 23, the Intel 82576 network card exhibits instability, manifesting as repeated card resets and link interruptions. The same hardware and configuration work normally in ImmortalWrt 21, OpenWrt 21, and Ubuntu 22.04.

Detailed Symptoms:

Repeated network card resets: "igb 0000:0b:00.0 eth0: Reset adapter"
Transmission queue timeout: "NETDEV WATCHDOG: eth0 (igb): transmit queue 0 timed out"
Fluctuating bridge interface status
Relevant Logs (ImmortalWrt 23):
```
2.1270021 igb: Intel(R) Gigabit Ethernet Network Driver
2.3361101 igb 0000：0b:00.0: added PHC on eth0
2.341155] igb 0000：0b:00.0: Intel(R) Gigabit Ethernet Network Connection
2.3447111 igb 0000：0b:00.0:eth0:(PCIe:5.0Gb/s:unknown)00:1b:21:89:5b:3b
2.3491231 igb 0000：0b:00.0: eth0:PBA No: E43709-003
5.0805721 igb 0000：0b:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
21.075025] NETDEV WATCHDOG: eth0 (igb): transmit queue 0 timed out
21.4205951 igb 0000：0b:00.0 eth0: Reset adapter
21.6303781 br-lan: port 1(eth0) entered disabled state
23.9505441 igb 0000：0b:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
24.070190] br-lan: port 1(eth0) entered blocking state
24.0741051 br-lan: port 1(eth0) entered forwarding state
34.1100821 igb 0000：0b:00.0 eth0: Reset adapter
34.3102711 br-lan: port 1(eth0) entered disabled state
```
[image]

Version Comparison:
Kernel Versions:
ImmortalWrt 21/OpenWrt 21
Kernel version: 5.4.255

ImmortalWrt 23/OpenWrt 23
Kernel version: 5.15.162

igb Driver Versions:
ImmortalWrt 21/OpenWrt 21: 5.14.16

ImmortalWrt 23/OpenWrt 23: Version information unavailable, but date is latest (based on modinfo output)

Attempted Solutions:

Adjusting network card parameters (ethtool -K eth0 gso off tso off gro off)
Modifying kernel parameters (net.core.netdev_budget, net.core.netdev_budget_usecs)
Disabling network card power-saving features
Checking PCI bus speed
Examining system load and memory allocation
Testing Process:

ImmortalWrt 21 and OpenWrt 21 work normally under the same hardware and virtualization environment
Ubuntu 22.04 also works normally under the same configuration
The issue occurs consistently in both ImmortalWrt 23 and OpenWrt 23
The problem persists, unaffected by system reboots or network card resets
Additional Relevant Information:

ESXi Version: 6.7
VM Configuration: All 4 ports of the 82576 network card are passed through
Network Card Firmware Version: To be confirmed

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

x86/64

### Device

VMware, Inc. VMware7,1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MOZHEYU, 2024-09-11:
The Intel Corporation 82576 Gigabit Network Connection network card is currently not usable. Switching from Firewall4 (Nftables) to Iptables allows it to work properly.

brada4, 2024-09-12:
No quad port 82576 in vmware 6.7 compat list?