openwrt/openwrt #16395: Segfault in ucode

petrn, 2024-09-15:
### Describe the bug

In activating a Luci menu item there is a segfault in Ucode library. This uses nginx, not uhttpd.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

x86/64

### Device

vmWare VM

### Image kind

Self-built image

### Steps to reproduce

Click on the custom menu item

### Actual behaviour

A custom item in Luci menu which used to work in 22.03 is showing Bad gateway error page. The log contains:

```
nginx[1852]: 2024/09/15 07:17:15 [error] 1898#0: *30 upstream prematurely closed connection while reading response header from upstream, client: 192.168.176.165, server: _lan, request: "GET /cgi-bin/luci/admin/services/router HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "st-router-1-doma.doma.uk.to", referrer: "https://st-router-1-doma.doma.uk.to/cgi-bin/luci/"
kern.info kernel: [53024.011909] ucode[29182]: segfault at 65636172748b ip 00007fb434c8ccf5 sp 00007ffc246a6cc8 error 4 in libucode.so.20230711[7fb434c84000+18000]
kern.info kernel: [53024.012335] Code: e2 0f 83 ea 05 80 fa 01 77 05 83 c8 20 88 07 c3 48 85 ff 74 0a 48 8b 7f 20 e9 a6 db 00 00 90 31 c0 c3 48 89 f8 48 85 ff 74 08 <48> 8b 47 20 48 83 c0 38 c3 41 57 48 89 f2 41 56 41 55 41 54 49 89
S
```

### Expected behaviour

the redirection should get to the correct page without showing any error

### Additional info

The failing menu is just a redirection to another path served by a different application:
```
{
	"admin/services/router": {
		"title": "ST Router",
		"order": 1,
		"action": {
			"type": "template",
			"path": "st_router"
		},
		"depends": {
		}
	}
}
```

### Diffconfig

```text
glibc build with multiple custom packages
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-09-18:
Can you backtrace crash in gdb?

petrn, 2024-09-18:
I have not tried to debug it with gdb, I have made a workaround by creating another action type in dispatcher.uc which directly calls http.redirect and now it is working at the expense of having to edit dispatcher.uc, so I avoid using template for this

brada4, 2024-09-18:
Try to repeat crash in command line so that it can be tried on other platform.
Likely diffconfig with c flags and core lib selections (s still worth it.

brada4, 2024-09-18:
No mater how "wrong" is your configuration, it should not end in crash.

petrn, 2024-09-18:
You are correct, that is why I have posted it here. I have tried the same menu json on another VM not using Nginx and glibc and that one does not crash and shows an error page like my RB5009. So the problem is either related to Nginx or to glibc. In another VM I have replaced uhttpd with ngingx and it works the same as withe uhttpd. So the issue seems to be related to glibc, at least that is what my tests are indicating.