openwrt/openwrt #16396: mt7620 crash since kernel > 6.6.45

ptpt52, 2024-09-15:
### Describe the bug

crash log:
```
<1>[  276.393904] CPU 0 Unable to handle kernel paging request at virtual address 7036837c, epc == 7036837c, ra == 80004574
<4>[  276.415127] Oops[#1]:
<4>[  276.419646] CPU: 0 PID: 11573 Comm: jshn Tainted: G           O       6.6.50 #0
<4>[  276.434213] $ 0   : 00000000 00000001 00000002 806d1698
<4>[  276.444644] $ 4   : 7fce7d9f 00000002 ffffff83 82979e00
<4>[  276.455068] $ 8   : 00000000 00000000 00000001 81995965
<4>[  276.465489] $12   : 00000000 8076eccc 00000003 85d5a630
<4>[  276.475911] $16   : 7fce7d9f 81995960 00000000 85d83e00
<4>[  276.486334] $20   : 85c8ac80 806d07a0 82ae3640 807b0000
<4>[  276.496758] $24   : 00000000 843b1cc4                  
<4>[  276.507180] $28   : 843b0000 843b1dd8 843b1df0 80004574
<4>[  276.517606] Hi    : 00000002
<4>[  276.523329] Lo    : 00000003
<4>[  276.529053] epc   : 7036837c 0x7036837c
<4>[  276.536692] ra    : 80004574 arch_align_stack+0x50/0x70
<4>[  276.547115] Status: 1100e403	KERNEL EXL IE 
<4>[  276.555454] Cause : 10800008 (ExcCode 02)
<4>[  276.563432] BadVA : 7036837c
<4>[  276.569156] PrId  : 00019650 (MIPS 24KEc)
<4>[  276.577133] Modules linked in: ksmbd rt2800soc(O) rt2800mmio(O) rt2800lib(O) qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rt2x00soc(O) rt2x00mmio(O) rt2x00lib(O) rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nft_fib_inet mac80211(O) libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash
<4>[  276.577771]  nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_pptp nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat(O) chacha_mips cdc_wdm br_netfilter natflow(O) natcap(O) nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 fuse ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet
<4>[  276.756961]  ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nls_ucs2_utils cifs_arc4 asn1_decoder nfs msdos ip6_gre ip_gre gre ip6_udp_tunnel udp_tunnel sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel tun lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv sha3_generic drbg md5 hmac geniv rng ecb des_generic libdes cmac arc4 uas mtk_sd mmc_block cqhci usb_storage mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug(O) vfat fat ext4 mbcache jbd2
<4>[  276.936754]  exfat usbcore nls_base usb_common mii crc32c_generic [last unloaded: ksmbd]
<4>[  277.132417] Process jshn (pid: 11573, threadinfo=71d86be0, task=a21c8eab, tls=77ef9df4)
<4>[  277.148373] Stack : 77e850ff 77dc6bc0 85d83e00 85c0c400 85d83800 801d39d4 00000006 00000000
<4>[  277.165055]         85d8385c 80820000 00000012 00000000 00000003 0041ff14 00401d50 00400000
<4>[  277.181732]         00420028 00400034 00000008 00401140 00000022 00000002 77dc6bc0 806d1698
<4>[  277.198409]         77db4000 85d83800 00000034 00000000 00000000 00000000 82ae36e0 00000000
<4>[  277.215084]         00000100 ecbcfece 808241dc fffffff8 85d83800 80823890 00000006 00000000
<4>[  277.231762]         ...
<4>[  277.236629] Call Trace:
<4>[  277.236634] 
<4>[  277.244604] [<801d39d4>] load_elf_binary+0xc44/0x1590
<4>[  277.254839] [<80174d54>] bprm_execve+0x1ec/0x578
<4>[  277.264053] [<801747a4>] copy_string_kernel+0x104/0x240
<4>[  277.274472] [<801756f0>] do_execveat_common+0x1b4/0x240
<4>[  277.284884] [<8000fdec>] do_page_fault+0xd4/0x554
<4>[  277.294419] [<80176378>] sys_execve+0x34/0x48
<4>[  277.303111] [<8000dee0>] syscall_common+0x34/0x58
<4>[  277.312502] 
<4>[  277.315456] Code: (Bad address in epc)
<4>[  277.315456] 
<4>[  277.325856] 
<4>[  277.329048] ---[ end trace 0000000000000000 ]---
```

no crash for `kernel version <= 6.6.45` 

### OpenWrt version

r27609-cf79b823996

### OpenWrt release

24.04_b202409140123

### OpenWrt target/subtarget

ramips/mt7620

### Device

Youku YK-L1c

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2024-09-15:
ping @hauke @dangowrt

namiltd, 2024-09-15:
Same issue on TP-Link Archer C5 v4, system restarts randomly on kernel above 6.6.45

aiamadeus, 2024-09-15:
Try revert https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.6.51&id=53f17409abf61f66b6f05aff795e938e5ba811d1

aiamadeus, 2024-09-15:
> https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/diff/fs/binfmt_flat.c?id=v6.6.46&id2=v6.6.47

https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.6.47 # binfmt_flat: Fix corruption when not offsetting data start

namiltd, 2024-09-15:
Kernel in snapshot has just been updated to 6.6.51. Testing has started.

namiltd, 2024-09-15:
The crash restarts seem to have stopped.
However, there is still a problem with frequent "phy1: Hardware restart was requested" after a higher load, after which 2.4GHz WiFi stops working.  I have temporarily enabled a script that restarts the Wi-Fi network in cron when it detects such a situation. However, this is a different problem and requires a separate look at it.

ptpt52, 2024-09-16:
crash again on kernel 6.6.51
```
<1>[  924.808612] CPU 0 Unable to handle kernel paging request at virtual address 70367f14, epc == 70367f14, ra == 80004574
<4>[  924.829814] Oops[#1]:
<4>[  924.834336] CPU: 0 PID: 25099 Comm: swconfig Tainted: G           O       6.6.51 #0
<4>[  924.849593] $ 0   : 00000000 00000001 00000002 fffff000
<4>[  924.860023] $ 4   : 7fc54000 7fc54000 00000001 00000000
<4>[  924.870445] $ 8   : 00000000 80629b1c 00000014 808e907f
<4>[  924.880867] $12   : 3ef2858b 3c605a55 d13feb91 b9379f1c
<4>[  924.891292] $16   : 7fc54000 00000001 85dbfea0 8457de00
<4>[  924.901714] $20   : 85bfd3c0 8457da5c 81da78c0 00000001
<4>[  924.912137] $24   : 275a8d4e 80798f8f                  
<4>[  924.922559] $28   : 85d12000 85d13d20 85d13de8 80004574
<4>[  924.932984] Hi    : 00000008
<4>[  924.938708] Lo    : 0000000c
<4>[  924.944432] epc   : 70367f14 0x70367f14
<4>[  924.952072] ra    : 80004574 arch_align_stack+0x50/0x70
<4>[  924.962495] Status: 1100e403	KERNEL EXL IE 
<4>[  924.970835] Cause : 10800008 (ExcCode 02)
<4>[  924.978812] BadVA : 70367f14
<4>[  924.984537] PrId  : 00019650 (MIPS 24KEc)
<4>[  924.992514] Modules linked in: ksmbd rt2800soc(O) rt2800mmio(O) rt2800lib(O) qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rt2x00soc(O) rt2x00mmio(O) rt2x00lib(O) rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nft_fib_inet mac80211(O) libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash
<4>[  924.993149]  nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_pptp nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat(O) chacha_mips cdc_wdm br_netfilter natflow(O) natcap(O) nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 fuse ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet
<4>[  925.172329]  ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nls_ucs2_utils cifs_arc4 asn1_decoder nfs msdos ip6_gre ip_gre gre ip6_udp_tunnel udp_tunnel sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel tun lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv sha3_generic drbg md5 hmac geniv rng ecb des_generic libdes cmac arc4 uas mtk_sd mmc_block cqhci usb_storage mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug(O) vfat fat ext4 mbcache jbd2
<4>[  925.352115]  exfat usbcore nls_base usb_common mii crc32c_generic [last unloaded: ksmbd]
<4>[  925.547780] Process swconfig (pid: 25099, threadinfo=cf1aface, task=7a431a5b, tls=77e9cdf4)
<4>[  925.564430] Stack : ffffffff 00000000 00005c08 80a5c680 8457da00 80173c04 85bfd3c0 00005c08
<4>[  925.581110]         80a5c680 00000000 00000001 8014e304 846bc780 00000003 00000000 00a032ce
<4>[  925.597786]         00000000 8082a508 00000001 00000014 8457de00 82bdda00 8457da5c 80368054
<4>[  925.614464]         00000001 80031008 8457da00 859cf200 00000000 1fd8711e ffffffff 85bfd3c0
<4>[  925.631141]         8457da54 80120bf8 82bdda00 844bbb00 00000100 801d2c3c 00000034 2c14baaa
<4>[  925.647820]         ...
<4>[  925.652687] Call Trace:
<4>[  925.652691] 
<4>[  925.660662] [<80173c04>] setup_arg_pages+0x48/0x2cc
<4>[  925.670545] [<8014e304>] free_unref_page+0xa0/0x118
<4>[  925.680890] [<80368054>] get_random_u32+0x64/0xdc
<4>[  925.690272] [<80031008>] flush_itimer_signals+0x34/0x5c
<4>[  925.701304] [<80120bf8>] arch_pick_mmap_layout+0x11c/0x1b0
<4>[  925.712545] [<801d2c3c>] load_elf_phdrs+0x78/0xcc
<4>[  925.722683] [<801d3108>] load_elf_binary+0x2e4/0x1540
<4>[  925.732902] [<8016abb0>] __kernel_read+0x164/0x2ac
[... cut ...] cve+0x1ec/0x578
<4>[  925.752600] [<801747a8>] copy_string_kernel+0x104/0x240
<4>[  925.763328] [<801756f4>] do_execveat_common+0x1b4/0x240
<4>[  925.773901] [<8000fdec>] do_page_fault+0xd4/0x554
<4>[  925.783435] [<8017637c>] sys_execve+0x34/0x48
<4>[  925.792124] [<8000dee0>] syscall_common+0x34/0x58
<4>[  925.801671] 
<4>[  925.804627] Code: (Bad address in epc)
<4>[  925.804627] 
<4>[  925.815029] 
<4>[  925.818167] ---[ end trace 0000000000000000 ]---
```

@namiltd 
no crash on your device?

namiltd, 2024-09-16:
I haven't noticed any crashes yet, but maybe it's just luck.

ptpt52, 2024-09-16:
I notice that this issue happens on mt798x:
```
<1>[97887.593961] Unable to handle kernel paging request at virtual address 00510b2901f7ec90
<1>[97887.601897] Mem abort info:
<1>[97887.604682]   ESR = 0x0000000096000004
<1>[97887.608705]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[97887.614002]   SET = 0, FnV = 0
<1>[97887.617066]   EA = 0, S1PTW = 0
<1>[97887.620192]   FSC = 0x04: level 0 translation fault
<1>[97887.625051] Data abort info:
<1>[97887.627944]   ISV = 0, ISS = 0x00000004, ISS2 = 0x00000000
<1>[97887.633410]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
<1>[97887.638455]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
<1>[97887.643748] [00510b2901f7ec90] address between user and kernel address ranges
<0>[97887.650868] Internal error: Oops: 0000000096000004 [#1] SMP
<7>[97887.656422] Modules linked in:
<7>[97887.656425] CMD[switch_ports_st] args=[/bin/sh]
<7>[97887.663972]  ksmbd qcserial pppoe ppp_async option l2tp_ppp cdc_mbim wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 nft_fib_inet mt7915e(O) mt76_connac_lib(O) mt76(O) mac80211(O) libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cp210x chacha_neon cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_DNETMAP(O) xt_DHCPMAC(O) xt_CLASSIFY xfrm_interface usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake poly1305_neon nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit
<7>[97887.664155]  nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c libchacha ipvlan iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables compat_xtables(O) compat(O) cdc_wdm br_netfilter natflow(O) natcap(O) nf_nat nf_conntrack crypto_safexcel fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev(O) xt_set ip_set_list_set
<7>[97887.753995]  ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink jool_siit(O) jool(O) nf_defrag_ipv4 jool_common(O) ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nls_ucs2_utils cifs_arc4 asn1_decoder nfs msdos bonding tls ip6_gre ip_gre gre nat46(O) nf_defrag_ipv6 ip6_vti ip_vti sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun xfrm_user xfrm_ipcomp af_key xfrm_algo lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_arm64 sha1_ce sha1_generic seqiv md5 echainiv geniv des_generic libdes cbc authencesn
<7>[97887.842996]  authenc arc4 nls_iso8859_1 ntfs3 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug(O) vfat fat exfat dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax usbcore usb_common aquantia mii tpm encrypted_keys trusted [last unloaded: ksmbd]
<7>[97887.963956] CPU: 1 PID: 30276 Comm: switch_ports_st Tainted: G           O       6.6.50 #0
<7>[97887.972199] Hardware name: GL.iNet GL-MT6000 (DT)
<7>[97887.976886] pstate: a0400005 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
<7>[97887.983827] pc : __kmem_cache_alloc_node+0xb0/0x25c
<7>[97887.988694] lr : __kmem_cache_alloc_node+0x3c/0x25c
<7>[97887.993554] sp : ffffffc08a44bbd0
<7>[97887.996852] x29: ffffffc08a44bbd0 x28: ffffff800ba74ca0 x27: 0000000000000000
<7>[97888.003968] x26: 0000000000000000 x25: 0000000000000cc0 x24: 000000000000001a
<7>[97888.011083] x23: ffffffc080bb8000 x22: 00000000ffffffff x21: 0000000000000000
<7>[97888.018199] x20: 0000000000000cc0 x19: ffffff8000001400 x18: ffffffc08a44ba60
<7>[97888.025315] x17: 0000000000000000 x16: ffffffc08096cf38 x15: 000000046474e552
<7>[97888.032431] x14: 0000000000000010 x13: 000000046474e552 x12: 0000000000000010
<7>[97888.039547] x11: 0000000000000001 x10: 0000000000000c40 x9 : 0000000000000c40
<7>[97888.046662] x8 : 000000000048f3c0 x7 : 000000000048f3c0 x6 : 0000000000000005
<7>[97888.053778] x5 : 00000000000faa51 x4 : 00000000000faa50 x3 : 69f682dd812005bf
[... cut ...]
<7>[97888.082079]  bprm_execve+0x248/0x534
<7>[97888.085641]  do_execveat_common+0x17c/0x208
<7>[97888.089807]  __arm64_sys_execve+0x34/0x44
<7>[97888.093800]  invoke_syscall.constprop.0+0x4c/0xe0
<7>[97888.098487]  do_el0_svc+0x3c/0xb8
<7>[97888.101786]  el0_svc+0x18/0x4c
<7>[97888.104828]  el0t_64_sync_handler+0x118/0x124
<7>[97888.109169]  el0t_64_sync+0x150/0x154
<0>[97888.112818] Code: b9402a62 f9405e63 8b020001 dac00c21 (f8626802) 
<4>[97888.118891] ---[ end trace 0000000000000000 ]---
```

danpawlik, 2024-09-16:
@ptpt52 but in the stacktrace is:
```
Comm: switch_ports_st Tainted: G           O       6.6.50 #
```

Are You sure you have 6.6.51?

ptpt52, 2024-09-16:
@danpawlik  it is kernel 6.6.50 but I said that the crash happen on mt798x too.

ptpt52, 2024-09-16:
Based on my testing, I found that reverting commit fd6ce0dea95272eec20eb222b08c7e685827caee can resolve this issue. It has been observed on mt7620 and mt798x, with a particularly high occurrence rate on mt7620. After reverting this commit, the issue no longer appears.

ptpt52, 2024-09-16:
ping @hauke

namiltd, 2024-09-17:
Isn't it the case that CONFIG LIST HARDENED enables additional code checking? Disabling the checks won't fix this code.

namiltd, 2024-09-23:
With CONFIG_LIST_HARDENED disabled, the restarts actually stopped.

danpawlik, 2024-09-23:
@namiltd indeed, and I think that it will also help solve bug: https://github.com/openwrt/openwrt/issues/16273 .

hauke, 2024-09-26:
Please try to activate `CONFIG_DEBUG_LIST` in addition and check if you see any error messages related to lists.
Some of the error checks in `CONFIG_LIST_HARDENED` are done by dereferencing a pointer.  when you activate `CONFIG_DEBUG_LIST` it will always run the slow debug code which returns proper error messages.

See this code: https://elixir.bootlin.com/linux/v6.6.52/source/include/linux/list.h#L71

namiltd, 2024-10-07:
> Please try to activate `CONFIG_DEBUG_LIST` in addition and check if you see any error messages related to lists. Some of the error checks in `CONFIG_LIST_HARDENED` are done by dereferencing a [...]

@ptpt52 Can you do this?

ptpt52, 2024-10-09:
@hauke @namiltd  nothing help but crash again with CONFIG_LIST_HARDENED enabled
and with CONFIG_DEBUG_LIST nothing help, no special logs

crash log:
```
1>[  230.897713] CPU 0 Unable to handle kernel paging request at virtual address 70364fd8, epc == 70364fd8, ra == 80004574
<4>[  230.908650] Oops[#1]:
<4>[  230.910980] CPU: 0 PID: 9968 Comm: swconfig Tainted: G           O       6.6.54 #0
<4>[  230.918714] $ 0   : 00000000 00000001 00000002 fffff000
<4>[  230.924080] $ 4   : 7fe32000 7fe32000 00000001 8595f530
<4>[  230.929433] $ 8   : 00000010 8062554c 00000003 00000008
<4>[  230.934786] $12   : 00459000 8596fd0c 8596fd0c ffffff00
<4>[  230.940140] $16   : 7fe32000 00000001 8593b3a8 85d51c00
<4>[  230.945493] $20   : 85c97d20 85d5105c 846f6460 00000001
<4>[  230.950847] $24   : 00000000 ffffffff                  
<4>[  230.956200] $28   : 85c24000 85c25d20 85c25de8 80004574
<4>[  230.961556] Hi    : 554a9555
<4>[  230.964493] Lo    : 7ffaa000
<4>[  230.967430] epc   : 70364fd8 0x70364fd8
<4>[  230.971351] ra    : 80004574 arch_align_stack+0x50/0x70
<4>[  230.976708] Status: 1100f403	KERNEL EXL IE 
<4>[  230.980993] Cause : 10800008 (ExcCode 02)
<4>[  230.985083] BadVA : 70364fd8
<4>[  230.988020] PrId  : 00019650 (MIPS 24KEc)
<4>[  230.992110] Modules linked in: ksmbd rt2800soc(O) rt2800mmio(O) rt2800lib(O) qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rt2x00soc(O) rt2x00mmio(O) rt2x00lib(O) rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nft_fib_inet mt76x2e(O) mt76x2_common(O) mt76x02_lib(O) mt76(O) mac80211(O) libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_numgen nft_nat
<4>[  230.992782]  nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_pptp nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat(O) chacha_mips cdc_wdm br_netfilter natflow(O) natcap(O) nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 fuse ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set
<4>[  231.084699]  ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nls_ucs2_utils cifs_arc4 asn1_decoder nfs msdos ip6_gre ip_gre gre ip6_udp_tunnel udp_tunnel sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel tun lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv sha3_generic drbg md5 hmac geniv rng ecb des_generic libdes cmac arc4 uas usb_storage leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug(O) vfat fat ext4 mbcache jbd2 exfat usbcore nls_base
<4>[  231.176776]  usb_common mii crc32c_generic [last unloaded: ksmbd]
<4>[  231.274698] Process swconfig (pid: 9968, threadinfo=65eed57d, task=e0e053dc, tls=77e23df4)
<4>[  231.283148] Stack : ffffffff 00000000 00005c36 00005c36 85d51000 801727ac 00000001 85c97d20
<4>[  231.291720]         85c975a0 807a7280 00000001 8014d1f0 85c61780 00000003 00000000 80164a60
<4>[  231.300288]         00000000 00000000 00000000 00000000 0000000a 80191390 00000000 8595f200
<4>[  231.308853]         00000001 80030e84 85d51000 8595f200 00000000 0ba99306 ffffffff 85c97d20
<4>[  231.317423]         85d51054 8011fee0 85d53180 85a7d900 00000100 801d1150 00000034 483fa8cc
<4>[  231.325991]         ...
<4>[  231.328493] Call Trace:
<4>[  231.328498] 
<4>[  231.332504] [<801727ac>] setup_arg_pages+0x48/0x2cc
<4>[  231.337841] [<8014d1f0>] free_unref_page+0xa0/0x118
<4>[  231.343000] [<80164a60>] filp_close+0x14/0x34
<4>[  231.347467] [<80191390>] do_close_on_exec+0x108/0x154
<4>[  231.352805] [<80030e84>] flush_itimer_signals+0x34/0x5c
<4>[  231.358481] [<8011fee0>] arch_pick_mmap_layout+0x1a4/0x1c4
[... cut ...]
<4>[  231.379679] [<80173900>] bprm_execve+0x1ec/0x578
<4>[  231.384410] [<80173350>] copy_string_kernel+0x104/0x240
<4>[  231.389922] [<8017429c>] do_execveat_common+0x1b4/0x240
<4>[  231.395273] [<8000fdac>] do_page_fault+0xd4/0x554
<4>[  231.400252] [<80174f24>] sys_execve+0x34/0x48
<4>[  231.404718] [<8000dea0>] syscall_common+0x34/0x58
<4>[  231.409713] 
<4>[  231.411232] Code: (Bad address in epc)
<4>[  231.411232] 
<4>[  231.416563] 
<4>[  231.418790] ---[ end trace 0000000000000000 ]---
```

namiltd, 2024-10-09:
Have you changed this option in target/linux/generic/config-6.6 ?

namiltd, 2024-10-12:
What is interesting is that the information at the time of the crash does not contain anything about lists.

namiltd, 2024-10-13:
What is interesting is that the information at the time of the crash does not contain anything about lists.
The list contains an invalid address. During the test it will encounter such an address and a crash will occur. When the test is not performed, the problem does not exist.
The swconfig module needs to be analyzed. I suggest adding a few dev_err in places where you operate on lists to find the exact line where the error occurs.

ptpt52, 2024-10-15:
> Probably fixed by [506d0c4](https://github.com/openwrt/openwrt/commit/506d0c421b1b0df6adb75506b3b20622b264b15f) Before applying the patch, running tcpdump on wan interface on Xiaomi mi-r3 resulted in a crash. After applying the patch, in this case it works stable

but mt7620 should never use the mtk_eth_soc driver.?

nbd168, 2024-10-16:
@ptpt52 from what I can tell, similar looking crashes were reproduced on both mt7620 and mediatek arm + mips hardware. The Xiaomi device mentioned here is mt7621 based, so it uses mtk_eth_soc. So my guess is that the issues on newer targets are fixed by https://github.com/openwrt/openwrt/commit/506d0c421b1b0df6adb75506b3b20622b264b15f, however there is likely a completely different source of crashes remaining on mt7620.

closed 2024-10-16 by openwrt-bot

danpawlik, 2024-10-16:
@nbd168 I did not test, due I changed the wireless card, but I'm 90% sure, that similar issue is for mt7925: https://github.com/openwrt/openwrt/issues/16273

nbd168, 2024-10-16:
@danpawlik that one does not look similar or related to me at all. in the mt7620 crash case, the address is (semi-)random in a place that handles memory allocation. in the mt7625 case, the faulting address suggests that container_of was done on a NULL pointer, and some fields of the resulting pointer were dereferenced.

AuthorReflex, 2024-10-17:
> Probably fixed by [506d0c4](https://github.com/openwrt/openwrt/commit/506d0c421b1b0df6adb75506b3b20622b264b15f) Before applying the patch, running tcpdump on wan interface on Xiaomi mi-r3 in test environment resulted in a crash. After applying the patch, in this case it works stable PS: on real environment after patch is also stable

test #2: Xiaomi miwifi-mini
config: defaults, firewall redirect wan 22001->22
test: connect ssh to wan 22001, run tcpdump -i eth0.2
before patch: crash && reboot in few seconds
after patch: stable on 5+ minute, even with CONFIG_LIST_HARDENED=y

ptpt52, 2024-12-29:
the crash happens again in the latest code (kernel 6.6.67), even with CONFIG_LIST_HARDENED disabled

ptpt52, 2024-12-29:
crash report on latest code:
```
<1>[  474.109484] CPU 0 Unable to handle kernel paging request at virtual address 70360064, epc == 70360064, ra == 80004294
<4>[  474.120680] Oops[#1]:
<4>[  474.123022] CPU: 0 PID: 15281 Comm: timeout Tainted: G           O       6.6.67 #0
<4>[  474.130757] $ 0   : 00000000 00000001 00000002 806c568c
<4>[  474.136120] $ 4   : 7fc7af55 00000001 ffffff83 8180a100
<4>[  474.141474] $ 8   : 00000000 00000000 00000001 85aa8785
<4>[  474.146828] $12   : 00000000 8076326c 00000003 84754750
<4>[  474.152183] $16   : 7fc7af55 85aa8780 00000000 806c4794
<4>[  474.157538] $20   : 85a24300 77e97e38 85dcadc0 807a0000
<4>[  474.162893] $24   : 00000000 844c3cbc                  
<4>[  474.168246] $28   : 844c2000 844c3dd0 844c3de8 80004294
<4>[  474.173602] Hi    : 554a9555
<4>[  474.176540] Lo    : 7ffaa000
<4>[  474.179477] epc   : 70360064 0x70360064
<4>[  474.183399] ra    : 80004294 arch_align_stack+0x50/0x70
<4>[  474.188753] Status: 1100f403	KERNEL EXL IE 
<4>[  474.193037] Cause : 10800008 (ExcCode 02)
<4>[  474.197128] BadVA : 70360064
<4>[  474.200065] PrId  : 00019650 (MIPS 24KEc)
<4>[  474.204155] Modules linked in: ksmbd rt2800soc(O) rt2800mmio(O) rt2800lib(O) qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rt2x00soc(O) rt2x00mmio(O) rt2x00lib(O) rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nft_fib_inet mt76x2e(O) mt76x2_common(O) mt76x02_lib(O) mt76(O) mac80211(O) libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_numgen nft_nat
<4>[  474.204837]  nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_pptp nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat(O) chacha_mips cdc_wdm br_netfilter natflow(O) natcap(O) nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 fuse ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set
<4>[  474.296751]  ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nls_ucs2_utils cifs_arc4 asn1_decoder nfs msdos ip6_gre ip_gre gre ip6_udp_tunnel udp_tunnel sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel tun lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv sha3_generic drbg md5 kpp hmac geniv rng ecb des_generic libdes cmac arc4 uas usb_storage leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug(O) vfat fat ext4 mbcache jbd2 exfat usbcore
<4>[  474.388826]  nls_base usb_common mii crc32c_generic [last unloaded: ksmbd]
<4>[  474.487111] Process timeout (pid: 15281, threadinfo=4313ef53, task=da465070, tls=77e70df4)
<4>[  474.495558] Stack : 77dc5000 77dd7bc0 85a08600 00000001 843c5400 801ce2f8 00000100 80819580
<4>[  474.504131]         00000006 00000000 00000012 00000000 00000000 00000003 0046fa94 00000002
<4>[  474.512696]         00458da0 00400000 00470028 00400034 806c4794 00403ca0 00000000 00000007
<4>[  474.521263]         77dd7bc0 00000008 77dc5000 806c568c 00000034 00000000 00000000 00000000
<4>[  474.529830]         85dca500 00000000 00000100 04a22b1c 80819e34 fffffff8 843c5400 80819580
<4>[  474.538399]         ...
<4>[  474.540900] Call Trace:
<4>[  474.540905] 
<4>[  474.545243] [<801ce2f8>] load_elf_binary+0xc20/0x1540
<4>[  474.550935] [<80170754>] bprm_execve+0x1ec/0x578
<4>[  474.555829] [<80170d60>] copy_string_kernel+0x104/0x240
<4>[  474.561666] [<80171574>] do_execveat_common+0x1b4/0x240
<4>[  474.567174] [<8000fb9c>] do_page_fault+0x1e4/0x554
<4>[  474.572243] [<801721fc>] sys_execve+0x34/0x48
<4>[  474.576709] [<8000db80>] syscall_common+0x34/0x58
<4>[  474.581862] 
<4>[  474.583381] Code: (Bad address in epc)
<4>[  474.583381] 
<4>[  474.588711] 
<4>[  474.590423] ---[ end trace 0000000000000000 ]---

```

namiltd, 2024-12-29:
24.10-rc4 or SNAPSHOT?

ptpt52, 2024-12-29:
> SNAPSHOT

snapshot

namiltd, 2024-12-29:
I've been testing 24.10.0-rc4 for a few days now and so far I haven't experienced any reboots on ramips/mt7620.

ptpt52, 2024-12-29:
>https://github.com/namiltd/openwrt/commit/cf9f8f3d40f12179c7421f7736ad8090dd27ea1c

gona give it a try.

namiltd, 2024-12-29:
This patch comes from https://github.com/openwrt/openwrt/pull/16845
There is currently a newer version but for me this is the most stable one.

AuthorReflex, 2024-12-29:
@ptpt52 try to enable hardened lists (i.e. rollback 74238413b13dbf58d9da6836f9a41179d6bcb02e)
In my case with hardened lists mi-r3 in real environment works stable (with disabled hardened lists vice versa crashes is happening)

ptpt52, 2024-12-30:
revert back to kernel 6.6.65, no crash.

DragonBluep, 2024-12-30:
I encountered a similar problem when I tried to open a second SSH session last week. Based on @ptpt52's log reports. I think this kernel oops is caused by calling `copy_string_kernel()` or `copy_strings()` located in https://github.com/gregkh/linux/blob/3ee19fb278a687efa3c643027f235f5ea9a6e9b4/fs/exec.c#L1946-L1970 when kernel execute a new program. This is a very rare situation and I can no longer reproduce it again. If you have time and can easily reproduce it. Maybe you can insert some debug code here as a start to track it.

```
(gdb) l*(do_execveat_common+0x1b4)
0x2374 is in do_execveat_common (fs/exec.c:1952).
1947            if (retval < 0)
1948                    goto out_free;
1949            bprm->exec = bprm->p;
1950
1951            retval = copy_strings(bprm->envc, envp, bprm);
1952            if (retval < 0)
1953                    goto out_free;
1954
1955            retval = copy_strings(bprm->argc, argv, bprm);
1956            if (retval < 0)
```

P.S.
a) This should be the same report. https://forum.openwrt.org/t/problem-on-rt2x00-router-reboot/213680.
b) This commit is suspicious https://github.com/gregkh/linux/commit/dcd46d897adb70d63e025f175a00a89797d31a43
c) I encountered this problem when the CPU was under heavy load. This should be a potential condition for the crash.

ptpt52, 2024-12-30:
try revert https://github.com/gregkh/linux/commit/dcd46d897adb70d63e025f175a00a89797d31a43
and testing now.

ptpt52, 2024-12-30:
> try revert [gregkh/linux@dcd46d8](https://github.com/gregkh/linux/commit/dcd46d897adb70d63e025f175a00a89797d31a43) and testing now.

no crash after revert the commit
@DragonBluep

ptpt52, 2025-01-07:
> > try revert [gregkh/linux@dcd46d8](https://github.com/gregkh/linux/commit/dcd46d897adb70d63e025f175a00a89797d31a43) and testing now.
> 
> no crash after revert the commit @DragonBluep

seems no true
still catch crash on latest build with the revert commit
```
<1>[   84.651547] CPU 0 Unable to handle kernel paging request at virtual address 70360160, epc == 70360160, ra == 80004294
<4>[   84.662462] Oops[#1]:
<4>[   84.664786] CPU: 0 PID: 4928 Comm: flock Tainted: G           O       6.6.69 #0
<4>[   84.672253] $ 0   : 00000000 00000001 00000002 fffff000
<4>[   84.677616] $ 4   : 7fbae000 7fbae000 00000001 819b4d2d
<4>[   84.682971] $ 8   : 00000010 8061dacc 0010000b 80b68081
<4>[   84.688325] $12   : 809c46e0 8180b60c 00000001 ffffff00
<4>[   84.693679] $16   : 7fbae000 00000001 85ab9558 845a7a00
<4>[   84.699033] $20   : 85b0c1e0 859daa5c 85b37aa0 00000001
<4>[   84.704389] $24   : 00000000 ffffffff                  
<4>[   84.709742] $28   : 85aa4000 85aa5d20 85aa5de8 80004294
<4>[   84.715096] Hi    : 554a9555
<4>[   84.718035] Lo    : 7ffaa000
<4>[   84.720972] epc   : 70360160 0x70360160
<4>[   84.724896] ra    : 80004294 arch_align_stack+0x50/0x70
<4>[   84.730249] Status: 1100f403	KERNEL EXL IE 
<4>[   84.734533] Cause : 10800008 (ExcCode 02)
<4>[   84.738623] BadVA : 70360160
<4>[   84.741560] PrId  : 00019650 (MIPS 24KEc)
<4>[   84.745651] Modules linked in: rt2800soc(O) rt2800mmio(O) rt2800lib(O) qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rt2x00soc(O) rt2x00mmio(O) rt2x00lib(O) rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nft_fib_inet mt76x2e(O) mt76x2_common(O) mt76x02_lib(O) mt76(O) mac80211(O) libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_numgen nft_nat
<4>[   84.746316]  nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_pptp nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat(O) chacha_mips cdc_wdm br_netfilter natflow(O) natcap(O) nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 fuse ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set
<4>[   84.837702]  ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 ksmbd nls_ucs2_utils cifs_arc4 asn1_decoder nfs msdos ip6_gre ip_gre gre ip6_udp_tunnel udp_tunnel sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel tun lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv sha3_generic drbg md5 kpp hmac geniv rng ecb des_generic libdes cmac arc4 uas usb_storage leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug(O) vfat fat ext4 mbcache jbd2 exfat usbcore
<4>[   84.929783]  nls_base usb_common mii crc32c_generic
<4>[   85.026555] Process flock (pid: 4928, threadinfo=6103b4bc, task=7cada061, tls=77effdf4)
<4>[   85.034738] Stack : 00001601 85b0c780 000042de 80a1a140 859daa00 8016faa8 85b0c1e0 85b0c780
<4>[   85.043309]         807a1280 00000001 85aa5de8 8014ad78 85a4b8c0 00000001 00000000 80161fc0
<4>[   85.051876]         00000000 00000000 00000000 00000000 0000000d 8018e608 00000000 819b4a00
<4>[   85.060441]         00000001 80030918 859daa00 819b4a00 00000000 a27d38e1 ffffffff 85b0c1e0
<4>[   85.069008]         859daa54 8011e358 85b3dd80 85b3ea00 00000100 801cd494 00000034 7b4e2350
<4>[   85.077578]         ...
<4>[   85.080079] Call Trace:
<4>[   85.080084] 
<4>[   85.084088] [<8016faa8>] setup_arg_pages+0x48/0x2cc
<4>[   85.089086] [<8014ad78>] free_unref_page+0x40/0x118
<4>[   85.094076] [<80161fc0>] filp_close+0x14/0x34
<4>[   85.098536] [<8018e608>] do_close_on_exec+0x108/0x154
<4>[   85.103876] [<80030918>] flush_itimer_signals+0x34/0x5c
[... cut ...]
<4>[   85.130101] [<80170784>] bprm_execve+0x1ec/0x578
<4>[   85.134825] [<80170d90>] copy_string_kernel+0x104/0x240
<4>[   85.140173] [<8017155c>] do_execveat_common+0x16c/0x1b8
<4>[   85.145515] [<8000fa8c>] do_page_fault+0xd4/0x554
<4>[   85.150498] [<80172184>] sys_execve+0x34/0x48
<4>[   85.154963] [<8000db80>] syscall_common+0x34/0x58
<4>[   85.159956] 
<4>[   85.161476] Code: (Bad address in epc)
<4>[   85.161476] 
<4>[   85.166806] 
<4>[   85.168451] ---[ end trace 0000000000000000 ]---
```

namiltd, 2025-01-07:
I haven't noticed this problem on the router I have (Mercusys AC12G v1 (8M)). I have mt7620/mt7612e in it but in a configuration without USB. Try to remove kmod-usb2 kmod-usb-ohci kmod-usb-ledtrig-usbport from the configuration and test.

DragonBluep, 2025-01-08:
@ptpt52 @nbd168 Now I can reproduce this bug with this command:
```
while true ; do var=$(iw phy0-ap0 station dump | grep "Station") ; done
```
MT7620 usually crashes within 10 minutes. But I can't reproduce it on other SoCs.

1. The stack track log from @ptpt52 is correct. The oops happens on `get_random_u32_below(PAGE_SIZE)`
https://elixir.bootlin.com/linux/v6.6.67/source/arch/mips/kernel/process.c#L715
2. I think it's not a compiler bug because the same code can run smoothly on mt7628.
3. The source code of `get_random_u32_below()` should be correct.
4. It seems that some instructions trigger the hardware defect.
5. As a workaround, always return the original parameter `sp` in `arch_align_stack()` seems work.

namiltd, 2025-01-08:
@DragonBluep On my device your script does not cause a crash. 
My router is Mercusys AC12G v1 (8M). It is almost the same as TP-Link Archer C5 v4 but it does not have USB.
It has:
 - Mediatek MT7620DA - main chipset
 - Mediatek MT7612EN - AC WIFI
 - Realtek RTL8367S - Switch LAN\WAN

I suspected that the reason I don't have a crash is the lack of USB, but it may also be because my router has an additional switch.
The issue may be in the switch driver built into the MT7620 - dsa: mediatek,MT7530. Have there been any changes to kernel 6.6.66 in this driver?

DragonBluep, 2025-01-09:
> @DragonBluep On my device your script does not cause a crash. 

How long did you run the script? Please keep it running for at least 2 hours. Maybe you need to revert https://github.com/openwrt/openwrt/commit/74238413b13dbf58d9da6836f9a41179d6bcb02e.

>My router is Mercusys AC12G v1 (8M). It is almost the same as TP-Link Archer C5 v4 but it does not have USB. It has:
> 
> * Mediatek MT7620A - main chipset
> * Mediatek MT7612EN - AC WIFI
> * Realtek RTL8367S - Switch LAN\WAN (no DSA)
> 
> I suspected that the reason I don't have a crash is the lack of USB,

No. Me neither.

> but it may also be because my router has an additional switch.

No. I spent a lot of time locating this error. It's clear that it crashes when `get_random_u32_below(PAGE_SIZE)` returned some value. Not a common problem like NULL pointer. No idea.

>The issue may be in the switch driver built into the MT7620 - dsa: mediatek,MT7530. Have there been any changes to kernel 6.6.66 in this driver?
> 
> I only found this: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.6.66&id=b8a0428f1eb2b3d9764ccd697010eba2cfa15046

ptpt52, 2025-01-09:
I just apply this hack
```diff
diff --git a/arch/mips/kernel/process.c b/arch/mips/kernel/process.c
index 738ca605d..280d30f05 100644
--- a/arch/mips/kernel/process.c
+++ b/arch/mips/kernel/process.c
@@ -713,7 +713,7 @@ unsigned long mips_stack_top(void)
  */
 unsigned long arch_align_stack(unsigned long sp)
 {
-       if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space)
+       if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space && 0)
                sp -= get_random_u32_below(PAGE_SIZE);
 
        return sp & ALMASK;
```

for now, no crash. Let's keep watching.

namiltd, 2025-01-09:
Due to optimizations, "&& 0" should be removed from the code during compilation anyway. So how is that supposed to change anything?

namiltd, 2025-01-09:
Maybe get_random_u32_below(PAGE_SIZE) returns a value greater than sp and this gives a negative result?

ptpt52, 2025-01-09:
> for now, no crash. Let's keep watching.

running hours and no crash till now.

namiltd, 2025-01-10:
Try this patch:
```diff
diff --git a/arch/mips/kernel/process.c b/arch/mips/kernel/process.c
index b630604..62d59a4 100644
--- a/arch/mips/kernel/process.c
+++ b/arch/mips/kernel/process.c
@@ -711,8 +711,8 @@ unsigned long mips_stack_top(void)
  */
 unsigned long arch_align_stack(unsigned long sp)
 {
-	if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space)
+	if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space && (sp > 0))
-		sp -= get_random_u32_below(PAGE_SIZE);
+		sp -= get_random_u32_below((PAGE_SIZE<sp)?PAGE_SIZE:sp);
 
 	return sp & ALMASK;
 }
```

KanjiMonster, 2025-01-10:
I don't think that `sp` will ever be 0 or near 0 (less than PAGE_SIZE). The error address in the kernel logs (EPC) is always 0x7*, IIRC the page at offset 0 is always unmapped (to catch NULL pointer accesses), and the stack grows downwards, so if it were close to 0 then there would be virtually no stack space.

Might be interesting to print sp before and after arch_align_stack(). Maybe the stack pointer isn't aligned to a page, or if it substracts too much, it runs out of space elsewhere.

ptpt52, 2025-01-10:
> @DragonBluep and @ptpt52 Try this patch:
> 
> ```diff
> diff --git a/arch/mips/kernel/process.c b/arch/mips/kernel/process.c
> index b630604..62d59a4 100644
> --- a/arch/mips/kernel/process.c
> +++ b/arch/mips/kernel/process.c
> @@ -711,8 +711,8 @@ unsigned long mips_stack_top(void)
>   */
>  unsigned long arch_align_stack(unsigned long sp)
>  {
> -	if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space)
> +	if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space && (sp > 0))
> -		sp -= get_random_u32_below(PAGE_SIZE);
> +		sp -= get_random_u32_below((PAGE_SIZE<sp)?PAGE_SIZE:sp);
>  
>  	return sp & ALMASK;
>  }
> ```

Let me try.

namiltd, 2025-01-10:
A small correction (sign <= and sp+1):
```diff
diff --git a/arch/mips/kernel/process.c b/arch/mips/kernel/process.c
index b630604..62d59a4 100644
--- a/arch/mips/kernel/process.c
+++ b/arch/mips/kernel/process.c
@@ -711,8 +711,8 @@ unsigned long mips_stack_top(void)
  */
 unsigned long arch_align_stack(unsigned long sp)
 {
-	if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space)
+	if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space && (sp > 0))
-		sp -= get_random_u32_below(PAGE_SIZE);
+		sp -= get_random_u32_below((PAGE_SIZE<=sp)?PAGE_SIZE:sp+1);
 
 	return sp & ALMASK;
 }
```

namiltd, 2025-01-10:
@ptpt52 A small correction (changed sign to <=, changed sp to (sp+1)):
```diff
diff --git a/arch/mips/kernel/process.c b/arch/mips/kernel/process.c
index b630604..62d59a4 100644
--- a/arch/mips/kernel/process.c
+++ b/arch/mips/kernel/process.c
@@ -711,8 +711,8 @@ unsigned long mips_stack_top(void)
  */
 unsigned long arch_align_stack(unsigned long sp)
 {
-	if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space)
+	if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space && (sp > 0))
-		sp -= get_random_u32_below(PAGE_SIZE);
+		sp -= get_random_u32_below((PAGE_SIZE<=sp)?PAGE_SIZE:(sp+1));
 
 	return sp & ALMASK;
 }
```

DragonBluep, 2025-01-11:
> I don't think that `sp` will ever be 0 or near 0 (less than PAGE_SIZE). 

Yep. `sp` is stack pointer address. It's a very big value. But PAGE_SIZE is only 4096 here.

> The error address in the kernel logs (EPC) is always 0x7*, IIRC the page at offset 0 is always unmapped (to catch NULL pointer accesses), and the stack grows downwards, so if it were close to 0 then there would be virtually no stack space.

MT7620 only support max 256 MiB RAM. There are a lot of reserved address. So sp will never close to 0 nor 4096.

> 
> Might be interesting to print sp before and after arch_align_stack(). Maybe the stack pointer isn't aligned to a page, 

Unalign access is acceptable on mt7620. In fact most architectures use unalign access.

> or if it substracts too much, it runs out of space elsewhere.

Make sense. Linux default stack size is 8 MiB. I'll try if I can catch the stack range.

ptpt52, 2025-01-11:
@namiltd  crash again with your patch.

DragonBluep, 2025-01-11:
I didn't find anything strange.
```
unsigned long arch_align_stack(unsigned long sp)
{
	static unsigned long max;

	if (max < sp)
		max = sp;
	pr_err("%lx\t%lx", sp, max);

	if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space)
		sp -= get_random_u32_below(PAGE_SIZE);

	return sp & ALMASK;
}

[  410.033573] 7f831ea9 7fffe000
[  410.046309] 7fc65000 7fffe000
[  410.049497] CPU 0 Unable to handle kernel paging request at virtual address 7036c2d8, epc == 7036c2d8, ra == 800042f4

[  153.196481] 7fc88ea9 7fffe000
[  153.208524] 7fc79000 7fffe000
[  153.211946] CPU 0 Unable to handle kernel paging request at virtual address 7036c2d8, epc == 7036c2d8, ra == 800042f4

[  151.954537] 7fd5c000 7fffe000
[  151.958120] 7fd5bea9 7fffe000
[  151.961364] CPU 0 Unable to handle kernel paging request at virtual address 7036c2d8, epc == 7036c2d8, ra == 800042f4
```

namiltd, 2025-01-11:
Insert an additional printf before each subsequent line to detect where in the arch_align_stack() function the crash is generated.
Split if (!(current->personality & ADDR_NO_RANDOMIZE) && randomize_va_space) into several lines of code, e.g.:
```
unsigned int test;
test = current->personality;
pr_err("Line 1");
test = test & ADDR_NO_RANDOMIZE;
pr_err("Line 2");
if (!(test) && randomize_va_space) {
    pr_err("Line 3");
    sp -= get_random_u32_below(PAGE_SIZE);
    pr_err("Line 4");
}
```

KanjiMonster, 2025-01-11:
I just noticed the `RA` is always the same, so this is very likely always one of the `get_random_u*()` calls of `get_random_u32_below()`.

So the issue isn't that sp is modified in a wrong way, but rather that the get_random() call somehow breaks sometimes. The question is why it happens.

namiltd, 2025-01-11:
So you need to add the pr_err between the individual get_random_u32_below() lines from include\linux\random.h

KanjiMonster, 2025-01-11:
The more I look at the crashes, the less sense they make. `EPC` is a non-kernel address, so it means the CPU tried to execute code at a user space address.

Especially this one doesn't make sense at all:

```
<4>[  925.652687] Call Trace:
<4>[  925.652691] 
<4>[  925.660662] [<80173c04>] setup_arg_pages+0x48/0x2cc
<4>[  925.670545] [<8014e304>] free_unref_page+0xa0/0x118
<4>[  925.680890] [<80368054>] get_random_u32+0x64/0xdc
<4>[  925.690272] [<80031008>] flush_itimer_signals+0x34/0x5c
<4>[  925.701304] [<80120bf8>] arch_pick_mmap_layout+0x11c/0x1b0
<4>[  925.712545] [<801d2c3c>] load_elf_phdrs+0x78/0xcc
<4>[  925.722683] [<801d3108>] load_elf_binary+0x2e4/0x1540
<4>[  925.732902] [<8016abb0>] __kernel_read+0x164/0x2ac
<4>[  925.743396] [<80174d58>] bprm_execve+0x1ec/0x578
```

What's the get_random_u32() in the stack doing there? I don't think that should be possible. I wonder if there's some stack corruption happening.

[28 more replies]

commit 74238413 (2024-10-16): ramips: disable CONFIG_LIST_HARDENED on MT7620

In #16396, crashes were reported on MT7620, which were introduced by enabling
CONFIG_LIST_HARDENED. The cause seems to be random memory corruption somewhere
in the kernel. Unfortunately the crash traces do not point to the real cause
of the crash in any way.
Since MT7620 is really ancient hardware that likely only has few users left,
I don't expect that anybody will invest a significant amount of time to track
down the real cause. Because of that, let's disable CONFIG_LIST_HARDENED on
this target only, and leave it enabled on all other platforms. [... cut]