openwrt/openwrt #16398: Interface not restarted when wifi STA connection is disabled and re-enabled

ryanrsrs, 2024-09-15:
### Describe the bug

This device is configured as a simple wifi client connecting to my existing home wifi network. STA mode, not AP.

In Luci, if I disable, then re-enable the wifi connection, it will authenticate and associate to the wifi network, but will not pass traffic. I have to manually restart the wwan interface before the connection starts working. I didn't need to do this in 23.05.4.

### OpenWrt version

r27352-47370eb9ba

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

Netgear WNDR3700 v2

### Image kind

Official downloaded image

### Steps to reproduce

Disable the wifi client phy1-sta0.

In Luci > Network > Wireless:
Find entry "SSID: 204_5 | Mode: Client", click Disable.

Wifi connection goes down, as expected.

Click Enable to bring it back up.

### Actual behaviour

Once re-enabled, the wifi connection successfully authenticates and associates with the network, but does not pass packets.

```
root@OpenWrt:~# iwinfo 
phy1-sta0 ESSID: "204_5"
          Access Point: C4:3D:C7:B0:7C:EC
          Mode: Client  Channel: 36 (5.180 GHz)  HT Mode: HT40
          Center Channel 1: 38 2: unknown
          Tx-Power: 17 dBm  Link Quality: 70/70
          Signal: -12 dBm  Noise: -95 dBm
          Bit Rate: 6.0 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11a/n
          Hardware: 168C:0029 168C:A094 [Atheros AR9220]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

root@OpenWrt:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP qlen 1000
    link/ether c6:3d:c7:b0:7c:ec brd ff:ff:ff:ff:ff:ff
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN qlen 1000
    link/ether c4:3d:c7:b0:7c:ed brd ff:ff:ff:ff:ff:ff
4: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether c6:3d:c7:b0:7c:ec brd ff:ff:ff:ff:ff:ff
    inet 192.168.60.1/24 brd 192.168.60.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd35:8d46:eb5c::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::c43d:c7ff:feb0:7cec/64 scope link 
       valid_lft forever preferred_lft forever

8: phy1-sta0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether c4:3d:c7:b0:7c:ec brd ff:ff:ff:ff:ff:ff
    inet6 fe80::c63d:c7ff:feb0:7cec/64 scope link 
       valid_lft forever preferred_lft forever
```

On my AP (different device also running OpenWrt), I can see the WNDR3700v2 re-associate.

To get things working, I have to go to
Luci > Network > Interfaces
Find interface "wwan : phy1-sta0", click Restart.

Then packets flow and everything works.

Now the wifi interface has an ip address:
```
root@OpenWrt:~# iwinfo 
phy1-sta0 ESSID: "204_5"
          Access Point: C4:3D:C7:B0:7C:EC
          Mode: Client  Channel: 36 (5.180 GHz)  HT Mode: HT40
          Center Channel 1: 38 2: unknown
          Tx-Power: 17 dBm  Link Quality: 70/70
          Signal: -32 dBm  Noise: -95 dBm
          Bit Rate: 270.0 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11a/n
          Hardware: 168C:0029 168C:A094 [Atheros AR9220]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

root@OpenWrt:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP qlen 1000
    link/ether c6:3d:c7:b0:7c:ec brd ff:ff:ff:ff:ff:ff
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN qlen 1000
    link/ether c4:3d:c7:b0:7c:ed brd ff:ff:ff:ff:ff:ff
4: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether c6:3d:c7:b0:7c:ec brd ff:ff:ff:ff:ff:ff
    inet 192.168.60.1/24 brd 192.168.60.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd35:8d46:eb5c::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::c43d:c7ff:feb0:7cec/64 scope link 
       valid_lft forever preferred_lft forever
8: phy1-sta0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether c4:3d:c7:b0:7c:ec brd ff:ff:ff:ff:ff:ff
    inet 192.168.5.53/24 brd 192.168.5.255 scope global phy1-sta0
       valid_lft forever preferred_lft forever
    inet6 fe80::c63d:c7ff:feb0:7cec/64 scope link 
       valid_lft forever preferred_lft forever
```

### Expected behaviour

On 23.05.4, I do not need to restart the wwan interface. It happens automatically when the wifi connection is re-enabled.

### Additional info

uci export
https://paste.armbian.com/agohihisek.txt

Annotated logread (scroll past the boot messages)
23.05.4 https://paste.armbian.com/bulomoqofu.txt
SNAPSHOT https://paste.armbian.com/ixuresowop.txt

I can do additional debugging, including building my own images if that's helpful.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vochong, 2024-09-16:
@ryanrsrs

Is your problem similar to this issue I filed a while ago:  https://github.com/openwrt/openwrt/issues/15389

ryanrsrs, 2024-09-16:
No.

ryanrsrs, 2024-09-18:
I see the same issue on a different target, GL-AR750, so I do not think it is target-specific. Probably some ifup/down/hotplug script change.

I am bisecting this.

ryanrsrs, 2024-09-18:
This commit introduced the regression:
https://github.com/openwrt/openwrt/commit/2bec6f48e6f8438f81bf26979d9ea296331d265d

ryanrsrs, 2024-09-19:
I am now doing debugging on a GL-AR750, not the Netgear. I don't think minor target differences matter.

As before, enabling a wifi STA connection doesn't bring the link up fully. You need to also restart the wwan interface. In 23.05.4, you don't need to manually restart the interface.

This netifd commit introduced the regression:
https://github.com/openwrt/netifd/commit/a8e90853c93632996bdba4cd227aa24253a3ef7c

It is still present in the latest netifd.

@nbd168 maybe you can point me in the right direction before I start looking into the logic?

```
root@OpenWrt:~# logread -f

In Luci, click Network > Wireless > Mode: Client [enable].

Thu Sep 19 03:20:28 2024 daemon.notice netifd: radio0 (3538): WARNING: Variable 'data' does not exist or is not an array/object
Thu Sep 19 03:20:29 2024 daemon.notice hostapd: Set new config for phy phy0:
Thu Sep 19 03:20:29 2024 daemon.notice wpa_supplicant[1723]: Set new config for phy phy0
Thu Sep 19 03:20:29 2024 daemon.notice netifd: radio0 (3538): sh: out of range
Thu Sep 19 03:20:29 2024 daemon.notice wpa_supplicant[1723]: Set new config for phy phy0
Thu Sep 19 03:20:29 2024 daemon.notice hostapd: Set new config for phy phy0:
Thu Sep 19 03:20:29 2024 daemon.notice wpa_supplicant[1723]: Set new config for phy phy0
Thu Sep 19 03:20:31 2024 kern.warn kernel: [  223.891427] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
Thu Sep 19 03:20:31 2024 kern.info kernel: [  223.906008] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
Thu Sep 19 03:20:31 2024 kern.info kernel: [  223.912306] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
Thu Sep 19 03:20:31 2024 kern.info kernel: [  223.920910] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 25560 iram: 24780'
Thu Sep 19 03:20:31 2024 kern.warn kernel: [  223.977951] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
Thu Sep 19 03:20:31 2024 kern.info kernel: [  223.997918] ath10k_pci 0000:00:00.0: rts threshold -1
Thu Sep 19 03:20:31 2024 daemon.notice netifd: Interface 'wwan' is enabled
Thu Sep 19 03:20:32 2024 daemon.notice netifd: Wireless device 'radio0' is now up
Thu Sep 19 03:20:32 2024 daemon.notice netifd: Interface 'wwan' is disabled
Thu Sep 19 03:20:32 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Thu Sep 19 03:20:32 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Thu Sep 19 03:20:32 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Thu Sep 19 03:20:34 2024 daemon.warn odhcpd[1901]: No default route present, overriding ra_lifetime to 0!
Thu Sep 19 03:20:36 2024 daemon.notice wpa_supplicant[1723]: phy0-sta0: SME: Trying to authenticate with 24:f2:7f:c4:cf:79 (SSID='204_5' freq=5680 MHz)
Thu Sep 19 03:20:36 2024 kern.info kernel: [  228.739324] phy0-sta0: authenticate with 24:f2:7f:c4:cf:79 (local address=94:83:c4:0e:8f:2e)
Thu Sep 19 03:20:36 2024 kern.info kernel: [  228.748132] phy0-sta0: send auth to 24:f2:7f:c4:cf:79 (try 1/3)
Thu Sep 19 03:20:36 2024 daemon.notice wpa_supplicant[1723]: phy0-sta0: SME: Trying to authenticate with 24:f2:7f:c4:cf:79 (SSID='204_5' freq=5680 MHz)
Thu Sep 19 03:20:36 2024 kern.info kernel: [  228.951178] phy0-sta0: authenticate with 24:f2:7f:c4:cf:79 (local address=94:83:c4:0e:8f:2e)
Thu Sep 19 03:20:36 2024 kern.info kernel: [  228.959969] phy0-sta0: send auth to 24:f2:7f:c4:cf:79 (try 1/3)
Thu Sep 19 03:20:36 2024 kern.info kernel: [  228.970616] phy0-sta0: authenticated
Thu Sep 19 03:20:36 2024 daemon.notice wpa_supplicant[1723]: phy0-sta0: Trying to associate with 24:f2:7f:c4:cf:79 (SSID='204_5' freq=5680 MHz)
Thu Sep 19 03:20:36 2024 kern.info kernel: [  228.990202] phy0-sta0: associate with 24:f2:7f:c4:cf:79 (try 1/3)
Thu Sep 19 03:20:36 2024 kern.info kernel: [  228.998679] phy0-sta0: RX AssocResp from 24:f2:7f:c4:cf:79 (capab=0x111 status=30 aid=5)
Thu Sep 19 03:20:36 2024 kern.info kernel: [  229.007186] phy0-sta0: 24:f2:7f:c4:cf:79 rejected association temporarily; comeback duration 1000 TU (1024 ms)
Thu Sep 19 03:20:38 2024 kern.info kernel: [  230.099730] phy0-sta0: associate with 24:f2:7f:c4:cf:79 (try 2/3)
Thu Sep 19 03:20:38 2024 kern.info kernel: [  230.143148] phy0-sta0: RX AssocResp from 24:f2:7f:c4:cf:79 (capab=0x111 status=0 aid=5)
Thu Sep 19 03:20:38 2024 kern.warn kernel: [  230.153274] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
Thu Sep 19 03:20:38 2024 kern.warn kernel: [  230.160640] ath10k_pci 0000:00:00.0: failed to enable peer stats info: -122
Thu Sep 19 03:20:38 2024 kern.info kernel: [  230.167981] phy0-sta0: associated
Thu Sep 19 03:20:38 2024 daemon.notice wpa_supplicant[1723]: phy0-sta0: Associated with 24:f2:7f:c4:cf:79
Thu Sep 19 03:20:38 2024 daemon.notice wpa_supplicant[1723]: phy0-sta0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Thu Sep 19 03:20:38 2024 daemon.notice wpa_supplicant[1723]: phy0-sta0: Unknown event 37
Thu Sep 19 03:20:38 2024 daemon.notice netifd: Network device 'phy0-sta0' link is up
Thu Sep 19 03:20:38 2024 daemon.notice wpa_supplicant[1723]: phy0-sta0: WPA: Key negotiation completed with 24:f2:7f:c4:cf:79 [PTK=CCMP GTK=CCMP]
Thu Sep 19 03:20:38 2024 daemon.notice wpa_supplicant[1723]: phy0-sta0: CTRL-EVENT-CONNECTED - Connection to 24:f2:7f:c4:cf:79 completed [id=1 id_str=]
Thu Sep 19 03:20:38 2024 daemon.notice wpa_supplicant[1723]: phy0-sta0: Unknown event 37
[... cut ...]
	option device 'radio0'
	option network 'wwan'
	option mode 'sta'
	option ssid '204_5'
	option encryption 'sae-mixed'
	option key 'xxxxxxxx'
	option ocv '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/ahb/18100000.wmac'
	option band '2g'
	option channel '1'
	option htmode 'HT20'
	option disabled '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'

root@OpenWrt:~# 
```

ryanrsrs, 2024-09-19:
Related:
https://github.com/openwrt/netifd/issues/33

nbd168, 2024-09-21:
Fixed in 3727731e6ae972491c6d3c41a1317adc1e4c68bd

closed 2024-09-21 by nbd168