openwrt/openwrt #16421: Xiaomi Mi Router AX3000T Snapshot - Luci doesn't show any connected stations

tabascoz, 2024-09-18:
### Describe the bug

Hi, in the latest git from today 09182024, Luci does not show any stations connected. I get a message "Collecting data..." but nothing ever appears. 

### OpenWrt version

r27433-7e1d092552

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T (OpenWrt U-Boot layout)

### Image kind

Official downloaded image

### Steps to reproduce

Install snapshot. Install luci. Reboot. Open Luci web-page. Navigate to Network -> wireless. 

### Actual behaviour

a message shows "Collecting data..." in associated stations section

### Expected behaviour

all connected stations and it's details should appear in associated stations section. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aamadorj, 2024-09-18:
I'm seeing the same issue with the same snapshot version but with a different device: GL.iNet GL-MT6000. I can add that no associated stations are shown at Status -> Overview, neither.

aamadorj, 2024-09-20:
With Snapshot version r27466-f368e2d5ec, downloaded from [here](https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/) today, the issue is gone. Luci is showing all the wireless information it should from my GL.iNet GL-MT6000.

tabascoz, 2024-09-20:
yep i can confirm. Thanks Its closed :)

closed 2024-09-20 by tabascoz