openwrt/openwrt #16423: Broadcom BCM47xx/53xx: Remove "brcmfmac-firmware-usb" from the core kernel

jen1-sys, 2024-09-18:
The name of the package "brcmfmac-firmware-usb" is misleading. This is a firmware for USB wifi adapter with these 2 chips "brcm43143" and "brcm43236b". The package will add 2 files to /lib/firmware/brcm: brcmfmac43143.bin and brcmfmac43236b.bin

The group "Broadcom BCM47xx/53xx" has PCIE WiFi chips and doesn't require this package. In fact no router at all requires this package, as it a driver for external USB WiFi adapter for the mentioned chips a=only and should be installed only upon request.

I successfully removed this package for my router Asus RT-AC88U by unchecking "Enable USB bus interface" under "Kernel modules ---> Wireless drivers --->". See pic 01
[image]

This will automatically uncheck the named package "brcmfmac-firmware-usb" under "Firmware --->". See pic 02
[image]

This will comment these lines in the .config file

```
# CONFIG_PACKAGE_brcmfmac-firmware-usb is not set
# CONFIG_BRCMFMAC_USB is not set
```

I tested my router Asus RT-AC88U with both 23.05 and the main branches and it worked flawlessly with no issues, and reduced the image size from 9,306,112 bytes to 8,650,752 bytes. That is a reduce size 655,360 bytes (640 KB) was used for this unnecessary package.

Here are some proves for the tests.
Build 23.05: link for the build https://www.mediafire.com/file/hu85ta0xwpmzslu/OpenWrt+23.05+bin.xz/file

[image]
[image]
[image]

Build snapshot (I used the main branch to build): Link for the build and image builder https://www.mediafire.com/file/h2bimdoqlkt92xr/24.tar.xz/file

[image]
[image]

If there are routers with these WiFi chips "brcm43143" and "brcm43236b" and I doubt it, then this package should be added to these routers separately and not to the core kernel.

Thanks in advance

jen1-sys, 2024-09-19:
BTW, this can be done easily by editing .config-package.in. Changes default from y to n for the .config-package.in line 23269

Before: lines 23266 to 23273

```
		config BRCMFMAC_USB
			bool "Enable USB bus interface support"
			depends on USB_SUPPORT
			default y
			help
			  Supported USB connected chipsets:
			  BCM43235, BCM43236, BCM43238 (all in revision 3 only)
			  BCM43143, BCM43242, BCM43566, BCM43569
```
After: 

```
		config BRCMFMAC_USB
			bool "Enable USB bus interface support"
			depends on USB_SUPPORT
			default n
			help
			  Supported USB connected chipsets:
			  BCM43235, BCM43236, BCM43238 (all in revision 3 only)
			  BCM43143, BCM43242, BCM43566, BCM43569
```

As said, if needed they can be install later, this should not be the default.

brada4, 2024-09-19:
Could you, please, follow the issue template? Without planning implementation details, just saying what you see and that you dont want to see it. Thanks