openwrt/openwrt #16428: I found a bug with the router TP-Link Archer C20 v5 (ISP)

testerbots723492340, 2024-09-20:
### Describe the bug

Hello everyone, I wanted to install openwrt my router TP-Link Archer C20 v5 (ISP), after installing via tftp server I got an eternal reboot. The documentation stated the version of my router, but did not specify the version for the provider (for ISP). I have restored the factory firmware, the router is working now. But I want to install openwrt, please help. I tried to install European and American versions of the firmware, but nothing works, still openwrt does not work.

Official page my router: https://www.tp-link.com/ru/support/download/archer-c20(isp)/

### OpenWrt version

https://openwrt.org/toh/tp-link/archer_c20_v5

### OpenWrt release

https://downloads.openwrt.org/releases/23.05.4/targets/ramips/mt76x8/openwrt-23.05.4-ramips-mt76x8-tplink_archer-c20-v5-initramfs-kernel.bin

### OpenWrt target/subtarget

https://openwrt.org/toh/tp-link/archer_c20_v5

### Device

https://www.tp-link.com/ru/support/download/archer-c20(isp)/

### Image kind

Official downloaded image

### Steps to reproduce

I installed openvrt firmware on my router TP-Link Archer C20 v5 (ISP)

### Actual behaviour

I got infinite reboot

### Expected behaviour

I wanted the firmware to work

### Additional info

I think if the router is supported, then why doesn't my router work, the same version of the router 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-09-20:
Helps if you do not cross-post.
First try to figure out in forums whether a code change (here) or a documentation change (in forums) fixes your router confusion.

https://forum.openwrt.org/t/problem-with-installation-tp-link-archer-c20-v5-isp/210879

brada4, 2024-09-22:
Would be nice if you close a non-issue and add info to use V4 OpenWRT image for V5 ISP device.

vetash, 2025-09-07:
Can confirm that, V5 (ISP) is indeed V4 and must to flash with owrt compiled for C20 v4.

ISP device basically marked on sticker like TP-Link Archer C20 (ISP)

brada4, 2025-09-07:
RU links are expired? Is it visually different from EU version eg https://www.tp-link.com/nl/home-networking/wifi-router/archer-c20/v5/