openwrt/openwrt #16433: Wifi not working correctly on latest 23.05.04 (big drop in speed)

happeeshopper, 2024-09-20:
### Describe the bug

Device: GL.iNet MT3000

Runing a basic test via laptop on speedtest.net with stock GL.iNet firmware I could repeatedly get 185mb/s

Using the same devices a few minutes later after flashing OpenWrt 23.05.4 the maximum I could get was 107mb/s.

I tried all settings but no improvement could be made.

I made a [thread](https://forum.openwrt.org/t/openwrt-50-drop-in-speed-compared-to-stock-gl-inet-fw-mt3000/210552) in the OpenWrt forum and a user [suggested](https://forum.openwrt.org/t/openwrt-50-drop-in-speed-compared-to-stock-gl-inet-fw-mt3000/210552/9?u=routerama) downgrading to 23.05.4, which fixed the problem and in fact OWrt firmware now outperforms the stock GL.i slightly and achieves over 190mb/s in the same test as before:

[image]

It seems there's something wrong with the wifi on 23.05.4.

I can't provide any further info now because I've had to flash the previous version to fix the problem.

### OpenWrt version

23.05.4

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

23.05.4

### Device

GL.iNet GL-MT3000

### Image kind

Official downloaded image

### Steps to reproduce

Use WiFi on the latest 23.05.4 firmware

### Actual behaviour

Big drop in speed

### Expected behaviour

No drop in speed

### Additional info

23.05.3 is unaffected

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

happeeshopper, 2024-09-21:
I'm new to the scene, this is the first time I've setup a router so I'm not aware of anything that's gone on previously.

All I know is the latest 23.05.4 firmware has a bug that seriously cripples wifi and the only way to stop it is to use 23.05.3. This may or may not be the same issue as before, I have no idea. The only thing I can tell you for certain is that there's a seriously debilitating bug in in the latest firmware.

There are no instructions or notes anywhere that warn people about this problem and neither are there any recommendations to flash a snapshot instead of the official firmware.

I don't know why you''ve labelled this as an invalid bug because it's most definitely valid and serious.

I've been trying to get around it for days and finally solved it with the help of a forum member.

I've done what any user would do... downloaded and flashed the latest firmware... and there's a wifi crippling bug in it.

It doesn't matter if it's been addressed before or not, it's either back, or it's a new bug.

happeeshopper, 2024-09-21:
That's cool, I don't know how anyones supposed to know that though, took me days of fiddling find out what was wrong. 

Surely this issue could be highlighted on the release page so users know about it.

happeeshopper, 2024-09-21:
> fix is here [56a84b3](https://github.com/openwrt/openwrt/commit/56a84b3c4754f96a64a17bcc9730358ac48ba993) it's just there is no 23.05.5 release

That's cool, I don't know how anyones supposed to know that though, took me days of fiddling find out what was wrong. 

Surely this issue could be highlighted on the release page so users know about it.

brada4, 2024-09-22:
You marked it solved in the forum, there is no bug thus.
https://forum.openwrt.org/t/openwrt-50-drop-in-speed-compared-to-stock-gl-inet-fw-mt3000/210552

happeeshopper, 2024-09-22:
> You marked it solved in the forum, there is no bug thus. https://forum.openwrt.org/t/openwrt-50-drop-in-speed-compared-to-stock-gl-inet-fw-mt3000/210552

It was only solved because I flashed 23.05.3, the bug is in 23.05.4.

happeeshopper, 2024-09-22:
> That’s fair but it has been discussed at length on the forum. If you like, test the next stable release with the fix: https://forum.openwrt.org/t/request-for-testing-openwrt-23-05-snapshots-23-05-5/210592
> 
> Otherwise just wait .5 should be out fairly soon it seems.

Cool I'll wait for the update.

brada4, 2024-09-22:
No action can be taken to fix 23.05.4 , you can close the report.

happeeshopper, 2024-09-23:
> No action can be taken to fix 23.05.4 , you can close the report.

No problem, but it's really bad form to leave a known bugged out release up as the latest stable release for months on end, without any indication to end users.

closed 2024-09-23 by happeeshopper