openwrt/openwrt #16438: realtek: RTL838x switch does not show correct speed of SFP modules

GoetzGoerisch, 2024-09-21:
### Describe the bug

SFP or SFP+ modules in the switch are reporting the incorrect link speed.

The operate at 1000Mbps but are shown and reported as 10Mbps.

Was already like this with 5.15 but remains with 6.6

### OpenWrt version

r27466-f368e2d5ec

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

realtek/rtl838x

### Device

HPE 1920-24G (JG924A

### Image kind

Official downloaded image

### Steps to reproduce

- Plug in SFP to RJ45 or SFP+-DAC
- Review reported link speed in UI

### Actual behaviour

Link speed is reported as 10Mbps
`rtl83xx-switch switch@1b000000 lan25: Link is Up - 10Mbps/Half - flow control off`

Actual throughput is 1000Mbps

### Expected behaviour

Correct link speed is reported

### Additional info

Modules as reported by kernel-log

```
sfp sfp-0: module OEM              SFP-GE-T         rev      sn <REDACTED>      dc 23110101
rtl83xx-switch switch@1b000000 lan25: Link is Up - 10Mbps/Half - flow control off
```
Model: https://www.amazon.de/gp/product/B07BQWFBC8
```
sfp sfp-3: module OEM              SFP-H10GB-CU1M   rev R    sn <REDACTED>  dc 240217
rtl83xx-switch switch@1b000000 lan28: Link is Up - 10Mbps/Full - flow control off
```
Model: https://www.amazon.de/gp/product/B01DCZCQTC

```
ethtool lan28
Settings for lan28:
	Supported ports: [ TP MII FIBRE ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Half 1000baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Half 1000baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 10Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 27
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Supports Wake-on: d
	Wake-on: d
	Link detected: yes

```

Previous discussion on the forum: https://forum.openwrt.org/t/support-for-rtl838x-based-managed-switches/57875/2804

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2024-10-26:
Linked to:
- https://github.com/openwrt/openwrt/pull/16457

evs93, 2024-10-26:
> No real idea, but can you have a look at rtl83xx_phylink_mac_link_up() and print out the debugging info there.

To get this moving along, and just to sanity check:

Build latest code.
 kernel/module loglevel set correctly?
You want the pr_debug statement from rtl83xx_phylink_mac_link_up

https://github.com/openwrt/openwrt/blob/a2aabc9a7a4ec8a66477fbdde84a73b5942b460e/target/linux/realtek/files-6.6/drivers/net/dsa/rtl83xx/dsa.c#L949

Neustradamus, 2024-10-27:
@GoetzGoerisch, @evs93: The @plappermaul PR has been merged, can you test?
- https://github.com/openwrt/openwrt/pull/16457

evs93, 2024-10-27:
> @GoetzGoerisch, @evs93: The @plappermaul PR has been merged, can you test?
> 
>     * [realtek: phy/ethernet fixes #16457](https://github.com/openwrt/openwrt/pull/16457)

I'll test again at some point.

To clarify:  my initial testing indicated behaviour didn't change before/after.

It was my initial confusion as I didn't know whether it was existing behaviour, and then I edited to indicate that it was existing behaviour.

plappermaul, 2025-04-25:
Can you add those debug messages and post the link up output again:

```
static void rtl83xx_phylink_mac_link_up(struct dsa_switch *ds, int port,
				   unsigned int mode,
				   phy_interface_t interface,
				   struct phy_device *phydev,
				   int speed, int duplex,
				   bool tx_pause, bool rx_pause)
{
	struct dsa_port *dp = dsa_to_port(ds, port);
	struct rtl838x_switch_priv *priv = ds->priv;
	u32 mcr, spdsel;

        pr_info("%s port %d speed = %d\n", __func__, port, speed); <<<< ADD
	if (speed == SPEED_1000)
```

```
static void rtl83xx_pcs_get_state(struct phylink_pcs *pcs,
				  struct phylink_link_state *state)
{
	struct rtl838x_pcs *rtpcs = container_of(pcs, struct rtl838x_pcs, pcs);
	struct rtl838x_switch_priv *priv = rtpcs->priv;
	int port = rtpcs->port;
	u64 speed;
	u64 link;

	if (port < 0 || port > priv->cpu_port) {
		state->link = false;
		return;
	}

	state->link = 0;
	link = priv->r->get_port_reg_le(priv->r->mac_link_sts);
	if (link & BIT_ULL(port))
		state->link = 1;
	pr_debug("%s: link state port %d: %llx\n", __func__, port, link & BIT_ULL(port));

	state->duplex = 0;
	if (priv->r->get_port_reg_le(priv->r->mac_link_dup_sts) & BIT_ULL(port))
		state->duplex = 1;

	speed = priv->r->get_port_reg_le(priv->r->mac_link_spd_sts(port));

	pr_info("%s: got combined speed %08x for port %d\n", __func__, (u32)speed, port); <<<< ADD

	speed >>= (port % 16) << 1;
```

On a RTL8380 with direct SFP (not RTL8214FC via QSGMII and FIBER/COPPER switch) I get:

```
[  138.225740] rtl83xx_phylink_mac_link_up port 24 speed = 1000
[  138.232291] rtl83xx-switch switch@1b000000 lan9: Link is Up - 1Gbps/Full - flow control off
[  138.242829] rtl83xx_pcs_get_state: got combined speed 00220000 for port 24
[  138.250761] switch: port 9(lan9) entered blocking state
[  138.256660] switch: port 9(lan9) entered forwarding state
```

dekarl, 2025-04-26:
Switched to a HPE 1920-24G (identical besides 8 ports more according to the wiki) with an US-8-150W as Link Partner.

Running OpenWrt SNAPSHOT, r29342-6b1d0ce88c with the two lines added.

```
[ 1855.197824] sfp sfp-3: module UBNT             UF-MM-1G         rev      sn FT17110800935    dc 171108
[ 1855.228883] rtl8214fc_media_set: port 27, set_fibre: 1
[ 1855.386713] hwmon hwmon1: temp1_input not attached to any thermal zone
[ 1855.408717] rtl8214fc_power_set: Powering off COPPER (port 27)
[ 1855.480857] rtl8214fc_power_set: Powering on FIBRE (port 27)
[ 1913.804894] rtl83xx_phylink_mac_link_up port 27 speed = 10
[ 1913.823280] rtl83xx-switch switch@1b000000 lan28: Link is Up - 10Mbps/Half - flow control off
[ 1913.851969] switch: port 28(lan28) entered blocking state
[ 1913.870063] switch: port 28(lan28) entered forwarding state
```

I'm missing a line with rtl83xx_pcs_get_state in it. Grep does not find any in dmesg either.

plappermaul, 2025-04-26:
Interesting ... Currently i have no switch with RTL8214FC at hand ... So this is a quick assumption without deep investigation. Speed determination of RTL8214FC does not work out of the box because it must decide in which register to look (depending on active operation mode). So we might need a dedicated read_status() routine. Maybe something like this, that simply sets the speed to 1000 full duplex at the moment.

```
static int rtl8214fc_read_status(struct phy_device *phydev) <<< ADD
{
        /* Fixed for now, need to code that out properly ... */
	phydev->speed = SPEED_1000;
	phydev->duplex = DUPLEX_FULL;

	return 0;
}
...
{
		.match_phy_device = rtl8214fc_match_phy_device,
		.name		= "Realtek RTL8214FC",
		.features	= PHY_GBIT_FIBRE_FEATURES,
		.probe		= rtl8214fc_phy_probe,
		.read_page	= rtl821x_read_page,
		.write_page	= rtl821x_write_page,
                .read_status    = rtl8214fc_read_status, <<< ADD
		.suspend	= rtl8214fc_suspend,
		.resume		= rtl8214fc_resume,
		.set_loopback	= genphy_loopback,
		.set_port	= rtl8214fc_set_port,
		.get_port	= rtl8214fc_get_port,
		.set_eee	= rtl8214fc_set_eee,
		.get_eee	= rtl8214fc_get_eee,
},

```

Give it a try and check if speed is reported from the fixed values.

dekarl, 2025-04-26:
It is different now.

No more messages about link up at all, ethtool reports 1000Mb/s Full Duplex, but not link detected.

On the other end a Half Duplex 1000Mb/s link is detected.

So it did not only change the reported values, but also the actual behaviour .

plappermaul, 2025-04-26:
Thanks for your help. So we know where it comes from. The normal call sequence from phy.h is

```
static inline int phy_read_status(struct phy_device *phydev)
{
	if (!phydev->drv)
		return -EIO;

	if (phydev->drv->read_status)
		return phydev->drv->read_status(phydev);
	else
		return genphy_read_status(phydev);
}
```

So must head into some direction like this:

static int rtl8214fc_read_status(struct phy_device *phydev)
{
        int ret;

        ret = genphy_read_status(phydev);
        if (ret)
              return ret;

        if (rtl8214fc_media_is_fibre(phydev)) {
               /* detect speed here somehow ... */
           	phydev->speed = SPEED_1000;
        } else {
               /* do nothing assuming that copper detection works totally fine */
        }

	return 0;
}

plappermaul, 2025-04-27:
Here the code from the GPL snippets for documentation purpose. It seems to boil down to:

- Set special register 29 (0x1d) on page 0 to either 3 or 1
- Read internal control register 16 (0x10) from either page 0 or 2624 (0xa40)
- Restore special register 29

```
/* Function Name:
 *      phy_8214fc_speed_get
 * Description:
 *      Get 8214FC MP (mass production) link speed status of the specific port
 * Input:
 *      unit - unit id
 *      port - port id
 * Output:
 *      pSpeed - pointer to PHY link speed
 * Return:
 *      RT_ERR_OK     - OK
 *      RT_ERR_FAILED - invalid parameter
 * Note:
 *      None
 */
int32
phy_8214fc_speed_get(uint32 unit, rtk_port_t port, rtk_port_speed_t *pSpeed)
{
    rtk_port_media_t    media;
    uint32              link_sts;
    uint32              phyData0;
    uint32              page, reg;
    int32               ret;

    if ((ret = _phy_8218b_14fc_intMedia_get(unit, port, &media, &link_sts)) != RT_ERR_OK)
        return ret;

    reg = PHY_INT_REG_BASE + PHY_CONTROL_REG; /* = 16 */

    if (PORT_MEDIA_FIBER == media)
    {
        page = PHY_PAGE_0;
        if ((ret = hal_miim_write(unit, port, PHY_PAGE_0, 29, 0x0003)) != RT_ERR_OK)
            goto ERR;
    }
    else
    {
        page = 0xa40;
        if ((ret = hal_miim_write(unit, port, PHY_PAGE_0, 29, 0x0001)) != RT_ERR_OK)
            goto ERR;
    }

    if ((ret = hal_miim_read(unit, port, page, reg, &phyData0)) != RT_ERR_OK)
        goto ERR;

    *pSpeed = ((phyData0 & SpeedSelection1_MASK) >> (SpeedSelection1_OFFSET -1))
            | ((phyData0 & SpeedSelection0_MASK) >> SpeedSelection0_OFFSET);

ERR:
    hal_miim_write(unit, port, PHY_PAGE_0, 29, 0x0000);
    /* PHY_PAGE_SELECTION_REG = 31 (0x1f) */
    hal_miim_write(unit, port, PHY_PAGE_0, PHY_PAGE_SELECTION_REG, 0x0000);
    return ret;
} /* end of phy_8214fc_speed_get */
```

plappermaul, 2025-04-29:
The only way for me to provide a fix is to test it on my Zyxel GS1920 which is unfortunately a RTL839x. We know that it has not yet proper RTL8214FC. To proceed I ask you to test and approve https://github.com/openwrt/openwrt/pull/18635 This will be needed as a first step to proceed.

plappermaul, 2025-05-01:
I'm confused. Playing and Testing around I found Copper does not work with the RTL8214FC. At least it does not get a link up. Is this expected?

plappermaul, 2025-05-05:
Can someone give this a try in the phy driver.

Insert this:

```
static int rtl8214fc_read_status(struct phy_device *phydev)
{
	bool changed;
	int ret;

	if (rtl8214fc_media_is_fibre(phydev)) {
		phydev->port = PORT_FIBRE;
		ret = genphy_c37_read_status(phydev, &changed);
	} else {
		phydev->port = PORT_MII;
		ret = genphy_read_status(phydev);
	}
	
	return ret;
}

static int rtl8214fc_config_aneg(struct phy_device *phydev)
{
	int ret;

	if (rtl8214fc_media_is_fibre(phydev))
		ret = genphy_c37_config_aneg(phydev);
	else
		ret = genphy_config_aneg(phydev);
	
	return ret;
}

static int rtl8214fc_get_features(struct phy_device *phydev)
{
	int ret = 0;
	
	ret = genphy_read_abilities(phydev);
	if (ret)
		return ret;

	linkmode_set_bit(ETHTOOL_LINK_MODE_1000baseX_Full_BIT, phydev->supported);	

	return 0;
}
```

And modify this

```
	{
		.match_phy_device = rtl8214fc_match_phy_device,
		.name		= "Realtek RTL8214FC",
/*		.features	= PHY_GBIT_FIBRE_FEATURES,  */ <--- REMOVE
		.probe		= rtl8214fc_phy_probe,
		.read_page	= rtl821x_read_page,
		.write_page	= rtl821x_write_page,
		.suspend	= rtl8214fc_suspend,
		.resume		= rtl8214fc_resume,
		.set_loopback	= genphy_loopback,
		.read_status    = rtl8214fc_read_status, <--- ADD
		.config_aneg	= rtl8214fc_config_aneg, <--- ADD
		.get_features	= rtl8214fc_get_features, <--- ADD
		.set_port	= rtl8214fc_set_port,
		.get_port	= rtl8214fc_get_port,
		.set_eee	= rtl8214fc_set_eee,
		.get_eee	= rtl8214fc_get_eee,
	},

```

GoetzGoerisch, 2025-05-06:
Do you have a branch for easy use?

dekarl, 2025-05-06:
```diff
diff --git a/target/linux/realtek/files-6.6/drivers/net/phy/rtl83xx-phy.c b/target/linux/realtek/files-6.6/drivers/net/phy/rtl83xx-phy.c
index 94ccbe5769..10a3a2ac60 100644
--- a/target/linux/realtek/files-6.6/drivers/net/phy/rtl83xx-phy.c
+++ b/target/linux/realtek/files-6.6/drivers/net/phy/rtl83xx-phy.c
@@ -3899,6 +3899,47 @@ static int rtl9300_serdes_probe(struct phy_device *phydev)
        return 0;
 }

+static int rtl8214fc_read_status(struct phy_device *phydev)
+{
+       bool changed;
+       int ret;
+
+       if (rtl8214fc_media_is_fibre(phydev)) {
+               phydev->port = PORT_FIBRE;
+               ret = genphy_c37_read_status(phydev, &changed);
+       } else {
+               phydev->port = PORT_MII;
+               ret = genphy_read_status(phydev);
+       }
+
+       return ret;
+}
+
+static int rtl8214fc_config_aneg(struct phy_device *phydev)
+{
+       int ret;
+
+       if (rtl8214fc_media_is_fibre(phydev))
+               ret = genphy_c37_config_aneg(phydev);
+       else
+               ret = genphy_config_aneg(phydev);
+
+       return ret;
+}
+
+static int rtl8214fc_get_features(struct phy_device *phydev)
+{
+       int ret = 0;
+
+       ret = genphy_read_abilities(phydev);
+       if (ret)
+               return ret;
+
+       linkmode_set_bit(ETHTOOL_LINK_MODE_1000baseX_Full_BIT, phydev->supported);
+
+       return 0;
+}
+
 static struct phy_driver rtl83xx_phy_driver[] = {
        {
                PHY_ID_MATCH_EXACT(PHY_ID_RTL8214C),
@@ -3914,13 +3955,15 @@ static struct phy_driver rtl83xx_phy_driver[] = {
        {
                .match_phy_device = rtl8214fc_match_phy_device,
                .name           = "Realtek RTL8214FC",
-               .features       = PHY_GBIT_FIBRE_FEATURES,
                .probe          = rtl8214fc_phy_probe,
                .read_page      = rtl821x_read_page,
                .write_page     = rtl821x_write_page,
                .suspend        = rtl8214fc_suspend,
                .resume         = rtl8214fc_resume,
                .set_loopback   = genphy_loopback,
+               .read_status    = rtl8214fc_read_status,
+               .config_aneg    = rtl8214fc_config_aneg,
+               .get_features   = rtl8214fc_get_features,
                .set_port       = rtl8214fc_set_port,
                .get_port       = rtl8214fc_get_port,
                .set_eee        = rtl8214fc_set_eee,

```

I tested with this patch.

Works on a HPE 1920-24G. Now it shows the link as 1Gbps/Full instead of 10Mbps/Half
```
[  100.932252] rtl83xx-switch switch@1b000000 lan28: Link is Up - 1Gbps/Full - flow control off
```

ethtool also shows it as advertised by the link partner
```
Settings for lan28:
	Supported ports: [ TP MII ]
	Supported link modes:   100baseT/Half 100baseT/Full
	                        1000baseX/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  100baseT/Half 100baseT/Full
	                        1000baseX/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  1000baseX/Full
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Port: FIBRE
	PHYAD: 27
	Transceiver: external
	Auto-negotiation: on
	Supports Wake-on: d
	Wake-on: d
	Link detected: yes

```

plappermaul, 2025-05-06:
https://github.com/openwrt/openwrt/pull/18724

GoetzGoerisch, 2025-05-10:
Fixed on `main` by https://github.com/openwrt/openwrt/commit/4cfd1c450181893a8063588eafd4eadef6a7e235 #18724
Fix pending for 24.10 https://github.com/openwrt/openwrt/pull/18755

Thank you @plappermaul!

plappermaul, 2025-06-05:
Please close

closed 2025-06-06 by GoetzGoerisch