openwrt/openwrt #16439: realtek 6.6: Upgrade to 6.6 breaks GS1900-8 storage

GoetzGoerisch, 2024-09-21:
### Describe the bug

Upgrading from 23.05.4 with sysupgrade breaks the GS1900-8 configuration and function.
- Storage is not setup correctly from what I can tell.
- Is does not keep its setting.

CC:
@plappermaul 
@svanheule

### OpenWrt version

r27466-f368e2d5ec

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

realtek/rtl838x

### Device

Zyxel GS1900-8v1/v2 Switch

### Image kind

Official downloaded image

### Steps to reproduce

- Download a latest SNAPSHOT build with `luci-ssl owut nano`
- Flash with UI - Sysupgrade while keeping settings

Actual image: 
https://sysupgrade.openwrt.org/store/f4b2a4affc7a3bdd9bb096ea22b8dd4f/openwrt-cd5af01b50ac-realtek-rtl838x-zyxel_gs1900-8-squashfs-sysupgrade.bin

### Additional info

From kernel-log:

5.15
```
2 uimage-fw partitions found on MTD device firmware
Creating 2 MTD partitions on "firmware":
0x000000000000-0x000000320000 : "kernel"
0x000000320000-0x0000006d0000 : "rootfs"
mtd: setting mtd7 (rootfs) as root device
1 squashfs-split partitions found on MTD device rootfs
0x0000005d0000-0x0000006d0000 : "rootfs_data"
0x000000930000-0x000001000000 : "runtime2"
Probing RTL838X eth device pdev: 82095c00, dev: 82095c10
...
Freeing unused kernel image (initmem) memory: 1260K
...
jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
jffs2_build_filesystem(): unlocking the mtd device... 
done.
jffs2_build_filesystem(): erasing all blocks after the end marker... 
...
done.
jffs2: notice: (1849) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
overlayfs: upper fs does not support tmpfile.
```

6.6 differences to 5.15
```
2 uimage-fw partitions found on MTD device firmware
Creating 2 MTD partitions on "firmware":
[... 570 lines cut ...]
[  101.147300] rtl83xx-switch switch@1b000000 lan7: entered allmulticast mode
[  101.156012] rtl83xx-switch switch@1b000000 lan7: entered promiscuous mode
[  101.361843] RTL8380 Link change: status: 1, ports 8000
[  101.433541] rtl83xx-switch switch@1b000000 lan4: Link is Up - 10Mbps/Full - flow control rx/tx
[  101.502775] switch: port 4(lan4) entered blocking state
[  101.508682] switch: port 4(lan4) entered forwarding state
[  101.516661] rtl83xx-switch switch@1b000000 lan8: configuring for phy/internal link mode
[  101.543005] 8021q: adding VLAN 0 to HW filter on device lan8
[  101.655137] switch: port 8(lan8) entered blocking state
[  101.661042] switch: port 8(lan8) entered disabled state
[  101.667150] rtl83xx-switch switch@1b000000 lan8: entered allmulticast mode
[  101.675884] rtl83xx-switch switch@1b000000 lan8: entered promiscuous mode
[  101.736681] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[  101.852712] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[  101.860205] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[  104.912002] RTL8380 Link change: status: 1, ports 8000
[  105.588719] Too few erase blocks (2)
[  105.861069] rtl83xx-switch switch@1b000000 lan8: Link is Up - 1Gbps/Full - flow control off
[  105.870685] switch: port 8(lan8) entered blocking state
[  105.876713] switch: port 8(lan8) entered forwarding state
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-09-21:
This looks like there is no free space

howels, 2024-09-21:
Too few erase blocks means no space for the overlay, remove packages from the sysypgrade image

closed 2024-09-21 by GoetzGoerisch

svanheule, 2024-09-21:
See https://github.com/openwrt/openwrt/pull/16442 for a patch that merges the firmware partitions

svanheule, 2024-12-22:
The two firmware partitions have been merged, so now OpenWrt will have more space. To install (or upgrade to) this new layout, the user will still need to be able to provide a factory (initramfs) image that fits inside the original partition.

Since this is a restriction of the vendor FW, I don't think there is an easy way around this. If/when the default OpenWrt release/snapshot builds no longer manage to fit inside the original partition, the user will have to start from an old build and then use sysupgrade to flash a newer version.

Neustradamus, 2024-12-22:
Good news, https://github.com/openwrt/openwrt/pull/16442 has been merged!

Thanks to all :)