openwrt/openwrt #16444: WiFi 5G Broadcom BCM43460 not work

dima-s30, 2024-09-21:
### Describe the bug

On OpenWRT firmware wifi module on chip BCM43460 did't work. Wia Web Device not ready.

In boot log can see not supported device and error  -95 

```
[   10.049039] b43-phy0: Broadcom 43431 WLAN found (core revision 29)
[   10.074097] b43-phy0: Found PHY: Analog 9, Type 7 (HT), Revision 1
[   10.080344] b43-phy0: Found Radio: Manuf 0x17F, ID 0x2059, Revision 0, Version 1
[   10.104570] b43-phy1: Broadcom 43460 WLAN found (core revision 42)
[   10.111524] b43-phy1 ERROR: FOUND UNSUPPORTED PHY (Analog 12, Type 11 (AC), Revision 1)
[   10.119681] b43: probe of bcma2:1 failed with error -95
[   10.125122] Broadcom 43xx driver loaded [ Features: NL ]
```

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

bcm53xx/generic

### Device

D-Link DWL-8610AP

### Image kind

Self-built image

### Diffconfig

```text
CONFIG_TARGET_bcm53xx=y
CONFIG_TARGET_bcm53xx_generic=y
CONFIG_TARGET_bcm53xx_generic_DEVICE_dlink_dwl-8610ap=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_LUCI_LANG_ru=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_kmod-crypto-arc4=y
CONFIG_PACKAGE_kmod-crypto-ecb=y
CONFIG_PACKAGE_kmod-crypto-sha1=y
CONFIG_PACKAGE_kmod-crypto-user=y
CONFIG_PACKAGE_kmod-gre=y
CONFIG_PACKAGE_kmod-iptunnel=y
CONFIG_PACKAGE_kmod-l2tp=y
CONFIG_PACKAGE_kmod-mppe=y
CONFIG_PACKAGE_kmod-pppol2tp=y
CONFIG_PACKAGE_kmod-pptp=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-i18n-base-ru=y
CONFIG_PACKAGE_luci-i18n-firewall-ru=y
CONFIG_PACKAGE_luci-i18n-opkg-ru=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_ppp-mod-pppol2tp=y
CONFIG_PACKAGE_ppp-mod-pptp=y
CONFIG_PACKAGE_pptpd=y
CONFIG_PACKAGE_resolveip=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireless-tools=m
CONFIG_PACKAGE_xl2tpd=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2024-10-27:
What was the last working version?

dima-s30, 2024-10-30:
> What was the last working version?

Unfortunately, there are no working versions for this device.