openwrt/openwrt #16447: 华硕小旋风 TUF-AX4200 2.G LAN口不显示

xiaoxiudog, 2024-09-22:
### Describe the bug

华硕小旋风 TUF-AX4200 国行版本有两个2.5G接口，其中一个2.G LAN口不显示，无法使用

### OpenWrt version

*

### OpenWrt release

*

### OpenWrt target/subtarget

*

### Device

*

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-09-22:
Other 2.5Gbps serdes is connected to switch asic,  so your whole report is fake. You will not get warranty return for device missing absent feature.
 You are supposed to fill completetemplate along  bug reports IN ENGLISH.

brada4, 2024-10-27:
I think I understood what you want - LAN 1+2 can be configured in a 2Gbps LACP aggregate using luci-app-bonding (or any other pair of ports)

xiaoxiudog, 2024-10-27:
> ### Describe the bug 描述 bug
> 华硕小旋风 TUF-AX4200 国行版本有两个2.5G接口，其中一个2.G LAN口不显示，无法使用
> 
> ### OpenWrt version OpenWrt 版本
> 
> ### OpenWrt release OpenWrt 版本
> 
> ### OpenWrt target/subtarget OpenWrt 目标/子目标
> 
> ### Device 装置
> 
> ### Image kind 图像种类
> Official downloaded image官方下载的镜像
> 
> ### Steps to reproduce 重现步骤
> _No response 无响应_
> 
> ### Actual behaviour 实际行为
> _No response 无响应_
> 
> ### Expected behaviour 预期行为
> _No response 无响应_
> 
> ### Additional info 其他信息
> _No response 无响应_
> 
> ### Diffconfig
> _No response 无响应_
> 
> ### Terms 条款
> * [x]  I am reporting an issue for OpenWrt, not an unsupported fork.我报告的是 OpenWrt 的问题，而不是不受支持的分叉。