openwrt/openwrt #16448: Broken 2.5G WAN port on MT7981 - Cudy TR3000

LuisMitaHL, 2024-09-22:
### Describe the bug

The 2.5 WAN port is completely dead. No link, no syslog messages, nothing.

### OpenWrt version

r27494-2e626ae2d2

### OpenWrt release

. /etc/openwrt_release && echo $DISTRIB_RELEASE

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy TR3000 v1

### Image kind

Official downloaded image

### Steps to reproduce

1) Flash latest SNAPSHOT firmware from downloads.openwrt.org, do not preserve any settings
2) Watch syslog with logread -f
3) Connect something to WAN port.

### Actual behaviour

Nothing happens. 

### Expected behaviour

If I connect a ethernet cable, negotiation messages should appear on syslog and, being the port a WAN port, getting a IP address with DHCP.

### Additional info

The PHY was involved on another bugs (#15093) but in this case it's recognized correctly.
```
Sat Sep 21 16:29:55 2024 kern.info kernel: [ 1582.135502] mtk_soc_eth 15100000.ethernet eth0: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY (C45)] (irq=POLL)
Sat Sep 21 16:29:55 2024 kern.info kernel: [ 1582.151364] mtk_soc_eth 15100000.ethernet eth0: configuring for phy/2500base-x link mode
```

Another error that I think is related was #16281, but can't cleanly revert the commit.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

LuisMitaHL, 2024-09-22:
The device works on 23.05-SNAPSHOT. The last good commit on main is 963be1423f64156fa66a8a94cf0207ce2945bf34
I'm bisecting from that.

The symptoms changed: with the latest commit all is dead, but after trying another commits, the port is unstable:
```
Sun Sep 22 05:18:33 2024 kern.info kernel: [  220.047957] mtk_soc_eth 15100000.ethernet eth0: Link is Down
Sun Sep 22 05:18:33 2024 kern.info kernel: [  220.054354] br-lan: port 1(eth0) entered disabled state
Sun Sep 22 05:18:33 2024 daemon.notice netifd: Network device 'eth0' link is down
Sun Sep 22 05:18:34 2024 daemon.notice netifd: bridge 'br-lan' link is down
Sun Sep 22 05:18:34 2024 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Sep 22 05:18:38 2024 kern.info kernel: [  225.246786] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Sun Sep 22 05:18:38 2024 kern.info kernel: [  225.246826] br-lan: port 1(eth0) entered blocking state
Sun Sep 22 05:18:38 2024 daemon.notice netifd: Network device 'eth0' link is up
Sun Sep 22 05:18:38 2024 daemon.notice netifd: bridge 'br-lan' link is up
Sun Sep 22 05:18:38 2024 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Sep 22 05:18:38 2024 kern.info kernel: [  225.260282] br-lan: port 1(eth0) entered forwarding state
Sun Sep 22 05:18:40 2024 kern.info kernel: [  227.326102] mtk_soc_eth 15100000.ethernet eth0: Link is Down
Sun Sep 22 05:18:40 2024 kern.info kernel: [  227.332497] br-lan: port 1(eth0) entered disabled state
Sun Sep 22 05:18:40 2024 daemon.notice netifd: Network device 'eth0' link is down
Sun Sep 22 05:18:41 2024 daemon.notice netifd: bridge 'br-lan' link is down
Sun Sep 22 05:18:41 2024 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Sep 22 05:18:42 2024 kern.info kernel: [  229.405790] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Sun Sep 22 05:18:42 2024 kern.info kernel: [  229.405828] br-lan: port 1(eth0) entered blocking state
Sun Sep 22 05:18:42 2024 daemon.notice netifd: Network device 'eth0' link is up
Sun Sep 22 05:18:42 2024 daemon.notice netifd: bridge 'br-lan' link is up
Sun Sep 22 05:18:42 2024 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Sep 22 05:18:42 2024 kern.info kernel: [  229.419287] br-lan: port 1(eth0) entered forwarding state
Sun Sep 22 05:18:43 2024 kern.info kernel: [  230.445374] mtk_soc_eth 15100000.ethernet eth0: Link is Down
Sun Sep 22 05:18:43 2024 kern.info kernel: [  230.451769] br-lan: port 1(eth0) entered disabled state
Sun Sep 22 05:18:43 2024 daemon.notice netifd: Network device 'eth0' link is down
Sun Sep 22 05:18:44 2024 daemon.notice netifd: bridge 'br-lan' link is down
Sun Sep 22 05:18:44 2024 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Sep 22 05:18:47 2024 kern.info kernel: [  233.564847] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Sun Sep 22 05:18:47 2024 kern.info kernel: [  233.564885] br-lan: port 1(eth0) entered blocking state
Sun Sep 22 05:18:47 2024 daemon.notice netifd: Network device 'eth0' link is up
Sun Sep 22 05:18:47 2024 daemon.notice netifd: bridge 'br-lan' link is up
Sun Sep 22 05:18:47 2024 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Sep 22 05:18:47 2024 kern.info kernel: [  233.578360] br-lan: port 1(eth0) entered forwarding state
Sun Sep 22 05:18:48 2024 kern.info kernel: [  234.604443] mtk_soc_eth 15100000.ethernet eth0: Link is Down
Sun Sep 22 05:18:48 2024 kern.info kernel: [  234.610845] br-lan: port 1(eth0) entered disabled state
Sun Sep 22 05:18:48 2024 daemon.notice netifd: Network device 'eth0' link is down
Sun Sep 22 05:18:49 2024 daemon.notice netifd: bridge 'br-lan' link is down
Sun Sep 22 05:18:49 2024 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Sep 22 05:18:49 2024 kern.info kernel: [  235.644423] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
Sun Sep 22 05:18:49 2024 kern.info kernel: [  235.644670] br-lan: port 1(eth0) entered blocking state
Sun Sep 22 05:18:49 2024 kern.info kernel: [  235.658091] br-lan: port 1(eth0) entered forwarding state
Sun Sep 22 05:18:49 2024 daemon.notice netifd: Network device 'eth0' link is up
Sun Sep 22 05:18:49 2024 daemon.notice netifd: bridge 'br-lan' link is up
Sun Sep 22 05:18:49 2024 daemon.notice netifd: Interface 'lan' has link connectivity
```
The endpoint is Gigabit. The port can't negotiate 1Gbps without dropping the link.

LuisMitaHL, 2024-09-22:
I finished the bisect
```
1069514978c419cc5bdfc18c5670b46d0443a686 is the first bad commit
commit 1069514978c419cc5bdfc18c5670b46d0443a686 (HEAD)
Author: Daniel Golle <daniel@makrotopia.org>
Date:   Tue Aug 20 17:20:29 2024 +0100

    mediatek: backport pending Ethernet PHY driver patches
    
    Use pending patchset for 2.5GE PHY driver, unifying LED handling
    accross all MediaTek Ethernet PHYs.
    
    Signed-off-by: Daniel Golle <daniel@makrotopia.org>

 target/linux/mediatek/files-6.6/arch/arm64/boot/dts/mediatek/mt7988a.dtsi                           |   17 +
 target/linux/mediatek/files-6.6/drivers/net/phy/mediatek-2p5ge.c                                    |  321 ------------
 target/linux/mediatek/filogic/config-6.6                                                            |    2 +-
 target/linux/mediatek/patches-6.6/733-01-net-phy-mediatek-Re-organize-MediaTek-ethernet-phy-d.patch | 3513 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 target/linux/mediatek/patches-6.6/733-02-net-phy-mediatek-Fix-spelling-errors-and-rearrange-v.patch |   62 +++
 target/linux/mediatek/patches-6.6/733-03-net-phy-mediatek-Move-LED-helper-functions-into-mtk-.patch |  742 ++++++++++++++++++++++++++
 target/linux/mediatek/patches-6.6/733-04-net-phy-mediatek-Improve-readability-of-mtk-phy-lib..patch |   70 +++
 target/linux/mediatek/patches-6.6/733-05-net-phy-mediatek-Integrate-read-write-page-helper-fu.patch |  141 +++++
 target/linux/mediatek/patches-6.6/733-06-net-phy-mediatek-Hook-LED-helper-functions-in-mtk-ge.patch |  146 ++++++
 target/linux/mediatek/patches-6.6/733-07-net-phy-mediatek-add-MT7530-MT7531-s-PHY-ID-macros.patch   |   54 ++
 target/linux/mediatek/patches-6.6/733-08-net-phy-mediatek-Change-mtk-ge-soc.c-line-wrapping.patch   |  182 +++++++
 target/linux/mediatek/patches-6.6/733-09-net-phy-mediatek-Add-token-ring-access-helper-functi.patch |  614 ++++++++++++++++++++++
 target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch     |  209 ++++++++
 target/linux/mediatek/patches-6.6/733-11-net-phy-add-driver-for-built-in-2.5G-ethernet-PHY-on.patch |  477 +++++++++++++++++
 target/linux/mediatek/patches-6.6/733-12-net-phy-mediatek-Fix-alignment-in-callback-functions.patch |  130 +++++
 target/linux/mediatek/patches-6.6/733-13-net-phy-mediatek-Remove-unnecessary-outer-parens-of-.patch |   65 +++
 target/linux/mediatek/patches-6.6/733-net-phy-add-driver-for-MediaTek-2.5G-PHY.patch                |   39 --
 17 files changed, 6423 insertions(+), 361 deletions(-)
 delete mode 100644 target/linux/mediatek/files-6.6/drivers/net/phy/mediatek-2p5ge.c
 create mode 100644 target/linux/mediatek/patches-6.6/733-01-net-phy-mediatek-Re-organize-MediaTek-ethernet-phy-d.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-02-net-phy-mediatek-Fix-spelling-errors-and-rearrange-v.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-03-net-phy-mediatek-Move-LED-helper-functions-into-mtk-.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-04-net-phy-mediatek-Improve-readability-of-mtk-phy-lib..patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-05-net-phy-mediatek-Integrate-read-write-page-helper-fu.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-06-net-phy-mediatek-Hook-LED-helper-functions-in-mtk-ge.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-07-net-phy-mediatek-add-MT7530-MT7531-s-PHY-ID-macros.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-08-net-phy-mediatek-Change-mtk-ge-soc.c-line-wrapping.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-09-net-phy-mediatek-Add-token-ring-access-helper-functi.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-11-net-phy-add-driver-for-built-in-2.5G-ethernet-PHY-on.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-12-net-phy-mediatek-Fix-alignment-in-callback-functions.patch
 create mode 100644 target/linux/mediatek/patches-6.6/733-13-net-phy-mediatek-Remove-unnecessary-outer-parens-of-.patch
 delete mode 100644 target/linux/mediatek/patches-6.6/733-net-phy-add-driver-for-MediaTek-2.5G-PHY.patch
```

I'm closely following the main branch for months, and had my suspicions of this commit before.

This problem is complicated to debug. Sometimes the link works, but it is possible to get the bug by connecting all the cables somewhere (empty switches, etc) and turning on the router.
At other times it is enough to connect the cable after the router is turned on. Sometimes negotiates 1Gbps at full speed, sometimes 100Mb/s (without any traffic passing).
The behavior looks a lot like https://github.com/openwrt/netifd/commit/68c8a4f94cd3cfd654a52cbc8b57c5c9d99640dd, but it was not detected in the git bisect and the bug continues on the latest SNAPSHOT builds.

I rebuilt 963be1423f64156fa66a8a94cf0207ce2945bf34 (the last known working commit before 1069514978c419cc5bdfc18c5670b46d0443a686) and everything works OK without failing any attempt connecting a cable. If I try to `ethtool -s eth0 autoneg on` (disabled by default), the port works with no problems.

My impression is that 1069514978c419cc5bdfc18c5670b46d0443a686 introduced several bugs and https://github.com/openwrt/netifd/commit/68c8a4f94cd3cfd654a52cbc8b57c5c9d99640dd was blamed for that. The https://github.com/openwrt/netifd/commit/68c8a4f94cd3cfd654a52cbc8b57c5c9d99640dd revert fixed the problem for other MT798x owners (I own a Confiabits MT7981 and everything looks OK), but 1069514978c419cc5bdfc18c5670b46d0443a686 has broken more things that have not been discovered yet.

I appreciate your help trying to understand where the problem is.

map-b, 2024-09-22:
I would try a build without this netifd's commit revert https://github.com/openwrt/netifd/commit/02aa43d82d587d87523231be5a5e0bf956435725

Including the last commit from nbd.

LuisMitaHL, 2024-09-24:
Full syslog:
```
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Tue Sep 24 10:29:10 2024 kern.notice kernel: [    0.000000] Linux version 6.6.52 (looper@amelia) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r25978+1189-ea609fe486) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue Sep 24 10:29:06 2024
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Machine model: Cudy TR3000 v1
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable secmon@43000000
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000047c80000..0x0000000047d7ffff (1024 KiB) nomap non-reusable wmcpu-reserved@47c80000
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000047d80000..0x0000000047dbffff (256 KiB) nomap non-reusable wo-emi@47d80000
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000047dc0000..0x0000000047ffffff (2304 KiB) nomap non-reusable wo-data@47dc0000
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Zone ranges:
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000]   DMA32    empty
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000]   Normal   empty
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Early memory node ranges
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043030000-0x0000000047c7ffff]
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000047c80000-0x0000000047ffffff]
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000048000000-0x000000005fffffff]
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] psci: PSCIv1.1 detected in firmware.
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.2
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] percpu: Embedded 18 pages/cpu s34600 r8192 d30936 u73728
Tue Sep 24 10:29:10 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s34600 r8192 d30936 u73728 alloc=18*4096
Tue Sep 24 10:29:10 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] CPU features: detected: GIC system register CPU interface
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] alternatives: applying boot alternatives
Tue Sep 24 10:29:10 2024 kern.notice kernel: [    0.000000] Kernel command line:
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 0MB
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] software IO TLB: area num 2.
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] software IO TLB: mapped [mem 0x000000005fd68000-0x000000005fde8000] (0MB)
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] Memory: 496720K/524288K available (8640K kernel code, 896K rwdata, 2532K rodata, 448K init, 289K bss, 27568K reserved, 0K cma-reserved)
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] rcu:  RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000]       Tracing variant of Tasks RCU enabled.
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Tue Sep 24 10:29:10 2024 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[... cut ...]
I think the problem alternates:
- Sometimes the connection is not recognized at all
- Sometimes the router detects the connection, but cannot negotiate 1Gbps and stays at 100Mbps. The connection stays broken, does not work and is not solved by reconnecting.

I tried disconnecting LAN and testing this only with WAN connected. The error persists.

When I was doing the bisect, the router behavior seemed to alternate between these two states for no apparent reason as the commits seemed to be unrelated.

JulioMC81, 2024-09-25:
Hi! I have a similar problem with Netgear WAX 206 router. The WAN port (RTL8221B, 2500M), doesn't work with snapshot build. With 23.05 branch, works perfect.

Thanks!

LuisMitaHL, 2024-09-29:
How to fix the problem:
- Get latest OpenWrt from git
- Revert [733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch](https://github.com/openwrt/openwrt/commits/main/target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch) following this instructions: https://github.com/openwrt/openwrt/issues/16281#issuecomment-2322095969 or this: https://github.com/openwrt/openwrt/issues/16281#issuecomment-2327321928
- Using the latest netifd, reinstate the reverted commit https://github.com/openwrt/netifd/commit/68c8a4f94cd3cfd654a52cbc8b57c5c9d99640dd. Or you can skip this step and enable autoneg with `ethtool -s eth0 autoneg on`

The 2.5G interface can correctly negotiate 1Gbps and 100Mbps now (both with a 100Mbps switch, or a 1000Mbps switch + 4-wire cable)

@dangowrt sorry for tagging. I think you have more experience with the driver, I am willing to help with more testing

samed, 2024-09-30:
Same here, just came with some ethtool output. I have to run `ethtool -s eth0 autoneg on speed 1000 duplex full` in each reboot.

Before running:

```
Settings for eth0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Full
                                100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: No
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Full
                                100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: No
        Advertised FEC modes: Not reported
        Speed: 2500Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: off
        MDI-X: Unknown
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: no
```

After setting autoneg:

```
Settings for eth0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Full
                                100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: No
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: No
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Half 1000baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: yes
```

closed 2024-10-09 by openwrt-bot

faberto, 2024-12-09:
Still broken with snapshot from Thu Dec 5 17:40:14 2024

LuisMitaHL, 2024-12-10:
I upgraded my TR3000 from 23.05.5 to 24.10.0-rc2 and WAN port broke again. Setting `autoneg on` with ethtool restores connectivity, like https://github.com/openwrt/openwrt/issues/17183

I can see a new error on dmesg:
```
[   15.606166] RTL8221B-VB-CG 2.5Gbps PHY (C45) mdio-bus:01: rtl822xb_config_init failed: -110
[   15.725591] mtk_soc_eth 15100000.ethernet eth0: mtk_open: could not attach PHY: -110
```

<details><summary>Full dmesg</summary>
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.63 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28161-ea17e958b9) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue Dec  3 11:41:08 2024
[    0.000000] Machine model: Cudy TR3000 v1
[    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
[    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable secmon@43000000
[    0.000000] OF: reserved mem: 0x0000000047c80000..0x0000000047d7ffff (1024 KiB) nomap non-reusable wmcpu-reserved@47c80000
[    0.000000] OF: reserved mem: 0x0000000047d80000..0x0000000047dbffff (256 KiB) nomap non-reusable wo-emi@47d80000
[    0.000000] OF: reserved mem: 0x0000000047dc0000..0x0000000047ffffff (2304 KiB) nomap non-reusable wo-data@47dc0000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x0000000047c7ffff]
[    0.000000]   node   0: [mem 0x0000000047c80000-0x0000000047ffffff]
[    0.000000]   node   0: [mem 0x0000000048000000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s35560 r8192 d29976 u73728
[    0.000000] pcpu-alloc: s35560 r8192 d29976 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 0MB
[    0.000000] software IO TLB: area num 2.
[    0.000000] software IO TLB: mapped [mem 0x000000005fd68000-0x000000005fde8000] (0MB)
[    0.000000] Memory: 497040K/524288K available (9024K kernel code, 996K rwdata, 1664K rodata, 448K init, 307K bss, 27248K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 640 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: 0xffffffc080010080
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000000] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.000072] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)
[    0.000080] pid_max: default: 32768 minimum: 301
[... 195 lines cut ...]
[   15.246609] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/gmii link mode
[   15.257127] br-lan: port 1(eth1) entered blocking state
[   15.262360] br-lan: port 1(eth1) entered disabled state
[   15.267682] mtk_soc_eth 15100000.ethernet eth1: entered allmulticast mode
[   15.274699] mtk_soc_eth 15100000.ethernet eth1: entered promiscuous mode
[   15.606166] RTL8221B-VB-CG 2.5Gbps PHY (C45) mdio-bus:01: rtl822xb_config_init failed: -110
[   15.725591] mtk_soc_eth 15100000.ethernet eth0: mtk_open: could not attach PHY: -110
[   18.597180] br-lan: port 2(phy0-anna) entered blocking state
[... cut ...]
[   19.256557] br-lan: port 3(phy1-anna) entered disabled state
[   19.262269] mt798x-wmac 18000000.wifi phy1-anna: entered allmulticast mode
[   19.269377] mt798x-wmac 18000000.wifi phy1-anna: entered promiscuous mode
[   19.276376] br-lan: port 3(phy1-anna) entered blocking state
[   19.282037] br-lan: port 3(phy1-anna) entered forwarding state
[   19.623965] br-lan: port 3(phy1-anna) entered disabled state
[   20.535068] br-lan: port 3(phy1-anna) entered blocking state
[   20.540791] br-lan: port 3(phy1-anna) entered forwarding state

```
</details>

samed, 2024-12-11:
Experiencing the same, shall we create another ticket as this one is closed?

v4lkyr, 2024-12-12:
Same problem on Ruijie RG-X60 Pro too. It's the same 2.5G PHY chip - RTL8221B-VB-CG

v4lkyr, 2024-12-21:
> @LuisMitaHL @v4lkyr could you try swapping lan port with wan to see if it helps?

Well that's exactly how I am using the device atm, since WAN port is not working, right?

samed, 2024-12-21:
No, I mean set wan interface to `eth1` and br-lan's parent device to `eth0` to see if you get both ports are working. By using this way I got both ports working.

ab77, 2024-12-30:
Swapping the ports in the DTS like [this](https://github.com/openwrt/openwrt/issues/17183#issuecomment-2563900981), does appear to result in a working network stack, but it isn't a workable solution IMO. Setting the 2.5G port from `phy-mode = "2500base-x";` to `gmii` breaks the WAN port entirely. The only way to get a working WAN with stock TR3000 V1 DTS and current 23.05 is to issue `/etc/init.d/network restart` over the serial console after boot.

dangowrt, 2025-01-23:
Hopefully this can be fixed for real by https://github.com/openwrt/openwrt/pull/17701

ab77, 2025-01-23:
> Hopefully this can be fixed for real by [#17701](https://github.com/openwrt/openwrt/pull/17701)

@dangowrt thank you, how trivial would it be to port this patch to kernel-5.15 in 23.05?

dangowrt, 2025-01-23:
@ab77 The problem was only introduced after 23.05 had branched, so there shouldn't be a need to backport anything. Or did I get that wrong?

ab77, 2025-01-23:
> [@ab77](https://github.com/ab77) The problem was only introduced after 23.05 had branched, so there shouldn't be a need to backport anything. Or did I get that wrong?

@dangowrt  well, it could bit a different problem, but in the latest `upstream/openwrt-23.05`, TR3000V1 does not boot with a working WAN interface (`RTL8221B-VB-CG 2.5Gbps PHY (C45) mdio-bus:01: rtl822xb_config_init failed: -110` is logged).

The only way to make the WAN interface work is to issue a `/etc/init.d/network restart` or swap WAN and LAN ports in the DTS.

JulioMC81, 2025-01-23:
> [@ab77](https://github.com/ab77) The problem was only introduced after 23.05 had branched, so there shouldn't be a need to backport anything. Or did I get that wrong?

I'm using a Cudy TR3000 with the latest 23.05 snapshot and works perfect. The problem only occurs with the latest 24.05 branch.

Thanks!

ab77, 2025-01-23:
> I'm using a Cudy TR3000 with the latest 23.05 snapshot and works perfect. The problem only occurs with the latest 24.05 branch.
>
> Thanks!

Are you able to post a boot log from your board with anything sensitive redacted please?

JulioMC81, 2025-01-24:
> > > [@ab77](https://github.com/ab77) The problem was only introduced after 23.05 had branched, so there shouldn't be a need to backport anything. Or did I get that wrong?
> > 
> > 
> > I'm using a Cudy TR3000 with the latest 23.05 snapshot and works perfect. The problem only occurs with the latest 24.05 branch.
> > Thanks!
> 
> [@JulioMC81](https://github.com/JulioMC81) are you able to post a boot log from your board with anything sensitive redacted please? (also the DTS you are using and if you are booting with the WAN port cable plugged in during boot).

If I can, I'll try this weekend and send the boot log. Yes, I have an ONT (UFiber Loco) connected to the WAN port.

ab77, 2025-01-25:
So this also might only be an issue when it's connected to a 1gig port and working fine when connected to 2.5g?

dangowrt, 2025-01-25:
@ab77 Yes, exactly. The PHY not doing autonegotiation but rather being set to a fixed speed of 2500M is only an issue if the WAN port of the TR3000 is connected to a link-partner which isn't capable of 2500M speed, but only 1000M or 100M.

ab77, 2025-01-27:
@dangowrt is there a software workaround to make it work when WAN is connected to a non-2.5G port?

commit 7d256aff (2024-10-09): mediatek: update MediaTek PHY patchset

The previous iteration of MediaTek's PHY patches caused various weird bugs.
Drop culprit patch 733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch
and use the most recent iteration of the patchset which has been posted to the
netdev mailing list.

Link: https://patchwork.kernel.org/project/netdevbpf/list/?series=895513&state=*
Fixes: #16448