openwrt/openwrt #16452: ipq807x: emmc: all rootfs_data lost during sysupgrade

Lanchon, 2024-09-22:
### Describe the bug

the only ipq807x devices with emmc are:
- spectrum sax1v1k
- zyxel nbg7815

during sysupgrade the loop device within rootfs is created at the wrong offset and thus data in $UPGRADE_BACKUP is lost.

the offset is required to be a multiple of 64K:
https://github.com/openwrt/fstools/blob/408c2cc48e6694446c89da7f8121b399063e1067/libfstools/rootdisk.c#L18

but the code does not do that:
https://github.com/openwrt/openwrt/blob/df1011e0b7c0c2497d40d57eff0cd604b8aece48/target/linux/qualcommax/ipq807x/base-files/lib/upgrade/mmc.sh#L53-L59

am i right?

this hasn't been noticed because there are further issues with the spectrum sax1v1k port:
- the `rootfs_data` partition (a real GPT partition) is not touched at all during sysupgrade
- if data wipe is requested during sysupgrade, the request is ignored
- all files in `rootfs_data`, not just those being backed up, survive sysupgrade

i imagine zyxel nbg7815 must be in a similar situation. 

but for now i want to focus of this loop device bug.

in fact, why does [this file](https://github.com/openwrt/openwrt/blob/master/target/linux/qualcommax/ipq807x/base-files/lib/upgrade/mmc.sh) exist at all? why not use [emmc.sh](https://github.com/openwrt/openwrt/blob/master/package/base-files/files/lib/upgrade/emmc.sh) instead?

### OpenWrt version

r27466-f368e2d5ec

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Spectrum SAX1V1K

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kirdesde, 2024-09-22:
The file https://github.com/openwrt/openwrt/blob/master/target/linux/qualcommax/ipq807x/base-files/lib/upgrade/mmc.sh has been added as part of the commit for the QNAP301w and is working fine (at least for the 301w and the NBG7815)
The mmc.sh is more or less a copy from the ipq806x target and sets up a rootfs_data loop device.

The emmc upgrade script https://github.com/openwrt/openwrt/blob/master/package/base-files/files/lib/upgrade/emmc.sh doesn't do that. e.g. the device would need a rootfs_data partition.

This in turn would mean, OpenWrt would overwrite files from the stock firmware on that partition. To avoid that, using a loop device on the rootfs partition is best practise here (in addition not every emmc device even has a rootfs_partition)

Can you post a full log of the sax1v1k sysupgrade process?

kirdesde, 2024-09-22:
@gotofbi  adding you since you added support for that device

Lanchon, 2024-09-22:
> @gotofbi adding you since you added support for that device

i'm also adding A/B (dual slot) booting to this device. please don't do changes to fix just this bug. thank you!

Lanchon, 2024-09-22:
> The mmc.sh is more or less a copy from the ipq806x target (https://github.com/openwrt/openwrt/blob/main/target/linux/ipq806x/base-files/lib/upgrade/zyxel.sh, including the offset) and sets up a rootfs_data loop device.
> 
> The emmc upgrade script https://github.com/openwrt/openwrt/blob/master/package/base-files/files/lib/upgrade/emmc.sh doesn't do that. e.g. the device would need a rootfs_data partition.

no, emmc.sh works fine in devices without rootfs_data. loop is not needed.

instead of creating a loop, formatting in ext4, mounting, copying the $UPGRADE_BACKUP file to it, unmounting the fs and removing the loop,  the upgrade process simple writes $UPGRADE_BACKUP to the area where the loop would be, as seen here:

https://github.com/openwrt/openwrt/blob/04bdf9b3323e1d8caa04af4a930100894b2b869b/package/base-files/files/lib/upgrade/emmc.sh#L50-L58

upon boot, fstools will create the loop and format it using ext4 or f2fs (depending on loop size), and expand the backup there.

this is the expected working of a loop device in the fstools repo. the loop device is only needed when booting openwrt, and the filesystem is created by the *new* OS (the one actually using it), not the old OS (from which you are upgrading away). it is perfectly legal for the new OS to switch rootfs_data to btrfs or whatever if it wanted: the only required common format between them is the backup file.

this setup works [perfectly ok](https://github.com/openwrt/openwrt/blob/935b2b7dcef61b2893ed5dff307dd8f8a1156899/target/linux/ipq40xx/base-files/lib/upgrade/linksys.sh#L192) in the WHW03 v1 which i ported, and [does not have rootfs_data partitions](https://openwrt.org/toh/linksys/whw03_v1#flash_layout).

> BTW
> 
> ```
> define ROOTDEV_OVERLAY_ALIGN	(64ULL * 1024ULL)
> ```
>
> defines exactly the offset size of 65536

that is not an offset size, it is an offset granularity. it is used to [round up the start of the loop device to a multiple of that value](https://github.com/openwrt/fstools/blob/408c2cc48e6694446c89da7f8121b399063e1067/libfstools/rootdisk.c#L121-L122) (as long as the value is a power of 2).

> Can you post a full log of the sax1v1k sysupgrade process?

no use; the sax1v1k sysupgrade is broken. as said before, sax1v1k *does have a rootfs_data partition*, so writing the backup data to the back of the rootfs partition is a mistake. moreover, during next boot this data is completely ignored. settings are only retained though a sysupgrade (whether you want it or not) simply because the rootfs_data is never cleared. this is very dangerous: if you install a package, then do a sysupgrade, the package for the old OS will continue to be installed!

no matter the bugs in the sax1v1k device, here i am pointing to a different bug in mmc.sh file.

> [mmc.sh] is working fine [for] the NBG7815

i don't know [that device](https://openwrt.org/inbox/toh/zyxel/nbg7815_armor_g5). (it seems to have both nand flash and emmc? the OEM bootlog shows an MTD device?) without a description of the device i can't asses the impact of this bug on that device. the openwrt bootlog is not populated in the wiki.

but if you want to be sure it does not suffer from the same issues as the sax1v1k, you can create a random file in the root of the device and then do a sysupgrade keeping settings. if the file survives, then sysupgrade is *not* working.

Lanchon, 2024-09-22:
my point here is simple:

- mmc.sh is not aligning to ROOTDEV_OVERLAY_ALIGN (64 KiB)
- and i believe fstools will align a loop on rootfs to ROOTDEV_OVERLAY_ALIGN during boot
- which means that loop during sysupgrade and loop during subsequent boot will be created in different places
- which means fstools will not find the backed-up data

where am i wrong in this reasoning?

Lanchon, 2024-09-23:
btw, i decided to decouple things and fix the sax1v1k first: https://github.com/openwrt/openwrt/pull/16463

pwned-pixel, 2024-09-24:
> i don't know [that device](https://openwrt.org/inbox/toh/zyxel/nbg7815_armor_g5). (it seems to have both nand flash and emmc? the OEM bootlog shows an MTD device?) without a description of the [...]

As I never choose to preserve any settings I have never tested this szeanario. But I remember similar issues for NBG6815 in the past. Which was the main reason (in the past) for me to move everything into a setup script as I'm doing my own builds anyway.

So I've tested it for NBG7815 and I can confirm the file exists after the upgrade (created a dummy file within /).

Lanchon, 2024-09-24:
hi @pwned-pixel 

the wiki is missing the bootlog:
https://openwrt.org/inbox/toh/zyxel/nbg7815_armor_g5#openwrt_bootlog

so you could do this. with openwrt loaded:
- type `mount` and paste the contents here
- type `dmesg` and capture the whole output and put it in a pastebin or a gist
- if you could install packages gdisk, type `gdisk /dev/mmcblk0` then `p` for print, and paste output here

your device is dual boot (A/B boot aka dual system) and is a bit more complex. i don't know the maintainer intended for the device to work. it would help if the maintainer had pasted the bootlog to the wiki, but didn't.

pwned-pixel, 2024-09-24:
Hi @Lanchon 

> the wiki is missing the bootlog:

Y, there is no real maintainer available. But I know that @Ansuel has this device and made LED available for it. He has surley more insight about this specific device. Maybe he has time to say sth. about if needed.

> your device is dual boot (A/B boot aka dual system) and is a bit more complex.

We have dual partition that's true. But it is not used at all by OpenWrt. Rootfs_data isn't touched also. I've changed the sysupgrade process for me alternating the partitions on sysupgrade and it works flawlessly.

OEM firmware is just altering between HLOS/HLOS_1 and rootfs/rootfs_1. Rootfs_data is not touched by the sysupgrade process during OEM firmware upgrade. They do not save the any privious settings explicit and just leave the config in place on rootfs_data. But you can delete/reset settings during upgrade explicit (they just format it).

There isn't much on rootfs_data. It just holds OS settings. App settings and logs made by OS/ZyXEL are on mmcblkp11.

I did this test above with a vanilla OpenWrt (just to be sure) and reverted back to my moded system already. As I run a nss build I cannot provide the vanilla dmesg. So I just cut out the important parts. If you need more information I would install a vanilla build again to provide the full OpenWrt log.

```
NBG7815A:~# sgdisk -p /dev/mmcblk0
Disk /dev/mmcblk0: 7405568 sectors, 3.5 GiB
Sector size (logical/physical): 512/512 bytes
Disk identifier (GUID): 98101B32-BBE2-4BF2-A06E-2BB33D000C20
Partition table holds up to 12 entries
Main partition table begins at sector 2 and ends at sector 4
First usable sector is 34, last usable sector is 7405534
Partitions will be aligned on 2-sector boundaries
Total free space is 282557 sectors (138.0 MiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1              34           16417   8.0 MiB     FFFF  rootfs_data
   2           16418           18465   1024.0 KiB  FFFF  header
   3           18466           30753   6.0 MiB     FFFF  0:HLOS
   4           30754          153633   60.0 MiB    FFFF  rootfs
   5          153634          161825   4.0 MiB     FFFF  0:WIFIFW
   6          161826          163873   1024.0 KiB  FFFF  header_1
   7          163874          176161   6.0 MiB     FFFF  0:HLOS_1
   8          176162          299041   60.0 MiB    FFFF  rootfs_1
   9          299042          307233   4.0 MiB     FFFF  0:WIFIFW_1
  10          307234         1355809   512.0 MiB   FFFF  nbg
  11         1355810         7122977   2.8 GiB     FFFF  appdata

```
Partition 10 + 11 do not hold the origin name here. I've changed them.

```
NBG7815A:~# mount
/dev/root on /rom type squashfs (ro,relatime,errors=continue)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/loop0 on /overlay type ext4 (rw,noatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work,uuid=on,xino=off)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
bpffs on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
pstore on /sys/fs/pstore type pstore (rw,noatime)
/dev/mmcblk0p11 on /root/data type ext4 (rw,noatime)
```

```
[    1.053660] 22 fixed-partitions partitions found on MTD device spi0.0
[    1.057962] Creating 22 MTD partitions on "spi0.0":
[    1.064279] 0x000000000000-0x000000050000 : "0:sbl1"
[    1.069423] 0x000000050000-0x000000060000 : "0:mibib"
[    1.074563] 0x000000060000-0x000000080000 : "0:bootconfig"
[    1.079404] 0x000000080000-0x0000000a0000 : "0:bootconfig1"
[    1.084828] 0x0000000a0000-0x000000220000 : "0:qsee"
[    1.090422] 0x000000220000-0x0000003a0000 : "0:qsee_1"
[    1.095462] 0x0000003a0000-0x0000003b0000 : "0:devcfg"
[... 18 lines cut ...]
NBG7815A:~# dmesg | grep -i mmc
[    1.299099] mmc0: SDHCI controller on 7824900.mmc [7824900.mmc] using ADMA 64-bit
[    1.403940] mmc0: new HS200 MMC card at address 0001
[    1.420052] mmcblk0: mmc0:0001 M62704 3.53 GiB
[    1.435173]  mmcblk0: p1 p2 p3 p4 p5 p6 p7 p8 p9 p10 p11
[    1.449784] mmcblk0boot0: mmc0:0001 M62704 2.00 MiB
[    1.465377] mmcblk0boot1: mmc0:0001 M62704 2.00 MiB
[    1.480655] mmcblk0rpmb: mmc0:0001 M62704 512 KiB, chardev (247:0)
[   53.596189] EXT4-fs (mmcblk0p11): warning: mounting unchecked fs, running e2fsck is recommended
[   53.599416] EXT4-fs (mmcblk0p11): mounted filesystem 66012e89-f08b-41f0-8695-1f1a7a434cd4 r/w without journal. Quota mode: disabled.
```
mmcblk0p11 is not automounted on vanialla build AFAIR.

```
NBG7815A:~# df -T
Filesystem           Type       1K-blocks      Used Available Use% Mounted on
/dev/root            squashfs       39680     39680         0 100% /rom
tmpfs                tmpfs         442992      7056    435936   2% /tmp
/dev/loop0           ext4           19087       692     16868   4% /overlay
overlayfs:/overlay   overlay        19087       692     16868   4% /
tmpfs                tmpfs            512         0       512   0% /dev
/dev/mmcblk0p11      ext4         2829880        24   2813472   0% /root/data
/dev/mmcblk0p1       ext4            6876      5099      1205  81% /1
```
I've mounted /dev/mmcblk0p1 here to document the fs.

shvetsnikita, 2024-09-24:
> the wiki is missing the bootlog

I also noticed that all download links point to obsolete [ipq807x ](https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/)target instead of [qualcommax](https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/).

Lanchon, 2024-09-24:
@pwned-pixel 

i'm sorry i can't help you. there are too many unknowns, including you running modified versions of openwrt.

there seems to be code using dual partitions, but you say dual partitions are not used.

rootfs_data doesn't seem to be used, but you say it is being used.

/dev/mmcblk0p11 (appdata) shouldn't be mounted, but it is mounted on /root/data.

way too many unknows. i can't help you without having a device. sorry.

Lanchon, 2024-09-24:
@pwned-pixel 

ok so i GUESSED a solution... and i mean GUESSED. it is here:
https://github.com/Lanchon/openwrt/tree/nbg7815-fix-sysupgrade

you seem to be able to do you own builds, right?

if you want to test it and you can confirm it makes sysupgrade work, i can do a PR for it.

how do you know it works?
- "sysupgrade retaining settings" actually retain settings, but random files in / are NOT retained
- "sysupgrade clearing settings" actually clear settings
- settings backup, restore, and reset (independent of sysupgrade) work as expected
 
(also, please include the output of this command `fw_printenv`. you can do that from any openwrt, modified or not.)

pwned-pixel, 2024-09-25:
Hi @Lanchon,

> there seems to be code using dual partitions, but you say dual partitions are not used.

On vanilla OpenWrt the dual partition system is not used. The way Zyxel implemented dual boot is a bit weird. I made it work for myself patching the upgrade process. But there is no fail over like most ppl. would expect/want for a dual boot system. Zyxel didn't implement it. So a fail over would need a replacement of the Zyxel-loader. So like a 2nd u-boot instead the z-loader. But that is beyond my skills.

> rootfs_data doesn't seem to be used, but you say it is being used.

Same as above. It is not used. I don't know why.

> /dev/mmcblk0p11 (appdata) shouldn't be mounted, but it is mounted on /root/data.

This is true. It is mounted on my own build only.

>ok so i GUESSED a solution... and i mean GUESSED. it is here:
>https://github.com/Lanchon/openwrt/tree/nbg7815-fix-sysupgrade

That's exactly what I would have done also. Just mimic the patch for the other device.

>if you want to test it and you can confirm it makes sysupgrade work, i can do a PR for it.

I've tested it making a vanilla build just with your 2 patch files. The dummy file created in "/" is not there anymore after upgrade. I've tested backup, restore and reset also.

So for me it works on NBG7815. Thank you for bringing this patch. 👍

>please include the output of this command fw_printenv

```
root@OpenWrt:~# fw_printenv 
baudrate=115200
bootcmd=bootipq
bootdelay=2
boxid=S212L09100398
countrycode=E1
defenv=sf probe && sf erase 0x600000 0x10000
ethact=eth0
ethaddr=BC:CF:4F:XX:XX:XX
fdt_high=0x4A400000
fdtcontroladdr=4a9674b0
flash_type=8
fwimg=norplusemmc-ipq807x_64-single.100ABSK3C0.img
ipaddr=192.168.1.1
ledB=run ledinit && i2c mw 0x40 0x18 0xff && i2c mw 0x40 0x1b 0xff && i2c mw 0x40 0x1e 0xff
ledG=run ledinit && i2c mw 0x40 0x17 0xff && i2c mw 0x40 0x1a 0xff && i2c mw 0x40 0x1d 0xff
ledR=run ledinit && i2c mw 0x40 0x16 0xff && i2c mw 0x40 0x19 0xff && i2c mw 0x40 0x1c 0xff
ledRblink=i2c mw 0x40 0x16 0xff && i2c mw 0x40 0x19 0xff && i2c mw 0x40 0x1c 0xff && sleep 1 && i2c mw 0x40 0x16 0x00 && i2c mw 0x40 0x19 0x00 && i2c mw 0x40 0x1c 0x00 && sleep 1
ledallon=run ledinit && mw 0x44000000 ffffffff 3 && i2c write 0x44000000 0x32 0x16 9 && i2c write 0x44000000 0x35 0x16 9
ledinit=i2c dev 0 && i2c probe && i2c mw 0x40 0x0 0x40 && mw 0x44000000 a0a0a0a0 3 && i2c write 0x44000000 0x40 0x7 9 && mw 0x44000000 00000000 3 && i2c write 0x44000000 0x40 0x16 9
machid=80100f1
netmask=255.255.255.0
partno=NBG7815-EU0102F
preboot=dcache off
progimg=aq_load_fw 8 && tftpboot 0x44000000 $fwimg && imgaddr=0x44000000 && source $imgaddr:script && reset
progzyxelimg=setenv machid 80100f1 && setenv bootcmd bootipq && saveenv && aq_load_fw 8 && tftpboot 0x44000000 $fwimg && imgaddr=0x44000000 && source $imgaddr:script && reset
serialnum=xxxxxxxxxxxxxx
serverip=192.168.1.99
soc_version_major=2
soc_version_minor=0
stderr=serial@78B3000
stdin=serial@78B3000
stdout=serial@78B3000
readonly=ro
bootargs=ttyMSM0,115200n8
```

If you still need the dmesg output here it is:

https://0x0.st/XYth.txt

Lanchon, 2024-09-28:
@pwned-pixel i noticed you sent a PM about supporting dual firmware, then deleted it. in you need help, why don't you open a public thread on the forum so that everybody can chip in? thanks

Lanchon, 2025-01-11:
fixed in these commits:
https://github.com/openwrt/openwrt/pull/16463
https://github.com/openwrt/openwrt/pull/16505

closed 2025-01-11 by Lanchon