openwrt/openwrt #16465: 请问openwrt怎么设置纯IPV6方式？

xibeilang0616, 2024-09-23:
### Describe the bug

### **光猫桥接**
### **openwrt拨号**
### **在openwrt翻来找去，找不到关闭IPV4的地方。**
### **试验在光猫设置了纯IPV6桥接，openwrt拨号后还是有运营商网关设备的IPV4的内网地址、网关、DSN、公网IPV4地址**
### **实在请教各位大神**

### OpenWrt version

23.05-SNAPSHOT

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

IPV4&IPV6

### Device

CMCC rax3000m

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-09-23:
English?

xibeilang0616, 2024-09-23:
> English?

I'm Chinese, I don't speak English

brada4, 2024-09-23:
Probably you should ask your general question at immortalwrt chat then?
(google translation indicates it is not a bug report, it is a general usage question)

jow-, 2024-09-23:
Please use the forums or mailing lists for support questions. Also please ask your questions in english.

closed 2024-09-23 by jow-