openwrt/openwrt #16468: All radios on AX6S/AX3200 don't work because of 6GHz

HW-OLE, 2024-09-23:
### Describe the bug

After doing a sysupgrade to the newest SNAPSHOT all of my radios won't come up and the frequency is reported as 6GHz, although this device doesn't support it. I suspect it's a issue in the mt76 driver because Mediatek now maybe seems to offer a mt7622 with WiFI 7 (source: https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+log)

### OpenWrt version

r27563-7850e19767

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Steps to reproduce

Just create a 5GHz radio and notice it won't come up, it'll show the following
[image]

### Actual behaviour

All radios don't come up and the frequency shows that it's operating at 6GHz although this Device doesn't have 6.x GHz

### Expected behaviour

The radios come up normally and the radio shows a frequency in the 5GHz range because it's actually operating in that range

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

HW-OLE, 2024-09-23:
[image]
Sorry for editing but I had to blur sensible information

rany2, 2024-09-23:
> source: https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+log)

Can't reproduce on my Redmi AX6S on latest SNAPSHOT. Just to confirm, you're not applying any of the MTK patches you linked to?

HW-OLE, 2024-09-24:
No I didn't Just pure Stock OpenWRT. Will reflash the factory one tomorrow and start from Scratch

HW-OLE, 2024-09-24:
OK, the fix was fairly easy. I just had to recreate all of the SSIDs and now it's just working fine

closed 2024-09-24 by HW-OLE