openwrt/openwrt #16474: ath79:  Kernel panic happened when load ath9k

chilinkiot, 2024-09-24:
### Describe the bug

Hi,
     I built my board image using the latest main branch code. When the module kmod-ath9k is built-in, kernel panic will happen during device boot. The kernel log as below:
-----------------------------------------------------
[   20.395151] xt_time: kernel timezone is -0000
[   20.436829] usbcore: registered new interface driver cdc_ether
[   20.488437] usbcore: registered new interface driver cdc_ncm
[   20.692302] urngd: v1.0.2 started.
[   21.177173] usbcore: registered new interface driver wwan_qmi
[   21.276780] usbcore: registered new interface driver rndis_host
[   21.541417] Data bus error, epc == 80347d38, ra == 835a27b4
[   21.552562] Oops[#1]:
[   21.557085] CPU: 0 PID: 722 Comm: kmodloader Tainted: G           O       6.6.52 #0
[   21.572321] $ 0   : 00000000 00000001 deadc0de 00000100
[   21.582740] $ 4   : b81143fc 000143fc 83461d44 00000004
[   21.593159] $ 8   : 00000000 8072a75c 0000000c 80967dec
[   21.603579] $12   : 00000000 000055ea 00000000 fffffffb
[   21.613982] $16   : 83558080 00000018 81a85b74 81a85b78
[   21.624402] $20   : 81a85f73 80000003 00010000 00014000
[   21.634822] $24   : 00000010 00000020
[   21.645224] $28   : 81a84000 81a85af8 00031018 835a27b4
[   21.655645] Hi    : 0000547b
[   21.661364] Lo    : 0b700000
[   21.667084] epc   : 80347d38 ioread32+0xc/0x14
[   21.675948] ra    : 835a27b4 ath9k_hw_setup_statusring+0x1498/0x26b4 [ath9k_hw]
[   21.690566] Status: 1000fc03 KERNEL EXL IE
[   21.698895] Cause : 1080001c (ExcCode 07)
[   21.706862] PrId  : 00019374 (MIPS 24Kc)
[   21.714655] Modules linked in: ath9k(O+) ath9k_common(O) ath9k_hw(O) ath(O) simcom_wwan rndis_host quectel_wwan qmi_wwan nf_flow_table_inet meig_wwan mac80211(O) ipt_REJECT cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TPROXY xt_TEE xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usbnet ts_fsm ts_bm rng_core pptp ppp_async nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_ct nft_connlimit nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp(O) nf_nat_pptp nf_nat_irc
[   21.715184]  nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_flow_table nf_dup_ipv6 nf_dup_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_rtsp(O) nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount libcrc32c l2tp_ppp iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ip6table_raw ip_tables compat(O) act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4
[   21.894828]  ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_gre ifb ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel tunnel6 xfrm_user xfrm_ipcomp af_key xfrm_algo vxlan crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic seqiv drbg md5 hmac echainiv geniv ecb des_generic libdes deflate cmac cbc authencesn authenc arc4 crypto_acompress fsl_mph_dr_of ehci_fsl gpio_button_hotplug(O) phy_ar7200_usb phy_ar7100_usb mii crc32c_generic
[   22.177414] Process kmodloader (pid: 722, threadinfo=6fb4ee8e, task=dc7bf8c2, tls=77e4ddf4)
[   22.194037] Stack : 83558a58 00000220 00010001 83558080 81a85bdc 00000000 00000220 000003ff
[   22.210694]         81b58000 00010001 83558080 81a85bdc 835a0000 835e8968 99999999 835a393c
[   22.227352]         81a84000 81a85b60 0000003d 83558080 00020000 00007000 ffffffe0 835580a8
[   22.244010]         83558080 83558618 83558080 00000100 804b7728 835ea360 000186a0 00000000
[   22.260669]         83558080 00008000 01000100 00010001 835580a8 81a85bdc 00000000 835e8968
[   22.277326]         ...
[   22.282183] Call Trace:
[   22.287038] [<80347d38>] ioread32+0xc/0x14
[   22.295201] [<835a27b4>] ath9k_hw_setup_statusring+0x1498/0x26b4 [ath9k_hw]
[   22.309103]
[   22.312059] Code: 0000000f  8c820000  0000000f <03e00008> 00000000  0000000f  8c820000  0000000f  7c0210a0
[   22.331500]
[   22.334561] ---[ end trace 0000000000000000 ]---
[   22.343766] Kernel panic - not syncing: Fatal exception
[   22.354159] Rebooting in 1 seconds..

### OpenWrt version

r0-a10ba454

### OpenWrt release

v3.0.240924

### OpenWrt target/subtarget

ath79/generic

### Device

ChilinkIoT ZR2000 Board

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

chilinkiot, 2024-09-24:
This is my release code. My project is cloned from https://github.com/openwrt/openwrt.git， main branch.

> Invalid Release reported. `v3.0.240924` Is this from a clean repository?

DragonBluep, 2024-10-06:
This repository does not collect issues from devices that are not officially supported. And ath9k driver is maintained by the Linux upstream.