openwrt/openwrt #16478: Channel Analysis functionality stopped working on Xiaomi AX3000T in OpenWrt 23.05.5

VoidNullNil, 2024-09-24:
### Describe the bug

After upgrading to OpenWrt 23.05.5, the Channel Analysis feature on Xiaomi AX3000T no longer functions. The feature was previously working as expected in earlier versions of OpenWrt but has completely stopped functioning in this release. No channel data is shown, and the scan does not complete successfully.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Official downloaded image

### Steps to reproduce

Install or upgrade OpenWrt to version 23.05.5 on Xiaomi AX3000T.
Navigate to the Wireless configuration page.
Select "Channel Analysis" for any available wireless interface.
Observe that no scan is performed, and no channel data is displayed.

### Actual behaviour

No data is displayed, and the scan does not appear to initiate or complete.

### Additional info

base-files busybox ca-bundle dnsmasq dropbear firewall4 fstools kmod-crypto-hw-safexcel kmod-gpio-button-hotplug kmod-leds-gpio kmod-mt7915e kmod-nft-offload kmod-phy-aquantia libc libgcc libustream-mbedtls logd mtd netifd nftables odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd procd-seccomp procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd wpad-mbedtls luci adguard

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

adamast0r, 2024-10-05:
I have the same device and firmware image (OpenWrt 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-24.264.56413-c7a3562), however I can't replicate the issue here:

[image]

Both the list of wireless APs and the chart at the top on 2.4Mhz and 5Mhz show up....

VoidNullNil, 2024-10-05:
[image]

@adamast0r My firmware is custom-built, using `wpad-mbedtls` instead of the basic version. 
The list of build packages can be found in the Additional Info section.

VoidNullNil, 2024-10-06:
There are differences in the build packages compared to the first report, but the build process never finished. 🤦‍♂️(via firmware-selector.openwrt.org):
[image]
[image]

VoidNullNil, 2024-10-22:
I tested on [LuCI openwrt-23.05 branch (git-24.264.56413-c7a3562)](https://github.com/openwrt/luci) / [OpenWrt 23.05.5 (r24106-10cc5fcd00)](https://openwrt.org/) problem is fixed.
Thanks everyone for collaboration.

closed 2024-10-22 by VoidNullNil