openwrt/openwrt #16511: 23.05.5 mvebu/cortexa9 opkg install fails for some packages with "Checksum or size mismatch for package. Either the opkg or the package index are corrupt."

dev-aaront-org, 2024-09-26:
### Describe the bug

After upgrading a Linksys WRT3200ACM to 23.05.5 today, I was able to `opkg install` some packages but 3 failed:
```
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package chrony. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package chrony.
 * opkg_install_pkg: Checksum or size mismatch for package libelf1. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package ip-full.
 * opkg_install_pkg: Checksum or size mismatch for package zile. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package zile.
```

It looks like there are some problems with the package index. For example, [Packages.gz](https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.gz) has this for `chrony`:
```
Size: 124049
SHA256sum: cb4d04c29bfc291c4985d29eade777adbacab7bd25f680e13f7186bcbd666491
```

But the [ipk](https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a9_vfpv3-d16/packages/chrony_4.4-1_arm_cortex-a9_vfpv3-d16.ipk) doesn't match:
```
# ls -la chrony_4.4-1_arm_cortex-a9_vfpv3-d16.ipk
-rw-r--r-- 1 root root 124050 Sep 25 17:31 chrony_4.4-1_arm_cortex-a9_vfpv3-d16.ipk
# sha256sum chrony_4.4-1_arm_cortex-a9_vfpv3-d16.ipk
c619357c2eed472e281ea7fa0468902874b15eaf22c6f29621a49a0a6ce9b703  chrony_4.4-1_arm_cortex-a9_vfpv3-d16.ipk
```

FWIW, [Packages.manifest](https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.manifest) has the correct size and checksum. It looks like there are quite a few differences between the checksums in those files:
```
# grep -c ^SHA256sum: Packages.manifest
4405
# diff -u <(zgrep ^SHA256sum: Packages.gz) <(grep ^SHA256sum: Packages.manifest) | grep -c '^+SHA256sum:'
286
```

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT3200ACM

### Image kind

Official downloaded image

### Steps to reproduce

1. Flash 23.05.5 image.
2. `opkg update`
3. `opkg install chrony`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dev-aaront-org, 2024-10-01:
The package index has been updated since I filed this and it appears to be correct now. I am surprised that it's possible for the public packages directory to be out of sync like that, but at least it's correct now. I'll close this bug.

ynezz, 2024-10-02:
> I am surprised that it's possible for the public packages directory to be out of sync like that, but at least it's correct now.

If you look closely, you'll find out, that this is being served via Fastly CDN, so this might be some CDN POP sync issue which are hard to debug, so we need to catch more details and report that to Fastly. I'll write this down for next time.

### Getting Fastly debug headers

```shell
$ curl \
  --head \
  --header "Fastly-Debug: 1" \
  https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/chrony_4.6-r1_arm_cortex-a9_vfpv3-d16.ipk
...snip...
fastly-debug-path: (D cache-fra-etou8220147-FRA 1727845040) (F cache-fra-etou8220124-FRA 1727844929)
fastly-debug-ttl: (M cache-fra-etou8220147-FRA - - 111)
fastly-debug-digest: de1abd771c18602a2846b978b0c57a60df264c27aec0ef2795fb5d2d93873396
x-served-by: cache-fra-etou8220147-FRA
...snip...
```
### Decoding Fastly debug headers

https://www.fastly.com/documentation/reference/http/http-headers/Fastly-Debug-Path

| **Header**              | **Value**                                                                                                                     | **Description**                                                                                                  |
|-------------------------|-------------------------------------------------------------------------------------------------------------------------------|------------------------------------------------------------------------------------------------------------------|
| `fastly-debug-path`     | `(D cache-fra-etou8220147-FRA 1727845040) (F cache-fra-etou8220124-FRA 1727844929)`                                         | **D**: Decision point in the cache path.<br>**F**: Fallback cache server with timestamp identifiers.             |
| `fastly-debug-ttl`      | `(M cache-fra-etou8220147-FRA - - 111)`                                                                                     | **M**: TTL (Time To Live) metadata indicating the cached content expires in **111 seconds**.                     |
| `fastly-debug-digest`   | `de1abd771c18602a2846b978b0c57a60df264c27aec0ef2795fb5d2d93873396`                                                          | A **SHA-256 digest** representing a unique identifier for the cached content, used for cache validation.          |
| `x-served-by`           | `cache-fra-etou8220147-FRA`                                                                                                   | Specifies the **Fastly cache server** that served the request, located in **Frankfurt (FRA)**. `etou8220147`  is a unique server instance identifier.
                    |

ozangoktan, 2024-10-10:
Running into this with multiple packages now, here's the request with fastly debug info for one of them:

> curl  --head \
  --header "Fastly-Debug: 1"\
  https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/packages/tailscale_1.58.2-1_x86_64.ipk

> HTTP/2 200 
> server: nginx
> content-type: application/octet-stream
> last-modified: Thu, 10 Oct 2024 11:10:42 GMT
> etag: "6707b632-8a3733"
> access-control-allow-origin: *
> access-control-allow-methods: GET, POST, OPTIONS
> access-control-allow-headers: DNT, X-CustomHeader, Keep-Alive, User-Agent, X-Requested-With, If-Modified-Since, Cache-Control, Content-Type, Content-Range, Range
> via: 1.1 varnish, 1.1 varnish
> surrogate-key: 608204B6022BAD9B0ECB55D848B682255B6C2473D408964E7BD225768B4639A3 /releases/23.05.5/packages/x86_64/packages/tailscale_1.58.2-1_x86_64.ipk /releases/23.05.5/packages/x86_64/packages /releases/23.05.5/packages/x86_64 /releases/23.05.5/packages /releases/23.05.5 /releases /releases/23.05.5/packages/x86_64/packages/tailscale_1.58.2-1_x86_64.ipk /releases/23.05.5/packages/x86_64/packages /releases/23.05.5/packages/x86_64 /releases/23.05.5/packages /releases/23.05.5 /releases
> tmp-path: 
> accept-ranges: bytes
> age: 604
> date: Thu, 10 Oct 2024 14:55:45 GMT
> fastly-debug-path: (D cache-cph2320025-CPH 1728572145) (F cache-cph2320030-CPH 1728571542) (D cache-fra-eddf8230124-FRA 1728571542) (F cache-fra-eddf8230124-FRA 1728571542)
> fastly-debug-ttl: (M cache-cph2320025-CPH - - 604) (M cache-fra-eddf8230124-FRA - - 0)
> fastly-debug-digest: 74bc80f4f7b5cb6710fa2939caa009c0ffc4d2debc85094f48d2b8a65a500bea
> x-served-by: cache-fra-eddf8230124-FRA, cache-cph2320025-CPH
> x-cache: MISS, HIT
> x-cache-hits: 0, 0
> x-timer: S1728572145.185509,VS0,VE1
> alt-svc: h3=":443";ma=86400,h3-29=":443";ma=86400,h3-27=":443";ma=86400
> content-length: 9058099

ynezz, 2024-10-10:
> Running into this with multiple packages now, here's the request with fastly debug info for one of them:

Thanks! Can you run the same for Packages.gz please ?

spacelama, 2025-09-27:
Meh, who needs security and consistency.

````
opkg install --force-checksum curl

curl --head --header "Fastly-Debug: 1" https://downloads.openwrt.org/releases/24.10.3/packages/mips_24kc/packages/Packages.gz

HTTP/2 200 
server: nginx
content-type: application/octet-stream
last-modified: Fri, 26 Sep 2025 03:16:40 GMT
etag: "68d60598-7cbfc"
access-control-allow-origin: *
access-control-allow-methods: GET, POST, OPTIONS
access-control-allow-headers: DNT, X-CustomHeader, Keep-Alive, User-Agent, X-Requested-With, If-Modified-Since, Cache-Control, Content-Type, Content-Range, Range
via: 1.1 varnish, 1.1 varnish
surrogate-key: B5D93E501BCB466E61E80A29DFB6238F20905B1094AF074E77D581CB3211ACD3 /releases/24.10.3/packages/mips_24kc/packages/Packages.gz /releases/24.10.3/packages/mips_24kc/packages /releases/24.10.3/packages/mips_24kc /releases/24.10.3/packages /releases/24.10.3 /releases /releases/24.10.3/packages/mips_24kc/packages/Packages.gz /releases/24.10.3/packages/mips_24kc/packages /releases/24.10.3/packages/mips_24kc /releases/24.10.3/packages /releases/24.10.3 /releases
tmp-path: 
accept-ranges: bytes
age: 3830
date: Sat, 27 Sep 2025 17:23:05 GMT
fastly-debug-path: (D cache-mel11265-MEL 1758993786) (F cache-mel11226-MEL 1758985857) (D cache-fra-eddf8230120-FRA 1758985857) (F cache-fra-eddf8230120-FRA 1758984566)
fastly-debug-ttl: (M cache-mel11265-MEL - - 3830) (H cache-fra-eddf8230120-FRA 2308.668 0.000 1291)
fastly-debug-digest: d34727b00e3097660148675c81d79e41f9c4075a9d0623c46144149a12da49fc
x-served-by: cache-fra-eddf8230120-FRA, cache-mel11265-MEL
x-cache: HIT, HIT
x-cache-hits: 10, 0
x-timer: S1758993786.857486,VS0,VE1
alt-svc: h3=":443";ma=86400,h3-29=":443";ma=86400,h3-27=":443";ma=86400
content-length: 510972

curl --head --header "Fastly-Debug: 1" https://downloads.openwrt.org/releases/24.10.3/packages/mips
_24kc/packages/coreutils-base64_9.7-r1_mips_24kc.ipk

HTTP/2 200 
server: nginx
content-type: application/octet-stream
last-modified: Sat, 27 Sep 2025 12:42:32 GMT
etag: "68d7dbb8-36bd"
access-control-allow-origin: *
access-control-allow-methods: GET, POST, OPTIONS
access-control-allow-headers: DNT, X-CustomHeader, Keep-Alive, User-Agent, X-Requested-With, If-Modified-Since, Cache-Control, Content-Type, Content-Range, Range
via: 1.1 varnish, 1.1 varnish
surrogate-key: 744513DF46EE31AF7DA0826EEA6C3EBBBED2780A7E530A9BBC4B9201149768F0 /releases/24.10.3/packages/mips_24kc/packages/coreutils-base64_9.7-r1_mips_24kc.ipk /releases/24.10.3/packages/mips_24kc/packages /releases/24.10.3/packages/mips_24kc /releases/24.10.3/packages /releases/24.10.3 /releases /releases/24.10.3/packages/mips_24kc/packages/coreutils-base64_9.7-r1_mips_24kc.ipk /releases/24.10.3/packages/mips_24kc/packages /releases/24.10.3/packages/mips_24kc /releases/24.10.3/packages /releases/24.10.3 /releases
tmp-path: 
accept-ranges: bytes
date: Sat, 27 Sep 2025 17:25:40 GMT
age: 6
fastly-debug-path: (D cache-mel11255-MEL 1758993940) (F cache-mel11220-MEL 1758990212) (D cache-fra-eddf8230039-FRA 1758990212) (F cache-fra-eddf8230039-FRA 1758990212)
fastly-debug-ttl: (H cache-mel11255-MEL - - 6) (M cache-fra-eddf8230039-FRA - - 0)
fastly-debug-digest: 1c1f23dd0850e6f9db00fc50d1764f6c88178813c88ddadb71ae8afa91fb82a2
x-served-by: cache-fra-eddf8230039-FRA, cache-mel11255-MEL
x-cache: MISS, HIT
x-cache-hits: 0, 1
x-timer: S1758993940.124054,VS0,VE1
alt-svc: h3=":443";ma=86400,h3-29=":443";ma=86400,h3-27=":443";ma=86400
content-length: 14013

````