openwrt/openwrt #16515: mt7622 crash after attempting upgrade to v23.03.7

tabcat, 2024-09-26:
### Describe the bug

Attended system upgrade failed to upgrade to v23.03.7 and I believe is now booting into fail-safe mode due to kernel panic on boot.

### OpenWrt version

r19803-9a599fee93

### OpenWrt release

22.03.2

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

This happened while attempting to upgrade the system to v23.03.7

### Additional info

[Oops 1 Part1 6 0 105518.txt](https://github.com/user-attachments/files/17148872/Oops.1.Part1.6.0.105518.txt)
[Panic 2 Part1 6 0 117622.txt](https://github.com/user-attachments/files/17148870/Panic.2.Part1.6.0.117622.txt)

I've attempted system upgrades multiple times. Also attempted to reflash the device with sysupgrade cli to v23.0.3.2 which is the version displayed in failsafe mode. I thought it was v23.03.5 before attempting to upgrade to v23.03.7 initially.

Anything I can do to get it back to usable? Thanks

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

sevmonster, 2024-10-31:
This also happens to me with the same panic.

fantekah, 2024-11-22:
I've encountered the same issue on a Belkin RT3200 (UBI, installer v1.02) going from 22.03.5 to 22.03.7. I'm able to reproduce this error both with an attended sysupgrade through LuCI and normally flashing a standard 22.03.7 image through LuCI.

I tip my hat to the developers however, because at least it fails gracefully into recovery (initramfs) mode and not worse. Unfortunately, further flashing from the recovery mode seems useless until the pstore matter is discovered and then resolved in the right sequence ([link](https://github.com/dangowrt/owrt-ubi-installer/issues/136#issuecomment-1437669171)). This could be a bug or feature request by itself.

I had only wished to update my router, and 22.03.7 was the only option the attended sysupgrade presented me. I've since learned that 22.03.x is EOL, and I intend to move to 23.05.5, but I'm hesitant to do so now, lest some lurking issue causes worse trouble later.

@sevmonster  It's reassuring to hear you made it to 23.05.4. What method and image did you use? Why didn't you go to 23.05.5?

Still, it'd be nice to know the severity of whatever is leading to these errors in the pstore files. The ones I've recovered show the situation heading south in exactly the same way as tabcat:
```
<6>[ ] ramoops: using 0x10000@0x42ff0000, ecc: 0
<3>[ ] Cannot find regmap for /clock-controller@10209000: -22
<1>[ ] Unable to handle kernel paging request at virtual address fffffffffffffffe
``` 

User "daniel" on the OpenWRT forums has been eager to see pstore logs ([link](https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/640)), so perhaps he can make sense of the errors. The same f...fe error address makes one appearance in that massive thread by user "goonx", who describes the same issue ([link](https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/5971])). Unfortunately, goonx only posts twice, and it doesn't seem like anyone addressed it beyond suspecting file corruption. I think I've checked my hashes well enough to rule that out.

sevmonster, 2024-11-22:
> It's reassuring to hear that you made it to 23.05.4. What method and image did you use?

I did exactly as I said. Once I was able to boot, I cleared `pstore` so that the router would not boot into recovery, downgraded to 22.03.6 (the last working image in the series) and then upgraded to the latest 23.05.4. I used the `sysupgrade` images from the repository.

> Why didn't you go to 23.05.5?

It was not available when I was upgrading.

tech733, 2025-01-15:
I recently had this same issue going from 22.03.5 to 22.03.7 and spent a whole night trying to figure out how to fix it. This should really be documented. 
I followed these directions to fix: 

While running the production firmware enter this command in the shell:
`echo c > /proc/sysrq-trigger`
Once the router has rebooted into recovery mode, clear PSTORE and downgrade:
```
rm /sys/fs/pstore/*
sysupgrade https://downloads.openwrt.org/releases/22.03.3/targets/mediatek/mt7622/openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb
```

Do not reboot after rm /sys/fs/pstore/*, directly proceed to sysupgrade the correct file.