openwrt/openwrt #16521: No internet connectivity

potus002, 2024-09-26:
### Describe the bug

Hello,

my setup is very simple an openwrt router connected to ISP modem through a bridge with pppoe and require a vlan, the wan interface get a public ip and can ping ( ip and domain names), traceroute and lookup ok from diagnostic page in the router however there is no internet connectivity within the router when try to update opkg packages or through devices in the LAN

When Using FriendlyWRT it worked ok with no issue.

### OpenWrt version

SNAPSHOT

### OpenWrt release

r27592-676a2ab03b

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R6C

### Image kind

Official downloaded image

### Additional info

[System Log.txt](https://github.com/user-attachments/files/17157126/System.Log.txt)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

potus002, 2024-09-27:
@1715173329 I saw your reply in the forum regarding gmac driver, how to fix this? will it work I build a custom image with image builder and include a specific driver? where I can report this to upstream? thank you in advance

1715173329, 2024-09-27:
Not fix in sight.
You may write an email to `linux-rockchip@lists.infradead.org` and cc `netdev@vger.kernel.org`.

potus002, 2024-09-27:
> Not fix in sight. You may write an email to `linux-rockchip@lists.infradead.org` and cc `netdev@vger.kernel.org`.

does immortalwrt have the same issue? I thought there is a simple fix since friendlywrt work ok

1715173329, 2024-09-27:
FriendlyWrt uses bsp kernel, not mainline.