openwrt/openwrt #16531: qualcommax: nbg7815: kernel panic when rebooting

asvio, 2024-09-27:
### Describe the bug

Every time the device is rebooting or sysupgrading there is a kernel panic.

### OpenWrt version

r27596-a4c95f1444

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Zyxel NBG7815

### Image kind

Self-built image

### Steps to reproduce

Just do a reboot or sysupgrade

### Actual behaviour

Every time the device is rebooting there is a kernel panic. You only can catch it if you are connected via serial.
It's also happening when sysupgrade just after the firmware is writing on partitions.

### Expected behaviour

To have a clear reboot or sysupgrade

### Additional info

This is the log taken via serial console.

```
BusyBox v1.36.1 (2024-09-27 06:19:19 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r27596-a4c95f1444
 -----------------------------------------------------
root@NBG7815:~# reboot
root@NBG7815:~# [  319.912356] ath11k c000000.wifi phy0-ap0: left allmulticast mode
[  319.912422] ath11k c000000.wifi phy0-ap0: left promiscuous mode
[  319.917645] br-lan: port 6(phy0-ap0) entered disabled state
[  320.188147] br-home: port 2(lan2.90) entered disabled state
[  320.188294] br-home: port 1(10g.90) entered disabled state
[  320.193839] br-home: left promiscuous mode
[  320.198558] 10g.90: left allmulticast mode
[  320.202300] br-home: port 1(10g.90) entered disabled state
[  320.285669] 10g.90: left promiscuous mode
[  320.377638] br-IoT: port 4(phy1-ap1) entered disabled state
[  320.424809] ath11k c000000.wifi phy1-ap1 (unregistering): left allmulticast mode
[  320.424862] ath11k c000000.wifi phy1-ap1 (unregistering): left promiscuous mode
[  320.431298] br-IoT: port 4(phy1-ap1) entered disabled state
[  320.525222] lan2.90: left allmulticast mode
[  320.525382] br-home: port 2(lan2.90) entered disabled state
[  320.565928] lan2.90: left promiscuous mode
[  320.634592] lan3.90: left allmulticast mode
[... 389 lines cut ...]
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_TARGET_OPTIMIZATION="-O3 -pipe -mcpu=cortex-a53+crypto+crc"
CONFIG_TARGET_OPTIONS=y
CONFIG_USE_GC_SECTIONS=y
CONFIG_USE_LTO=y
CONFIG_USE_MOLD=y
CONFIG_WPA_MSG_MIN_PRIORITY=4
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

asvio, 2024-09-27:
The problem seems to have been introduced with the mac80211 update from 6.9.9 to 6.11.

I've reverted commits [https://github.com/openwrt/openwrt/commit/f03163adc5b43b008bd19ea8b3ffcf6589857e85](url) and [https://github.com/openwrt/openwrt/commit/384d079fd87620234abc5061be65050c3d54a4fe](url)  to make a new build and the problem has disappeared

pwned-pixel, 2024-09-27:
I can confirm that a reboot is resulting in the same error message/crash as in the initial post. I didn't test reverting the last two patches for mac80211 yet. But I'll do and report back.

qosmio, 2024-09-28:
I'm also seeing this issue, not only on reboots but reloading the driver.

The issue is related to the following 2 commits upstream:

wifi: ath11k: allocate dummy net_device dynamically

```diff
commit bca592ead82528b65fb028fbb71911359a642f16
Author:     Breno Leitao <leitao@debian.org>
AuthorDate: Mon Apr 22 05:39:03 2024 -0700
Commit:     David S. Miller <davem@davemloft.net>
CommitDate: Wed Apr 24 12:00:17 2024 +0100

    wifi: ath11k: allocate dummy net_device dynamically
    
    Embedding net_device into structures prohibits the usage of flexible
    arrays in the net_device structure. For more details, see the discussion
    at [1].
    
    Un-embed the net_device from struct ath11k_ext_irq_grp by converting it
    into a pointer. Then use the leverage alloc_netdev() to allocate the
    net_device object at ath11k_ahb_config_ext_irq() for ahb, and
    ath11k_pcic_ext_irq_config() for pcic.
    
    The free of the device occurs at ath11k_ahb_free_ext_irq() for the ahb
    case, and ath11k_pcic_free_ext_irq() for the pcic case.
    
    [1] https://lore.kernel.org/all/20240229225910.79e224cf@kernel.org/
    
    Signed-off-by: Breno Leitao <leitao@debian.org>
    Tested-by: Kalle Valo <kvalo@kernel.org>
    Acked-by: Kalle Valo <kvalo@kernel.org>
    Signed-off-by: David S. Miller <davem@davemloft.net>

diff --git a/drivers/net/wireless/ath/ath11k/ahb.c b/drivers/net/wireless/ath/ath11k/ahb.c
index 7c0a23517949..7f3f6479d553 100644
--- a/drivers/net/wireless/ath/ath11k/ahb.c
+++ b/drivers/net/wireless/ath/ath11k/ahb.c
@@ -442,6 +442,7 @@ static void ath11k_ahb_free_ext_irq(struct ath11k_base *ab)
 			free_irq(ab->irq_num[irq_grp->irqs[j]], irq_grp);
 
 		netif_napi_del(&irq_grp->napi);
+		free_netdev(irq_grp->napi_ndev);
 	}
 }
 
@@ -533,8 +534,12 @@ static int ath11k_ahb_config_ext_irq(struct ath11k_base *ab)
 
 		irq_grp->ab = ab;
 		irq_grp->grp_id = i;
-		init_dummy_netdev(&irq_grp->napi_ndev);
-		netif_napi_add(&irq_grp->napi_ndev, &irq_grp->napi,
+
+		irq_grp->napi_ndev = alloc_netdev_dummy(0);
+		if (!irq_grp->napi_ndev)
+			return -ENOMEM;
+
+		netif_napi_add(irq_grp->napi_ndev, &irq_grp->napi,
 			       ath11k_ahb_ext_grp_napi_poll);
 
 		for (j = 0; j < ATH11K_EXT_IRQ_NUM_MAX; j++) {
diff --git a/drivers/net/wireless/ath/ath11k/core.h b/drivers/net/wireless/ath/ath11k/core.h
index b3fb74a226fb..590307ca7a11 100644
--- a/drivers/net/wireless/ath/ath11k/core.h
+++ b/drivers/net/wireless/ath/ath11k/core.h
@@ -174,7 +174,7 @@ struct ath11k_ext_irq_grp {
 	u64 timestamp;
 	bool napi_enabled;
 	struct napi_struct napi;
-	struct net_device napi_ndev;
[... 154 lines cut ...]

Reverting those two patches resolves the issue. However, not sure it's the right approach.
Link to [patch](https://gist.githubusercontent.com/qosmio/7aba88c58bc944c453732916e4f29288/raw/5ecd69aaeb8227c0ae19694d09dc86040ef1828e/906-Revert-ath11k-allocate-dummy-net_device-and-ath11k_pcic_ext_irq_config.patch)

I haven't tested my other theory of just reverting these two lines. But also not sure if it's the right approach, as it would likely result in memory leaks for instances where the driver is reloaded vs. reboot.
```diff
--- a/drivers/net/wireless/ath/ath11k/ahb.c
+++ b/drivers/net/wireless/ath/ath11k/ahb.c
@@ -442,6 +442,7 @@ static void ath11k_ahb_free_ext_irq(struct ath11k_base *ab)
 			free_irq(ab->irq_num[irq_grp->irqs[j]], irq_grp);
 
 		netif_napi_del(&irq_grp->napi);
+		free_netdev(irq_grp->napi_ndev);
 	}
 }
--- a/drivers/net/wireless/ath/ath11k/pcic.c
+++ b/drivers/net/wireless/ath/ath11k/pcic.c
@@ -316,6 +316,7 @@ static void ath11k_pcic_free_ext_irq(struct ath11k_base *ab)
 			free_irq(ab->irq_num[irq_grp->irqs[j]], irq_grp);
 
 		netif_napi_del(&irq_grp->napi);
+		free_netdev(irq_grp->napi_ndev);
 	}
 }
```

qosmio, 2024-09-29:
Just confirming, the following patch is enough to mitigate this issue.

```diff
--- b/drivers/net/wireless/ath/ath11k/ahb.c
+++ a/drivers/net/wireless/ath/ath11k/ahb.c
@@ -442,7 +442,6 @@
 			free_irq(ab->irq_num[irq_grp->irqs[j]], irq_grp);

 		netif_napi_del(&irq_grp->napi);
-		free_netdev(irq_grp->napi_ndev);
 	}
 }

--- b/drivers/net/wireless/ath/ath11k/pcic.c
+++ a/drivers/net/wireless/ath/ath11k/pcic.c
@@ -316,7 +316,6 @@
 			free_irq(ab->irq_num[irq_grp->irqs[j]], irq_grp);

 		netif_napi_del(&irq_grp->napi);
-		free_netdev(irq_grp->napi_ndev);
 	}
 }
```

Works when rebooting, or reloading kernel mod manually.
```sh
+ service wpad stop
+ wifi down
+ rmmod ath11k_ahb
+ rmmod ath11k
+ rmmod mac80211
+ rmmod cfg80211
+ sleep 1
+ modprobe cfg80211
+ insmod mac80211
+ insmod ath11k
+ modprobe ath11k_ahb
+ service wpad start
+ sleep 2
+ wifi up
```

```
Sat Sep 28 19:53:41 2024 kern.info kernel: [  154.578805] ath11k c000000.wifi: ipq8074 hw2.0
Sat Sep 28 19:53:41 2024 kern.info kernel: [  154.578822] ath11k c000000.wifi: FW memory mode: 0
Sat Sep 28 19:53:41 2024 daemon.err procd: Could not find firmware /lib/firmware/ath11k/IPQ8074/hw2.0/firmware-2.bin: No such file or directory
Sat Sep 28 19:53:41 2024 kern.info kernel: [  154.603725] remoteproc remoteproc0: powering up cd00000.q6v5_wcss
Sat Sep 28 19:53:41 2024 kern.info kernel: [  154.603825] remoteproc remoteproc0: Booting fw image IPQ8074/q6_fw.mdt, size 668
Sat Sep 28 19:53:41 2024 kern.info kernel: [  154.948932] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
Sat Sep 28 19:53:41 2024 kern.info kernel: [  155.000234] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
Sat Sep 28 19:53:41 2024 kern.info kernel: [  155.000256] ath11k c000000.wifi: fw_version 0x2c0584a5 fw_build_timestamp 2024-07-11 16:11 fw_build_id WLAN.HK.2.12-01368-QCAHKSWPL_SILICONZ-1
Sat Sep 28 19:53:41 2024 kern.info kernel: [  155.115637] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
Sat Sep 28 19:53:41 2024 kern.info kernel: [  155.115657] remoteproc remoteproc0: powering up cd00000.q6v5_wcss
Sat Sep 28 19:53:41 2024 kern.info kernel: [  155.115746] remoteproc remoteproc0: Booting fw image IPQ8074/q6_fw.mdt, size 668
Sat Sep 28 19:53:41 2024 kern.info kernel: [  155.460995] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
Sat Sep 28 19:53:41 2024 kern.info kernel: [  155.512185] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
Sat Sep 28 19:53:41 2024 kern.info kernel: [  155.512209] ath11k c000000.wifi: fw_version 0x2c0584a5 fw_build_timestamp 2024-07-11 16:11 fw_build_id WLAN.HK.2.12-01368-QCAHKSWPL_SILICONZ-1
Sat Sep 28 19:53:42 2024 daemon.notice wpa_supplicant[12693]: Successfully initialized wpa_supplicant
Sat Sep 28 19:54:07 2024 daemon.notice netifd: radio0 (12485): Command failed: ubus call hostapd config_set { "phy": "phy0", "config": "", "prev_config": "/var/run/hostapd-phy0.conf" } (Request timed out)
Sat Sep 28 19:54:07 2024 daemon.notice hostapd: Set new config for phy phy1:
Sat Sep 28 19:54:07 2024 daemon.notice wpa_supplicant[12693]: Set new config for phy phy0
Sat Sep 28 19:54:07 2024 daemon.notice wpa_supplicant[12693]: Set new config for phy phy1
Sat Sep 28 19:54:07 2024 daemon.notice netifd: Wireless device 'radio0' is now down
Sat Sep 28 19:54:07 2024 daemon.notice netifd: Wireless device 'radio1' is now down
Sat Sep 28 19:54:07 2024 daemon.notice netifd: radio0 (12924): WARNING: Variable 'data' does not exist or is not an array/object
Sat Sep 28 19:54:07 2024 daemon.notice netifd: radio1 (12925): WARNING: Variable 'data' does not exist or is not an array/object
Sat Sep 28 19:54:09 2024 daemon.notice hostapd: Set new config for phy phy0:
Sat Sep 28 19:54:09 2024 daemon.notice hostapd: Set new config for phy phy1:
Sat Sep 28 19:54:09 2024 daemon.notice wpa_supplicant[12693]: Set new config for phy phy0
Sat Sep 28 19:54:09 2024 daemon.notice wpa_supplicant[12693]: Set new config for phy phy1
Sat Sep 28 19:54:09 2024 daemon.notice wpa_supplicant[12693]: Set new config for phy phy1
Sat Sep 28 19:54:09 2024 daemon.notice wpa_supplicant[12693]: Set new config for phy phy0
Sat Sep 28 19:54:09 2024 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Sat Sep 28 19:54:09 2024 daemon.notice hostapd: Restart interface for phy phy1
Sat Sep 28 19:54:09 2024 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Sat Sep 28 19:54:09 2024 daemon.notice hostapd: Restart interface for phy phy0
```

pwned-pixel, 2024-09-29:
Thank you for this fast find and fix.

robimarko, 2024-09-30:
We are missing this:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/net/core/dev.c?h=v6.11&id=f8d05679fb3faae478d604177b0c188b340371cd

After backporting that change it now works.

closed 2024-09-30 by openwrt-bot