openwrt/openwrt #16532: wifi-scripts: VHT160 capabilities are still emmited if channel width is less than 160 MHz

nachalni, 2024-09-27:
### Describe the bug

This commit(https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=80e4e2285fdf4a7b19c84532deafe2d1e690ed30) should have fixed the issue, but it is still present.
VHT160 capabilities are still set here, even though configured channel width is less than 160MHz : https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/network/config/wifi-scripts/files/lib/netifd/wireless/mac80211.sh;h=11b11c411f43585047d70bc632954dd06d2d528d;hb=HEAD#l370

### OpenWrt version

r27494-2e626ae2d2

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

MT7981 rfb

### Image kind

Official downloaded image

### Steps to reproduce

Use device HE80 mode with apple iphone(ax standard) and let signal drop to more than ~80db

### Actual behaviour

rx accoring to iwinfo xxx assoclist is stuck at very low rate

### Expected behaviour

rx rate should be much higher

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2024-09-27:
Can you show a generated hostapd config with the wireless config that has this issue? No VHT160 capabilities on any of my routers with `HE80` set:

`vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][MAX-MPDU-7991][MAX-A-MPDU-LEN-EXP7]` (chip supports 160Mhz)

`vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-2][BF-ANTENNA-2][MAX-MPDU-7991][MAX-A-MPDU-LEN-EXP7]` (chip does not support 160Mhz)

nachalni, 2024-09-30:
Checked the configs and scripts, I have not reverted local changes done to mac80211.sh when did the testing, everything is set fine with current Openwrt version, sorry

closed 2024-09-30 by nachalni