openwrt/openwrt #16537: Wireless (with multiple SSIDs) down when changing from 80Mhz to 160MHz

adamast0r, 2024-09-27:
### Describe the bug

I have a **configuration with multiple SSIDs (3) on one wireless radio**, everything works fine on 80Mhz.

When I change from 80Mhz to 160Mhz all those wireless SSIDs do not work anymore.

Both last stable release and snapshot are affected.

### OpenWrt version

OpenWrt 23.05.5 r24106-10cc5fcd00 and  SNAPSHOT (r27594-9c413397d9) 

### OpenWrt release

OpenWrt 23.05.5 / SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T (OpenWrt U-Boot layout)

### Image kind

Official downloaded image

### Steps to reproduce

Configure a wireless radio with multiple SSIDs on 5Ghz. The innitial configuration I have is:

```
config wifi-device 'radio2'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi'
	option channel '1'
	option band '2g'
	option htmode 'HE20'
	option cell_density '0'

config wifi-device 'radio3'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi+1'
	option channel '36'
	option band '5g'
	option htmode 'HE80'
	option cell_density '0'

config wifi-iface 'wifinet7'
	option device 'radio2'
	option mode 'ap'
	option ssid 'test'
	option encryption 'sae-mixed'
	option isolate '1'
	option key 'WWWWWWWWWWWWWWW'
	option network 'vlan13'

config wifi-iface 'wifinet8'
	option device 'radio3'
	option mode 'ap'
	option ssid 'Guest'
	option encryption 'sae-mixed'
	option key 'ZZZZZZZZZZZZZZ'
	option network 'vlan11'

config wifi-iface 'wifinet9'
	option device 'radio3'
	option mode 'ap'
	option ssid 'Work'
	option encryption 'sae-mixed'
	option isolate '1'
	option key 'YYYYYYYYYYYYYY'
	option network 'vlan12'

config wifi-iface 'wifinet10'
	option device 'radio3'
	option mode 'ap'
	option ssid 'IOT'
	option encryption 'sae-mixed'
	option key 'XXXXXXXXXXXXXX'
	option network 'vlan10'
```
 
Change in the Luci configuration for the wireless from 80Ghz to 160Ghz,
 I can see it changed the configuration:

```
config wifi-device 'radio3'
        option type 'mac80211'
        option path 'platform/18000000.wifi+1'
        option channel '36'
        option band '5g'
        option htmode 'HE160'
        option cell_density '0'
```

### Actual behaviour

wireless interface with multiple SSIDs goes down when changed from 80Mhz to 160MHz, can't see them anymore.

[image]

### Expected behaviour

I would expect that the wireless SSIDs would still allow me to work/connect after that change. I have seen other people report that they were using 160Mhz with the same router.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lukasz1992, 2024-09-28:
You need to put hostapd log here

adamast0r, 2024-09-28:
> You need to put hostapd log here

`Sat Sep 28 10:43:06 2024 daemon.notice wpa_supplicant[1483]: Set new config for phy phy1
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: Restart interface for phy phy1
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: Remove interface 'phy1'
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap0: interface state ENABLED->DISABLED
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap2: AP-DISABLED
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap2: CTRL-EVENT-TERMINATING
Sat Sep 28 10:43:06 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Sep 28 10:43:06 2024 kern.warn kernel: [43220.067875] Ignoring NSS change in VHT Operating Mode Notification from XX:XX:XX:XX:XX:XX with invalid nss 2
Sat Sep 28 10:43:06 2024 kern.info kernel: [43891.226390] br-wan: port 5(phy1-ap2) entered disabled state
Sat Sep 28 10:43:06 2024 kern.info kernel: [43891.276095] device phy1-ap2 left promiscuous mode
Sat Sep 28 10:43:06 2024 kern.info kernel: [43891.281075] br-wan: port 5(phy1-ap2) entered disabled state
Sat Sep 28 10:43:06 2024 daemon.notice netifd: Network device 'phy1-ap2' link is down
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: nl80211: Failed to remove interface phy1-ap2 from bridge br-wan: No such device
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap1: AP-STA-DISCONNECTED XX:XX:XX:XX:XX:XX
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap1: AP-DISABLED
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap1: CTRL-EVENT-TERMINATING
Sat Sep 28 10:43:06 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Sep 28 10:43:06 2024 kern.info kernel: [43891.466318] br-wan: port 4(phy1-ap1) entered disabled state
Sat Sep 28 10:43:06 2024 kern.info kernel: [43891.496042] device phy1-ap1 left promiscuous mode
Sat Sep 28 10:43:06 2024 kern.info kernel: [43891.501020] br-wan: port 4(phy1-ap1) entered disabled state
Sat Sep 28 10:43:06 2024 daemon.notice netifd: Network device 'phy1-ap1' link is down
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: nl80211: Failed to remove interface phy1-ap1 from bridge br-wan: No such device
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED XX:XX:XX:XX:XX:XX
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED XX:XX:XX:XX:XX:XX
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED XX:XX:XX:XX:XX:XX
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED XX:XX:XX:XX:XX:XX
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED XX:XX:XX:XX:XX:XX
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Sat Sep 28 10:43:06 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Sep 28 10:43:06 2024 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Sat Sep 28 10:43:06 2024 kern.info kernel: [43891.578005] device phy1-ap0 left promiscuous mode
Sat Sep 28 10:43:06 2024 kern.info kernel: [43891.583106] br-wan: port 3(phy1-ap0) entered disabled state
Sat Sep 28 10:43:07 2024 daemon.notice netifd: Network device 'phy1-ap0' link is down
Sat Sep 28 10:43:07 2024 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 hw_mode=a beacon_int=100 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=50 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-3][BF-ANTENNA-3][VHT160][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=50 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu_edca_qos_info_txop_request=0 he_mu_edca_ac_be_aifsn=8 he_
Sat Sep 28 10:43:07 2024 kern.info kernel: [43892.616027] br-wan: port 3(phy1-ap0) entered blocking state
Sat Sep 28 10:43:07 2024 kern.info kernel: [43892.621898] br-wan: port 3(phy1-ap0) entered disabled state
Sat Sep 28 10:43:07 2024 kern.info kernel: [43892.628002] device phy1-ap0 entered promiscuous mode
Sat Sep 28 10:43:07 2024 kern.info kernel: [43892.633324] br-wan: port 3(phy1-ap0) entered blocking state
Sat Sep 28 10:43:07 2024 kern.info kernel: [43892.639189] br-wan: port 3(phy1-ap0) entered forwarding state
Sat Sep 28 10:43:07 2024 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->HT_SCAN
Sat Sep 28 10:43:08 2024 daemon.notice netifd: Wireless device 'radio3' is now up
Sat Sep 28 10:43:08 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sat Sep 28 10:43:08 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Sat Sep 28 10:43:08 2024 daemon.err hostapd: could not get valid channel
Sat Sep 28 10:43:08 2024 daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->DFS
[... cut ...]
Sat Sep 28 10:43:16 2024 daemon.info hostapd: phy0-ap0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: associated (aid 4)
Sat Sep 28 10:43:16 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH XX:XX:XX:XX:XX:XX
Sat Sep 28 10:43:17 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH XX:XX:XX:XX:XX:XX
Sat Sep 28 10:43:18 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH XX:XX:XX:XX:XX:XX
Sat Sep 28 10:43:19 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH XX:XX:XX:XX:XX:XX`

lukasz1992, 2024-09-28:
You are limited with your regulatory settings.

`hostapd: could not get valid channel`

adamast0r, 2024-09-28:
@lukasz1992 thanks - didn't realise that the country was "default" and that was part of the problem, working at 160Mhz now. 

I think some warnings on the UI are needed, to prevent this from happening to other people. Alternative maybe we can show that the radios can do 160Mhz, but it is disable because you have "X" country selected?

adamast0r, 2024-10-05:
should this issue be moved to Luci repo ?

Neustradamus, 2024-10-11:
To follow this ticket.

ynezz, 2024-10-13:
> To follow this ticket.

@Neustradamus there is a button `Subscribe` on the right side for exactly that purpose.