openwrt/openwrt #16548: MediaTek MT7916AN 6 GHz wrong txpower

vortexilation, 2024-09-30:
### Describe the bug

AW7916-NPD or MediaTek MT7916AN at 6 GHz reported wrong 255 dBm txpowerlist , especially in luci .

```
root@router:~# iwinfo phy1 txpowerlist
 255 dBm (-2147483648 mW)
```

Patch used is in [here](https://gist.github.com/csyuanc/f0037b2accd349dc2c85e6ff36d2c18a), with latest mt7916 firmware

### OpenWrt version

r27593-7ea086bb89

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

generic

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vortexilation, 2024-09-30:
system log : 
```
Sep 30 15:26:59 router netifd: radio1 (14679): WARNING: Variable 'data' does not exist or is not an array/object
Sep 30 15:27:00 router hostapd: Set new config for phy phy1: 
Sep 30 15:27:00 router wpa_supplicant[4295]: Set new config for phy phy1
Sep 30 15:27:00 router netifd: radio1 (14679): sh: acs_survey: bad number
Sep 30 15:27:00 router wpa_supplicant[4295]: Set new config for phy phy1
Sep 30 15:27:00 router hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Sep 30 15:27:00 router hostapd: Restart interface for phy phy1
Sep 30 15:27:00 router hostapd: Configuration file: data:
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=PA
ieee80211d=1
hw_mode=g
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 120 240
beacon_int=100
#num_global_macaddr=1
ieee80211n=1
ht_coex=0
ht_capab=[HT40+]
ieee80211ax=1
he_su_beamformer=1
he_su_beamformee=1
he_mu_beamformer=1
he_bss_color=128
he_spr_sr_control=3
he_default_pe_duration=4
he_rts_threshold=1023
he_mu_edca_qos_info_param_count=0
he_mu_edca_qos_info_q_ack=0
he_mu_edca_qos_info_queue_request=0
he_mu_edca_qos_info_txop_request=0
he_mu_edca_ac_be_aifsn=8
he_mu_edca_ac_be_aci=0
he_mu_edca_ac_be_ecwmin=9
he_mu_edca_ac_be_ecwmax=10
he_mu_edca_ac_be_timer=255
he_mu_edca_ac_bk_aifsn=15
he_mu_edca_ac_bk_aci=1
he_mu_edca_ac_bk_ecwmin=9
he_mu_edca_ac_bk_ecwmax=10
he_mu_edca_ac_bk_timer=255
he_mu_edca_ac_vi_ecwmin=5
he_mu_edca_ac_vi_ecwmax=7
he_mu_edca_ac_vi_aifsn=5
he_mu_edca_ac_vi_aci=2
he_mu_edca_ac_vi_timer=255
he_mu_edca_ac_vo_
Sep 30 15:27:00 router netifd: Wireless device 'radio1' is now up
Sep 30 15:27:00 router ModemManager[14824]: hotplug: add network interface phy1-ap0: event processed
Sep 30 15:27:00 router ModemManager[14824]: hotplug: Couldn't report kernel event: error: couldn't find the ModemManager process in the bus
Sep 30 15:27:00 router kernel: [   61.476102] br-lan: port 6(phy1-ap0) entered blocking state
Sep 30 15:27:00 router kernel: [   61.481702] br-lan: port 6(phy1-ap0) entered disabled state
Sep 30 15:27:00 router kernel: [   61.487290] mt7915e 0000:05:00.0 phy1-ap0: entered allmulticast mode
Sep 30 15:27:00 router kernel: [   61.493705] mt7915e 0000:05:00.0 phy1-ap0: entered promiscuous mode
Sep 30 15:27:00 router hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sep 30 15:27:00 router hostapd: Hardware does not support configured mode
Sep 30 15:27:00 router hostapd: phy1-ap0: IEEE 802.11 Hardware does not support configured mode (1) (hw_mode in hostapd.conf)
Sep 30 15:27:00 router hostapd: Could not select hw_mode and channel. (-2)
Sep 30 15:27:00 router hostapd: phy1-ap0: interface state COUNTRY_UPDATE->DISABLED
Sep 30 15:27:00 router hostapd: phy1-ap0: AP-DISABLED 
Sep 30 15:27:00 router hostapd: phy1-ap0: Unable to setup interface.
Sep 30 15:27:00 router hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Sep 30 15:27:00 router kernel: [   61.502189] mt7915e 0000:05:00.0 phy1-ap0: left allmulticast mode
Sep 30 15:27:00 router kernel: [   61.508318] mt7915e 0000:05:00.0 phy1-ap0: left promiscuous mode
Sep 30 15:27:00 router kernel: [   61.514337] br-lan: port 6(phy1-ap0) entered disabled state
Sep 30 15:27:00 router hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sep 30 15:27:00 router hostapd: phy1-ap0: CTRL-EVENT-TERMINATING 
Sep 30 15:27:00 router hostapd: hostapd_free_hapd_data: Interface phy1-ap0 wasn't started
Sep 30 15:27:00 router hostapd: hostapd.add_iface failed for phy phy1 ifname=phy1-ap0
```

vortexilation, 2024-10-11:
frequencies grabbed from "iw list" seems detected as 12 dBm but "no IR" : 
```
                Frequencies:
                        * 5955.0 MHz [1] (12.0 dBm) (no IR)
                        * 5975.0 MHz [5] (12.0 dBm) (no IR)
                        * 5995.0 MHz [9] (12.0 dBm) (no IR)
                        * 6015.0 MHz [13] (12.0 dBm) (no IR)
                        * 6035.0 MHz [17] (12.0 dBm) (no IR)
                        * 6055.0 MHz [21] (12.0 dBm) (no IR)
                        * 6075.0 MHz [25] (12.0 dBm) (no IR)
                        * 6095.0 MHz [29] (12.0 dBm) (no IR)
                        * 6115.0 MHz [33] (12.0 dBm) (no IR)
                        * 6135.0 MHz [37] (12.0 dBm) (no IR)
                        * 6155.0 MHz [41] (12.0 dBm) (no IR)
                        * 6175.0 MHz [45] (12.0 dBm) (no IR)
                        * 6195.0 MHz [49] (12.0 dBm) (no IR)
                        * 6215.0 MHz [53] (12.0 dBm) (no IR)
                        * 6235.0 MHz [57] (12.0 dBm) (no IR)
                        * 6255.0 MHz [61] (12.0 dBm) (no IR)
                        * 6275.0 MHz [65] (12.0 dBm) (no IR)
                        * 6295.0 MHz [69] (12.0 dBm) (no IR)
                        * 6315.0 MHz [73] (12.0 dBm) (no IR)
                        * 6335.0 MHz [77] (12.0 dBm) (no IR)
                        * 6355.0 MHz [81] (12.0 dBm) (no IR)
                        * 6375.0 MHz [85] (12.0 dBm) (no IR)
                        * 6395.0 MHz [89] (12.0 dBm) (no IR)
                        * 6415.0 MHz [93] (12.0 dBm) (no IR)
                        * 6435.0 MHz [97] (12.0 dBm) (no IR)
                        * 6455.0 MHz [101] (12.0 dBm) (no IR)
                        * 6475.0 MHz [105] (12.0 dBm) (no IR)
                        * 6495.0 MHz [109] (12.0 dBm) (no IR)
                        * 6515.0 MHz [113] (12.0 dBm) (no IR)
                        * 6535.0 MHz [117] (12.0 dBm) (no IR)
                        * 6555.0 MHz [121] (12.0 dBm) (no IR)
                        * 6575.0 MHz [125] (12.0 dBm) (no IR)
                        * 6595.0 MHz [129] (12.0 dBm) (no IR)
                        * 6615.0 MHz [133] (12.0 dBm) (no IR)
                        * 6635.0 MHz [137] (12.0 dBm) (no IR)
                        * 6655.0 MHz [141] (12.0 dBm) (no IR)
                        * 6675.0 MHz [145] (12.0 dBm) (no IR)
                        * 6695.0 MHz [149] (12.0 dBm) (no IR)
                        * 6715.0 MHz [153] (12.0 dBm) (no IR)
                        * 6735.0 MHz [157] (12.0 dBm) (no IR)
                        * 6755.0 MHz [161] (12.0 dBm) (no IR)
                        * 6775.0 MHz [165] (12.0 dBm) (no IR)
                        * 6795.0 MHz [169] (12.0 dBm) (no IR)
                        * 6815.0 MHz [173] (12.0 dBm) (no IR)
                        * 6835.0 MHz [177] (12.0 dBm) (no IR)
                        * 6855.0 MHz [181] (12.0 dBm) (no IR)
                        * 6875.0 MHz [185] (12.0 dBm) (no IR)
                        * 6895.0 MHz [189] (12.0 dBm) (no IR)
                        * 6915.0 MHz [193] (12.0 dBm) (no IR)
                        * 6935.0 MHz [197] (12.0 dBm) (no IR)
                        * 6955.0 MHz [201] (12.0 dBm) (no IR)
                        * 6975.0 MHz [205] (12.0 dBm) (no IR)
                        * 6995.0 MHz [209] (12.0 dBm) (no IR)
                        * 7015.0 MHz [213] (12.0 dBm) (no IR)
                        * 7035.0 MHz [217] (12.0 dBm) (no IR)
                        * 7055.0 MHz [221] (12.0 dBm) (no IR)
                        * 7075.0 MHz [225] (12.0 dBm) (no IR)
                        * 7095.0 MHz [229] (12.0 dBm) (no IR)
                        * 7115.0 MHz [233] (12.0 dBm) (no IR)
```

vortexilation, 2024-10-15:
[Similar issue](https://github.com/openwrt/openwrt/issues/16356) , by using manual uci command : 
```
uci set wireless.radio1.htmode='HE160'
uci set wireless.radio1.txpower='10'
uci set wireless.radio1.channel='29'
uci set wireless.radio1.band='6g'
uci commit wireless
```
The interface seems up but not connectable with intel AX210 windows 11 client.

[image]

testuser7, 2024-10-15:
I think that `txpowerlist` is not supported for 6GHz band in `iwinfo`. The same is reported for QCN9024 (6GHz).
Maybe you should report the issue directly: https://github.com/openwrt/iwinfo

vortexilation, 2024-10-15:
@testuser7 
luci needed those txpowerlist support?, I am not quite understand on how to proceed or reported it into iwinfo.

mhalano, 2025-07-31:
I got this also on 2.4 and 5GHz radios. Maybe it's related?

[image]

[image]

mhalano, 2025-07-31:
After I restarted the 6GHz radio, using LuCi, I started getting these results:
```
root@router:~# iwinfo wifi-lan6 txpowerlist
 255 dBm (2147483647 mW)
root@router:~# iwinfo phy0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
  25 dBm ( 316 mW)
  26 dBm ( 398 mW)
* 27 dBm ( 501 mW)
```
So it's clear to me that all interfaces get the maximum TX power from `phy0`, and not from what is on top, like `wifi-lan{2,5,6}`.
This is by design? If yes, maybe there is other issue related to the wrong values being get from `phy0` until the radio is reset?

testuser7, 2026-01-22:
@vortexilation Can you check if tx power is correctly reported after this commit: https://github.com/openwrt/openwrt/commit/f6429413d92890181574f0f37b57d734bb73ba96 ?

MM999-coder, 2026-03-27:
Same problem on ACer W6m on 25.12.2.