openwrt/openwrt #16551: ramips: mt7621: broken 100 Mbps support on GMAC1 WAN port

DragonBluep, 2024-09-30:
### Describe the bug

For MT7621 devices using GMAC1 on OpenWrt master branch (ASUS RT-AC57U, H3C TX1806...), when the WAN port is connected to a 100 Mbps port (tested with a 100 Mbps modem and my Gigabit RTL8168 computer eth port), it can never establish a stable connection. The log output shows it keeps link up and link down. I waited for a few minutes, but it didn't get better.
```
# Using auto speed on my computer eth port:
[  527.244998] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 1Gbps/Full - flow control rx/tx
# The link status didn't change and it won't link down automatically. Then I set computer eth port
# speed to force 100 Mbps full and re-plug the cable:
[  638.524911] mtk_soc_eth 1e100000.ethernet wan: Link is Down
# Bingo! It will link down and up every few seconds:
[  719.645033] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
[  720.684937] mtk_soc_eth 1e100000.ethernet wan: Link is Down
[  724.844992] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
[  725.884927] mtk_soc_eth 1e100000.ethernet wan: Link is Down
[  756.045021] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
[  757.084920] mtk_soc_eth 1e100000.ethernet wan: Link is Down
[  784.125028] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
[  785.164932] mtk_soc_eth 1e100000.ethernet wan: Link is Down
```
**on OpenWrt v23.05.5, it always succeed on the first attempt:**
```
[  136.431534] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
[  136.431592] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
```
cc @arinc9 @dangowrt 

### OpenWrt version

r27603-fb3d681783

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AC57U v1

### Image kind

Official downloaded image

### Additional info

1. The original report: https://github.com/openwrt/openwrt/issues/16083#issuecomment-2382246090
2. All reports based on OpenWrt default configs.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-10-01:
Any material difference in
`ethtool wan` `ethrool lan1`
`mii-tool -v .. `
when the 100mbps device is connected?

AuthorReflex, 2024-10-02:
Looks is related to https://github.com/openwrt/openwrt/issues/15348

arinc9, 2024-10-04:
I can reproduce this issue on 23.05.5 as well on Unielec U7621-06 16M.

```
[  589.063003] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control rx/tx
[  593.568639] mt7530-mdio mdio-bus:1f lan1: Link is Down
[  595.178624] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control rx/tx
...
[ 1618.669539] mt7530-mdio mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control rx/tx
[ 1649.134976] mt7530-mdio mdio-bus:1f lan2: Link is Down
[ 1650.805890] mt7530-mdio mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control rx/tx
```

It seems this issue has been there for longer and it's become more apparent with newer OpenWrt releases.

DragonBluep, 2024-10-07:
> I can reproduce this issue on 23.05.5 as well on Unielec U7621-06 16M.

Yes. I spend some time to bisect it but finally find the 5.15 kernel is also not stable with my 100 Mbps modem, haha. But it seems that 5.15 kernel is a little better than 6.6 kernel.

> It seems this issue has been there for longer and it's become more apparent with newer OpenWrt releases.
> 
> Update: I can see it on 22.03.7 as well.

Have you applied your 2 Gbps patch https://github.com/openwrt/openwrt/pull/10238 on v22.03? I can workaround this issue by reverting that patch.

BUG-reporter, 2025-04-16:
The same as for my AC-58P: https://github.com/openwrt/openwrt/issues/17711

Djfe, 2025-05-19:
does #18585 workaround your issue so it doesn't happen with said modem?

DragonBluep, 2025-05-19:
> does [#18585](https://github.com/openwrt/openwrt/pull/18585) workaround your issue so it doesn't happen with said modem?

No. It's a different issue. **A workaround is mux a LAN port (connected to GMAC0) as the WAN function.**

gyanev, 2025-07-15:
I was having exactly the same problem with my Netgear R6220 that is also using mt7621. 

Just upgared from 24.10.0 to 24.10.2 r28739 and the issue is fixed for me. 

Hope this solve your problem too.

closed 2026-10-17 by DragonBluep