openwrt/openwrt #16552: Major bug in 23.05.5  for GL.iNet MT3000 - Keeps rebooting when data is used.

happeeshopper, 2024-09-30:
### Describe the bug

Today I flashed 23.05.5 and set up wifi and wireguard, everything was working perfectly. 

I powered the router off and went about some other things, before powering it back on a couple of hours later and connecting via laptop.

It connected fine and my google homepage opened, but as soon as I try to use an data the router crashes and reboots. This happens repeatedly, over and over.

After the restart, I can connect again and access luci without issue, but, if I:

- `update lists` in the software tab, it succeeds - but then causes the device to crash and reboot (every single time)
- try to load a web page with more content than google it such as Youtube it crashes and reboots (every single time)

Any attempt to use data crashes the device and causes it to reboot.

NOTE: This also happened on the previous 23.05.5 snapshot, I hoped it would be resolved in the official release, but it is not.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT3000

### Image kind

Official downloaded image

### Steps to reproduce

I can only state what I did:

1. Clean flash of 23.05.5
2. Setup wifi & wireguard and configure firewall (Everything was working fine at this point, rebooted a few times to check)
3. Power device off and leave it, try to connect again later and it crashes and reboots every time data is consumed.

### Actual behaviour

Router reboots every time data is used

### Expected behaviour

Not to reboot

### Additional info

This happened with the 23.05.5 snapshot also - it was working fine for a day or 2, then after making no changes, it suddenly started to do the same as described above for no apparent reason.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

happeeshopper, 2024-09-30:
I logged in via ssh over ethernet on desktop, issued the following commands and immediately started using wifi via laptop so to cause a crash.

The end of each output is when a crash/reboot occurred after I started using wifi:

```
logread -f
Mon Sep 30 19:05:37 2024 daemon.err uhttpd[1706]: [info] luci: accepted login on /admin/system/admin for root from EDIT
Mon Sep 30 19:06:58 2024 daemon.info hostapd: phy1-ap0: STA EDIT IEEE 802.11: authenticated
Mon Sep 30 19:06:58 2024 daemon.info hostapd: phy1-ap0: STA EDIT IEEE 802.11: associated (aid 1)
Mon Sep 30 19:06:59 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED EDIT auth_alg=open
Mon Sep 30 19:06:59 2024 daemon.info hostapd: phy1-ap0: STA edit WPA: pairwise key handshake completed (RSN)
Mon Sep 30 19:06:59 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED EDIT
Mon Sep 30 19:07:03 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.166 EDIT
Mon Sep 30 19:07:03 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.166 EDIT Home-PC
Mon Sep 30 19:07:07 2024 daemon.info dnsmasq-dhcp[1]: DHCPINFORM(br-lan) 192.168.1.166 EDIT
Mon Sep 30 19:07:07 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.166 EDIT Home-PC
```

```
tail -f /tmp/log/messages
Mon Sep 30 20:09:49 2024 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 192.168.1.167 edit
Mon Sep 30 20:09:49 2024 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.167 edit
Mon Sep 30 20:09:49 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.167 edit
Mon Sep 30 20:09:49 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.167 edit mx
Mon Sep 30 20:09:53 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.166 edit
Mon Sep 30 20:09:53 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.166 edit Home-PC
Mon Sep 30 20:09:56 2024 daemon.err uhttpd[1715]: [info] luci: accepted login on /admin/network/wireless for root from 192.168.1.166
Mon Sep 30 20:10:47 2024 daemon.err uhttpd[1715]: [info] luci: accepted login on /admin/status/logs for root from 192.168.1.167
Mon Sep 30 20:11:21 2024 authpriv.info dropbear[3068]: Child connection from 192.168.1.167:57704
Mon Sep 30 20:11:30 2024 authpriv.notice dropbear[3068]: Pubkey auth succeeded for 'root' with ssh-rsa key SHA256:edit from 192.168.1.167:57704
Mon Sep 30 20:16:26 2024 kern.info kernel: [  420.449024] device phy1-ap0 left promiscuous mode
Mon Sep 30 20:16:26 2024 kern.info kernel: [  420.453808] br-lan: port 2(phy1-ap0) entered disabled state
Mon Sep 30 20:16:26 2024 daemon.notice wpa_supplicant[1483]: Set new config for phy phy1
Mon Sep 30 20:16:26 2024 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Mon Sep 30 20:16:26 2024 daemon.notice hostapd: Reload config for bss 'phy1-ap0' on phy 'phy1'
Mon Sep 30 20:16:26 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED edit
Mon Sep 30 20:16:27 2024 daemon.notice netifd: Network device 'phy1-ap0' link is down
Mon Sep 30 20:16:27 2024 daemon.notice netifd: Network device 'phy1-ap0' link is up
Mon Sep 30 20:16:27 2024 daemon.notice hostapd: Reloaded settings for phy phy1
Mon Sep 30 20:16:27 2024 daemon.notice netifd: Wireless device 'radio1' is now up
Mon Sep 30 20:16:27 2024 kern.info kernel: [  421.533774] br-lan: port 2(phy1-ap0) entered blocking state
Mon Sep 30 20:16:27 2024 kern.info kernel: [  421.539379] br-lan: port 2(phy1-ap0) entered disabled state
Mon Sep 30 20:16:27 2024 kern.info kernel: [  421.545111] device phy1-ap0 entered promiscuous mode
Mon Sep 30 20:16:27 2024 kern.info kernel: [  421.550210] br-lan: port 2(phy1-ap0) entered blocking state
Mon Sep 30 20:16:27 2024 kern.info kernel: [  421.555777] br-lan: port 2(phy1-ap0) entered forwarding state
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: using nameserver 10.64.0.1#53
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: using nameserver 10.64.0.1#53
Mon Sep 30 20:16:28 2024 daemon.info dnsmasq[1]: using nameserver 194.168.4.100#53
[... cut ...]
Mon Sep 30 20:16:50 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED edit auth_alg=open
Mon Sep 30 20:16:50 2024 daemon.info hostapd: phy1-ap0: STA edit WPA: pairwise key handshake completed (RSN)
Mon Sep 30 20:16:50 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED edit
Mon Sep 30 20:16:54 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.166 edit
Mon Sep 30 20:16:54 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.166 edit Home-PC
client_loop: send disconnect: Broken pipe
```

LuisMitaHL, 2024-09-30:
Probably related to #16516

brada4, 2024-10-01:
`uci: Entry not found`

means there is a syntax error in /etc/config.

happeeshopper, 2024-10-01:
> `uci: Entry not found`
> 
> means there is a syntax error in /etc/config.

I'll check and try to fix that, but considering there's a known bug that's causing system crashes I highly doubt that correcting the uci error will fix the crashes.

brada4, 2024-10-01:
It never made into 23.05 , you need to install snapshot from 1-2 weeks ago to hit the bug.
Backup config
Set up back wifi etc and check if it crashes or it is config.

happeeshopper, 2024-10-01:
> It never made into 23.05 , you need to install snapshot from 1-2 weeks ago to hit the bug. Backup config Set up back wifi etc and check if it crashes or it is config.

So you think that 1x uci error can cause the whole system to repeatedly crash? 

Even if there is a parameter set incorrectly, it shouldn't cause the whole system to repeatedly crash and reboot.

Also consider that I can use the router via ethernet indefinitely and there are zero issues or crashes, everything works 100% - so how can that be if this config parameter is such a big problem?

I haven't changed many parameters in the config anyway, so I'll fix whatever that is and check again.

brada4, 2024-10-01:
Problem will not go away if you avoid diagnosing it. Like adding your parameters few at a time on top of default config.

happeeshopper, 2024-10-01:
> Problem will not go away if you avoid diagnosing it. Like adding your parameters few at a time on top of default config.

Sure, I know that, I'm trying to diagnose it.  I did a clean flash and I havent made the changes like I did yesterday and it seems to be working well after a couple of reboots, though it was lke that yesterday too.

I'll increment any other changes over the next couple of days and see how it goes.

I'll re-open if I get any issues.