openwrt/openwrt #16557: GL.iNet A1300 switch key configured for unknown code

cmsj, 2024-10-01:
### Describe the bug

In the Device Tree for the GL.iNet, the side switch's GPIO pins are configured to map to `KEY_SETUP` (see [here](https://github.com/openwrt/openwrt/blob/main/target/linux/ipq40xx/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq4018-gl-a1300.dts#L65)), but when the `gpio_button_hotplug` module is loaded, it fails to configure that switch:

```
[394742.068931] gpio-keys keys: does not support key code:141
[394742.068983] gpio-keys: probe of keys failed with error -22
```

Looking at the code for the probe, it's looking up the mappings in [this array](https://github.com/openwrt-mirror/openwrt/blob/master/package/kernel/gpio-button-hotplug/src/gpio-button-hotplug.c#L84) which doesn't have an entry for `KEY_SETUP`.

I'm not sure if the right choice here is to add more mappings or change the device tree to something like `KEY_RFKILL`, but I suspect it would be the former, since the kernel module doesn't seem to have even all of the recommended buttons listed on the openwrt wiki: https://openwrt.org/docs/guide-user/hardware/hardware.button

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ipq40xx/generic

### Device

GL.iNet GL-A1300

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

cmsj, 2024-10-01:
In GL.iNet's firmware it's configurable to toggle things like AdGuard or VPNs: https://docs.gl-inet.com/router/en/4/interface_guide/toggle_button_settings/

robimarko, 2024-10-01:
Yeah, we dont support configurable usage of buttons, but we can change it to one of the supported usages like RFKILL etc

cmsj, 2024-10-01:
If it's of any use, I opened a PR: https://github.com/openwrt/openwrt/pull/16564

I don't currently know how to test it (and afaik it's not possible to patch the device tree at runtime?), so I could be way off.

cmsj, 2024-11-10:
Closing now the PR has merged.

closed 2024-11-10 by cmsj