openwrt/openwrt #16569: kernel: status of 6.12 version

namiltd, 2024-10-02:
### Describe the bug

### Status of Linux kernel 6.12 migration in main:
- [ ] generic: 
  - under development: https://github.com/openwrt/openwrt/pull/14751

Targets:
- [ ] airoha
- [ ] apm821xx
- [ ] archs38
- [ ] armsr
- [ ] at91
- [ ] ath79
- [ ] bcm27xx
- [ ] bcm47xx
- [ ] bcm4908
- [ ] bcm53xx
- [ ] bmips
- [ ] d1
- [ ] gemini
- [ ] imx
- [ ] ipq40xx
- [ ] ipq806x
- [ ] ixp4xx
- [ ] kirkwood
- [ ] lantiq
- [ ] layerscape
- [ ] loongarch64
- [ ] malta
- [ ] mediatek
- [ ] mpc85xx
- [ ] mvebu
- [ ] mxs
- [ ] octeon
- [ ] omap
- [ ] pistachio
- [ ] qoriq
- [ ] qualcommax
- [ ] ramips
  - under development: https://github.com/openwrt/openwrt/pull/14751
- [ ] realtek
- [ ] rockchip
- [ ] sifiveu
- [ ] siflower
- [ ] starfive
- [ ] sunxi
- [ ] tegra
- [ ] uml
- [ ] x86
- [ ] zynq

### Summary:
- &ensp;0 -✓ approved generic
- &ensp;2 - under development
- &ensp;0 - approval request for testing
- &ensp;0 - ✓ as testing
- &ensp;0 - approval request for default
- &ensp;0 - ✓ as default

Total 2 of 43

### OpenWrt version

*

### OpenWrt release

*

### OpenWrt target/subtarget

*

### Device

*

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-10-04:
@Noltari Can you pin this tag?

dfateyev, 2024-10-04:
Just curious: are we planning to rebase master to kernel-6.12? I see that `openwrt-24.XX` hasn't been branched yet.

namiltd, 2024-10-04:
There are no plans, these are just test versions.

Neustradamus, 2024-10-05:
To follow this status ticket :)

ZevAlain, 2024-10-06:
let me follow

robimarko, 2024-10-06:
> Just curious: are we planning to rebase master to kernel-6.12? I see that `openwrt-24.XX` hasn't been branched yet.

No, 6.12 for sure wont happen before branching.

lorand-horvath, 2024-10-08:
I assume kernel 6.12 builds will not fit 8 MB flash devices anymore (or will be very close to the limit) ?
We're already at around 6.5 MB with kernel 6.6.

namiltd, 2024-10-08:
Not necessarily, in the first compilation attempts the kernel takes up a similar size as for version 6.6.54.

peterwillcn, 2024-10-13:
Kernel 6.12 PREEMPT_RT for a release of 25.xx.

namiltd, 2024-10-13:
@Noltari @lynxis I think it makes more sense to pin it, even if the implementation is distant, it prevents duplicate PR.

mgz0227, 2024-10-18:
Hi, I want to know, will you adapt it for x86 or wait for others to do it?

OW87, 2024-11-18:
> Kernel 6.12 PREEMPT_RT for a release of 25.xx.

What could be the benefits on a router ? Better general performance ?

Neustradamus, 2024-12-07:
Good news: Kernel 6.12.x is LTS, can be added in 24.10?

tapper82, 2024-12-07:
@Neustradamus Hi mate sorry it would take why to long to get all the bits dun for a newer Kernel. It will be the Kernel for the next OpenWRT release thoe. As far as I know.

namiltd, 2024-12-07:
No, version 24.10 stays with 6.6.

LearZhou, 2024-12-10:
For further discussions notice.

shuangqiao999, 2024-12-21:
Usage feedback: Using ping tool, TCP mode, ping80 port, with an interval of 1 millisecond, packet loss of 2%. The minimum and maximum sending time of each packet have been reduced to a few milliseconds, with an average time of around 4 to 5 milliseconds. The delay reduction is very significant. When RT is not enabled, the average time is over 300 milliseconds, but there will be no packet loss. The interval is 10 milliseconds, packet loss is 1%, and there is no packet loss at an interval of 20 milliseconds.   kernel 6.12

graysky2, 2024-12-21:
@shuangqiao999 - what is the call to `ping` with options you used for these tests (ie `ping -foo -bar`)?

shuangqiao999, 2024-12-21:
> @shuangqiao999 - what is the call to `ping` with options you used for these tests (ie `ping -foo -bar`)?

Using Android software with only a few options

graysky2, 2024-12-21:
I cannot reproduce your tests without more details, what are you pinging and from what?  I assume your android phone to the router/firewall?  I am not seeing any option to define an interval with iputils's ping

vincejv, 2024-12-22:
@shuangqiao999 it's best to provide which software you are talking about and how you reproduced that scenario in that software so the devs can track the bug

wigyori, 2024-12-27:
I've got d1, mxs, starfive, sifiveu ready to go for 6.12 once the final 6.12 support patches are merged into trunk.

namiltd, 2024-12-29:
@wigyori Please create PRs (they can be in separate repositories) and provide links to them.

openwgcf, 2024-12-29:
After enabling NAT66, the IPv6 network speed becomes extremely slow! [IPv6 NAT](https://openwrt.org/docs/guide-user/firewall/fw3_configurations/fw3_nat#ipv6_nat)

namiltd, 2024-12-29:
@openwgcf A similar problem occurred for kernel 6.6: https://github.com/openwrt/openwrt/issues/15857, I don't know if it has been fixed.

openwgcf, 2024-12-29:
> @openwgcf A similar problem occurred for kernel 6.6: #15857, I don't know if it has been fixed.

No fix.

sppmasterspp, 2024-12-30:
Some guys have kernel 6.12 [working on qualcommax](https://forum.openwrt.org/t/qualcommax-nss-build/148529/4298?u=sppmaster).

guidosarducci, 2025-01-02:
Happy New Year, and my thanks to everyone helping push this forward. For the target checklist above, I now have the `malta` target working for `mips64el`.

This also required fixes for the mac80211 build issues I believe others are seeing:
1. https://github.com/nbd168/backports/pull/1
2. https://github.com/openwrt/openwrt/pull/17456

namiltd, 2025-03-10:
It's been 5 months now and everything is still at a standstill.
Any ideas to move forward with implementing kernel 6.12?

shuangqiao999, 2025-03-10:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

graysky2, 2025-03-15:
I built an image after pulling this PR but found that the machine would boot and then reboot by itself in a loop. Here is the last thing I see before the thing power cycles:

Machine is an [ikoolcore-r2-max](https://www.ikoolcore.com/products/ikoolcore-r2-max) (Intel N150).
[image]

namiltd, 2025-03-15:
There is no PR here so I understand that you wanted to write it here: https://github.com/openwrt/openwrt/pull/16547
I am testing on VirtualBox version 6.12.19, the system works correctly.

graysky2, 2025-03-15:
My bad, yes.

mcbridematt, 2025-03-24:
There is an armsr tree here:
https://gitlab.com/traversetech/ls1088firmware/openwrt/-/commits/openwrt-main-sr-6-12?ref_type=heads
Only armv8 has been built and tested, armv7 TODO.

There are a couple of kmods which fail to build under 6.12, I think most will get resolved by updating to their mainline/backport versions:
- All mac80211/atheros, ralink and realtek (+CT versions). Most of these are related to a change in GPIO API.
- Earlier mediatek (mt76x0, but mt7915 is OK)
- kmod-nat46
- kmod-trelay 

From feeds:
- kmod-dahdi (telephony feed)
- ovpn-dco
- xtables-addons
- batman-adv
- openswitch (looks to be a dependency issue, to follow up)
- tcpbridge (undeclared libbpf dependency, to be checked further. part of tcprelay package)

namiltd, 2025-03-24:
@mcbridematt The patch https://github.com/namiltd/openwrt/pull/60/commits/a2955248c87cb84b2ef67188b1e2bfe077c30cb5 (https://github.com/namiltd/openwrt/pull/60) contains some fixes for mac80211 please try them out.

qingtian110, 2025-03-28:
> There are a couple of kmods which fail to build under 6.12, I think most will get resolved by updating to their mainline/backport versions:
>
> * All mac80211/atheros, ralink and realtek (+CT [...]

It should be buildable.

https://kmod.cooluc.com/aarch64_generic/6.12.20~0394888f551ffc433c8b2cebf7062e2d-r1/

namiltd, 2025-05-07:
It's amazing how quickly the number of approval requests for testing for targets is growing now.

Noltari, 2025-05-07:
> It's amazing how quickly the number of approval requests for testing for targets is growing now (after 212 days of struggle with generic and x86).

That makes sense considering the amount of work some targets need, and the fact that any work done before basic new kernel support is merged is subject to need additional changes (more work).

namiltd, 2025-05-09:
It's a shame that so few people wanted to help with generic and x86. It could have been done much earlier.

castillofrancodamian, 2025-05-21:
Kernel 6.12 for [Airoha](https://github.com/hurrian/openwrt-w1700k).

castillofrancodamian, 2025-06-08:
The "Total 36 of 44" is incorrect. It's actually 35. I think you forgot to deduct the total when the archs38 goal was removed.

namiltd, 2025-06-08:
Add generic.

castillofrancodamian, 2025-06-08:
[Under development](https://github.com/tmn505/openwrt/tree/tegra-next) in Tegra.

chunkeey, 2025-06-09:
I pushed uml default 6.12 to main.

https://github.com/openwrt/openwrt/pull/18977

udanfacy22, 2025-08-12:
Something's wrong with the numbers. Under the list of targets, it says "Complete 22 of 44," but the summary numbers say 23 targets + generic use 6.12 by default.

Djfe, 2025-08-17:
ipq40xx now runs as 6.12 by default.
It can be checked off now :)

namiltd, 2025-09-06:
In about a month, the rc version of the next LTS kernel should be available, it's not much time. So I guess the next OpenWRT release will include kernel v6.18.

Djfe, 2025-09-06:
> In about a month, the rc version of the next LTS kernel should be available, it's not much time. So I guess the next OpenWRT release will include kernel v6.18.

does that really make sense now that main ist almost stable and almost all targets switched? it would add another 6 months to the release easily. I would prefer tagging in the next 4 months over attempting to switch to 6.18

not my call to make obviously, I'm just curious if waiting really makes sense

namiltd, 2025-09-06:
It was similar with the transition from 5.15 to 6.1 and then to 6.6. The transition from 6.1 to 6.6 went much more quickly.

Djfe, 2025-09-06:
Was it? Weren't most targets still on 5.15 when that was decided (I could be wrong)?
Also wasn't the current kernel 6.7/6.8 when this was decidedlast year?
I mean if it helps: sure. I won't argue against it.

Due to shorter LTS circles (2 years) it could make sense to stay closer to kernel development/as close as possible.

In that case (decided at the end of the year) choosing 6.18 would mean either waiting for December to start adapting to 6.18 or starting early when it's not a confirmed lts candidate, yet.
tagging of the release would be earliest in march and the next release would be in early summer of 2026. 1.5 years without a major release :/

or what do you think a rough estimate for a timetable of a release would be. 6.12 could be tagged in 3-4 months, no?

graysky2, 2025-09-13:
@Torbins - never gonna happen... updating a major kernel version is a major undertaking. It only makes sense to do it with a LTS release and that happens annually.

sanderpqr, 2025-09-17:
> As a user, I would prefer to get new release faster with 6.12 kernel. If I need newer kernel, I would wait for snapshots to switch. In my case I would like to experiment with >=6.16 kernel and OpenVPN 2.7, because this combo finally should support DCO. But this is just an experiment, new release is much more useful to me.

DCO works in snapshots with 6.12, see https://github.com/openwrt/packages/pull/25645

Torbins, 2025-09-19:
> You need to install the `kmod-ovpn-dco-v2` package, and, depending on your platform, you may need to disable hardware acceleration (there's more info on this in the PR).
Of course, it is installed and loaded via modprobe. It looks like interface of this module is too old compared to what OpenVPN is expecting. Maybe someone will be able to backport module from recent kernel. Given that this version will be a long term support, there is a lot of sense in backporting.

> If you still have issues, it's better to discuss it in the [DCO threat in the form](http://forum.openwrt.org/t/openvpn-dco-how-to-enable/189788).
Thanks for the link, I will continue there.

Borromini, 2025-09-28:
Looks like we can fully move to 6.12 soon. Time for a release branch? :^)

hailfinger, 2025-10-22:
6.12 is now default for stm32: https://github.com/openwrt/openwrt/pull/20468

libiunc, 2025-10-26:
The siflower kernel bump is done, but sf19 (mips24kc) needs to be tested (I don't have this board)

CHKDSK88, 2025-10-28:
I prepared alternative kernel bump for mpc85xx: #20581

It is based on #18778 but it uses `./scripts/kernel_bump.sh` instead simple copy.

efahl, 2025-11-03:
> Also, [apk-tools](https://gitlab.alpinelinux.org/alpine/apk-tools/-/tags) 3.0.0 (which the next major release would be switching to from opkg) is still in the release candidate stage upstream (currently rc7).

Well, having tracked the progress of apk 3 for almost two years now, I'm not sure the alpine devs will reach the release any time soon.  What we've got now in OpenWrt is pretty solid, as we've got a much narrower path to tread (we don't use many of the sophisticated upgrade features), so I don't think it would be any problem to just stick with our known-good version.

Ansuel, 2025-11-04:
@namiltd ipq806x is 6.12 by default

namiltd, 2025-11-12:
> [@namiltd](https://github.com/namiltd) mpc85xx was switched to 6.12: [8d679ff](https://github.com/openwrt/openwrt/commit/8d679ff5ddc4875e18f3f03790efc1fe5bde71f3)

Already updated.

[5 more replies]