openwrt/openwrt #16570: Zyxel XGS1250-12 boot is stuck on calibration

Biergartenparadoxon, 2024-10-02:
### Describe the bug

Boot of Zyxel XGS1250-12 is stuck at calibration.
This happens almost on every boot.
When stuck UART isn't usable to get a terminal, so it's basically bricked.

Very rarely it boots successfully. Couldn't find a pattern to get it to boot successfully so I restored the original firmware once i got a successfull boot.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

realtek/rtl930x

### Device

Zyxel XGS1250-12

### Image kind

Official downloaded image

### Steps to reproduce

1. Connect serial.
2. Navigate to 'Management' in the OEM web interface and click on 'Firmware upgrade' to the left.
3. Upload the OpenWrt initramfs image, and wait till the switch reboots into OpenWrt.
4. Connect to the device through serial and change the U-boot boot command:
`fw_setenv bootcmd 'rtk network on; boota'`

### Actual behaviour

Boot gets stuck at calbration:
```
[    5.778239] start_1.2.3 Foreground Calibration
[    5.791705] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.815179] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.838634] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.862096] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.885579] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
... keeps repeating
```

### Expected behaviour

Device boots successfully

### Additional info

```
U-Boot Version V1.0.0.4 (Dec 01 2022 - 15:46:54)

Board: RTL9300 CPU:800MHz LX:175MHz DDR:600MHz
DRAM:  128 MB
SPI-F: MXIC/C22018/MMIO16-1/ModeC 1x16 MB (plr_flash_info @ 83f74a30)
Loading 65536B env. variables from offset 0xe0000
Net:   Net Initialization Skipped
No ethernet found.
 0 
Enable network
RTCORE Driver Module Initialize
  IOAL init
  Hardware-profile probe  GPIO probe (unit 0): (found)
  GPIO Init
[... 165 lines cut ...]
[    5.710019] start_1.1.2 Load DFE init. value
[    5.715784] end_1.1.2
[    5.718304] start_1.1.3 disable LEQ training,enable DFE clock
[    5.730701] end_1.1.3 --
[    5.733543] start_1.1.4 offset cali setting
[    5.739192] end_1.1.4
[    5.741704] start_1.1.5 LEQ and DFE setting
[    5.753369] end_1.1.5
[    5.762902] start_1.2.1 ForegroundOffsetCal_Manual
[    5.770224] end_1.2.1
[    5.778239] start_1.2.3 Foreground Calibration
[    5.791705] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.815179] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.838634] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.862096] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.885579] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.909050] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.932518] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
[    5.955972] rtl9300_do_rx_calibration_2_3: fgcal_gray: 0, fgcal_binary 0
... keeps repeating
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2024-10-11:
To follow this ticket.

ynezz, 2024-10-13:
> To follow this ticket.

@Neustradamus there is a button `Subscribe` on the right side for exactly that purpose.

plappermaul, 2024-10-14:
Can you do me a favour and add the following debug message to drivers/net/phy/rtl83xx-phy.c:

```
  /* ##FGCAL read binary */
  fgcal_binary = rtl9300_sds_field_r(sds_num, 0x1f, 0x14, 5, 0);

  pr_info("%s: fgcal_gray: %d, fgcal_binary %d\n",
           __func__, fgcal_gray, fgcal_binary);

  offset_range = rtl9300_sds_field_r(sds_num, 0x2e, 0x15, 15, 14);

  pr_info("offset_range: %d\n", offset_range);   /* <---- ADD THIS PR_INFO */

  if (fgcal_binary > 60 || fgcal_binary < 3) {
    if (offset_range == 3) {
      pr_info("%s: Foreground Calibration result marginal!", __func__);
      break;
    } else {
      offset_range++;
      rtl9300_sds_field_w(sds_num, 0x2e, 0x15, 15, 14, offset_range);
      rtl9300_do_rx_calibration_2_2(sds_num);
    }
  } else {
    break;
  }

```

Biergartenparadoxon, 2024-10-15:
Only if there is a way to failsafe this first. AFAIK uboot is password protected on this model and I don't want to brick this device and successfull boots are veEery rare with openwrt ATM. And I don't know how to restore the device w/o a successfull boot.

I don't have a clue about kernel development. Would something like
```
void rtl9300_do_rx_calibration_2_3(int sds_num)
{
	u32 fgcal_binary, fgcal_gray;
	u32 offset_range;
        u32 retries = 0;                            // <-- This

	pr_info("start_1.2.3 Foreground Calibration\n");

	while(1) {
                if (retries > 10) break;           // <-- And this
		if (!(sds_num % 2))
			rtl930x_write_sds_phy(sds_num, 0x1f, 0x2, 0x2f);
		else
			rtl930x_write_sds_phy(sds_num -1 , 0x1f, 0x2, 0x31);
```
be safe?

Either way could take a while. I need to figure out how to build openwrt from source at all first.

SimonFischer04, 2024-10-18:
seems like some weird race condition to me.
was only able to reproduce it twice (out of like 30), but only while randomly unplugging/plugging in cables during boot.

Unfortunately both successfully attempts didn't include the pr_info. With it, I could not reproduce it at all.

@Biergartenparadoxon you could just set your bootcmd to something bogus like bootu, then on boot you always get dropped into uboot shell. 
There you can just tftpboot the image and if you trigger the bug, just replug the device and you are back in uboot.

This is how I tested as well.

Biergartenparadoxon, 2024-10-21:
Ok. That trick with the bootcmd is cool but still needs a successfull boot. But maybe hoping to get at least one might be reasonable. Maybe that switch just had a bad day^^ and it's not that rare.

Race condition might be the right track. All of the successfull boots were after clearing the partitions with the reset button. (Not including reboots from already successfull boots and not all of the boots after reseting were successfull). Rebuilding the partitions might've gave it the few extra millis it needed.

@plappermaul would be `main` + that `pr_info` be more usefull or `10cc5fcd00` + that `pr_info` for repruduceability?

Neustradamus, 2024-10-27:
@Biergartenparadoxon: Have you seen the @plappermaul PR?
- https://github.com/openwrt/openwrt/pull/16457

Neustradamus, 2024-10-27:
The @plappermaul PR has been merged, can you test?
- https://github.com/openwrt/openwrt/pull/16457

Neustradamus, 2024-11-26:
@oliv3r, @svanheule, @TobleMiner, @achterin, @plaurien, @chapichapo, @Biergartenparadoxon, @plappermaul, @SimonFischer04, @0xharshal, @mr-wewlad, @Borromini, @jonasjelonek, @lucize, @tristan-k, @pierrepinon: Can you look?

SimonFischer04, 2025-04-18:
@Neustradamus I already looked shortly after this issue came up.
But could not really reproduce it.
With all the recent fixes I think this is closed stale.