openwrt/openwrt #16599: Помогите пажалуйста |  Please help me

livxbox5, 2024-10-04:
Я не могу вернуться на столовую проверку на роутере RT-Asus RT-AX53U (AX-1800U) 
помогите пожалуйста ТГ-  https://t.me/tigor11 Дискорд - killeryt

I can't go back to the canteen check on the router RT-Asus RT-AX53U (AX-1800U)
please help TG- https://t.me/tigor11 Discord - killeryt

livxbox5, 2024-10-04:
Помогите как  обратно вернуть на сток-рекавори asus rt-ax1800u пж

dfateyev, 2024-10-04:
Have you checked the router's page: https://openwrt.org/toh/asus/rt-ax53u

livxbox5, 2024-10-06:
А возможно прошить его через ssh и командную строку???
или обязательно нужно программа ?

Is it possible to flash it via ssh and the command line???
or do you need a program ?

livxbox5, 2024-10-06:
просто почему-то программа не может найти его нажимаю на кнопки и настройка Ethernet не помогло он его не хочет находить 

it's just that for some reason the program can't find it, I press the buttons and the Ethernet setup didn't help, he doesn't want to find it.

brada4, 2024-10-08:
Documentation provided describes installation and reverting to OEM using ONLY ssh client program.
If you have constructive proposals to make wiki more readable to non-english audiences you are welcome 5o submit your own writing via forum.
This place is for bugs, ie defects needing a code change to repair.

lotigara, 2024-12-08:
> столовую проверку

Did you mean "stock firmware" ("стоковая прошивка")?

brada4, 2024-12-08:
Thats kind of derivable from google translation, it is more of a documentation formatting issue that one-line revert procedure is stirred into other paragraph.

@livxbox5 here the properly formatted version from device page, shout out if something does not translate to understandable actions.

Starting from implementation of nested partitions (from version 22.03.4), you can use the following method directly from OpenWRT:
- Rename the asus stock firmware to asus.trx or whatever you like (optional)
- Transfer the trx file to the device in /tmp using winscp or something else
- Write that file using `mtd -r write asus.trx firmware`

lotigara, 2024-12-08:
> Starting from implementation of nested partitions (from version 22.03.4), you can use the following method directly from OpenWRT:
>
> * Rename the asus stock firmware to asus.trx or whatever you like [...]

I speak Russian, so I can translate (and clarify):
> Начиная с версии OpenWrt 22.03.4, вы можете использовать следующий способ восстановления стоковой прошивки с установленной OpenWrt:
> - Переименуйте файл стоковой прошивки Asus в `asus.trx`
> - Перенесите этот файл на устройство с OpenWrt в директорию `/tmp` используя, к примеру, WinSCP
> - Запишите файл с прошивкой в ПЗУ, выполнив команду `mtd -r write /tmp/asus.trx firmware` в оболочке OpenWrt (вы можете использовать PuTTY)