openwrt/openwrt #16604: EA7500-v1 : DSA implemtetation broken for wan port

ErwanMAS, 2024-10-05:
### Describe the bug

Router is working with Openwrt 23.05.5 .

With last snaphot kernel ( 6.6.53 ) or a little older 6.6.33 , wan port does not work .

Can not get a dhcp address !

Wan seem to be part of local lan , because every external packets are coming right into br-lan .

 

### OpenWrt version

r27645-f9a0485c99

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

Linksys EA7500 V1 WiFi Router

### Image kind

Self-built image

### Steps to reproduce

install Openwrt SNAPHOT  and see that the router don't do interaction  on wan interface .

### Actual behaviour

DHCP request is send , but unable to receive DHCP reply .

if a Static address , unable to ping this address from outside . 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

castiel652, 2024-10-08:
When you did sysupgrade did you check keep settings?

ErwanMAS, 2024-10-08:
i don't have  to keep settings , because i clear all configurations with `firstboot -y && reboot` .

ErwanMAS, 2024-10-10:
@Ansuel   you can be interested by this

castiel652, 2024-10-11:
> i don't have to keep settings , because i clear all configurations with `firstboot -y && reboot` .

Can you post the boot log and /etc/config/network

ErwanMAS, 2024-10-11:
**logread  | sed 's|\([0-9a-f]\)\([0-9a-f]:[0-9a-f][0-9a-f]:[0-9a-f][0-9a-f]:[0-9a-f][0-9a-f]:[0-9a-f][0-9a-f]:[0-9a-f]\)\([0-9a-f]\)|\1X:XX:XX:XX:XX:X\3|'**
``` 
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Wed Oct  2 18:30:55 2024 kern.notice kernel: [    0.000000] Linux version 6.6.53 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 13.3.0 r27645-f9a0485c99) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Wed Oct  2 18:30:47 2024
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] OF: fdt: Machine model: Linksys EA7500 V1 WiFi Router
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x40000000..0x40ffffff (16384 KiB) nomap non-reusable nss@40000000
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x41000000..0x411fffff (2048 KiB) nomap non-reusable smem@41000000
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x41200000..0x414fffff (3072 KiB) nomap non-reusable rsvd@41200000
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] Zone ranges:
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000042000000-0x000000004fffffff]
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000]   HighMem  empty
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] Early memory node ranges
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000042000000-0x000000004fffffff]
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000042000000-0x000000004fffffff]
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] percpu: Embedded 13 pages/cpu s21396 r8192 d23660 u53248
Wed Oct  2 18:30:55 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s21396 r8192 d23660 u53248 alloc=13*4096
Wed Oct  2 18:30:55 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Wed Oct  2 18:30:55 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=14
Wed Oct  2 18:30:55 2024 kern.notice kernel: [    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8 init=/sbin/init rootfstype=squashfs ubi.mtd=14,2048 ubi.block=0,0 root=/dev/ubiblock0_0 rootwait ro
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 56840
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] Memory: 214744K/229376K available (7619K kernel code, 628K rwdata, 2056K rodata, 1024K init, 249K bss, 14632K reserved, 0K cma-reserved, 0K highmem)
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000002] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000021] Switching to timer-based delay loop, resolution 160ns
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000316] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000341] CPU: Testing write buffer coherency: ok
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.000391] pid_max: default: 32768 minimum: 301
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.008538] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.008561] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.016266] qcom_scm: convention: smc legacy
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.017797] RCU Tasks Trace: Setting shift to 1 and lim to 1 rcu_task_cb_adjust=1.
[... cut ...]
Wed Oct  2 18:30:55 2024 kern.info kernel: [    0.094281] IP idents hash table entries: 4096 (order: 3, 32768 bytes, linear)
[... 540 lines cut ...]
	option macaddr '1X:XX:XX:XX:XX:Xc'

config device
	option name 'lan4'
	option macaddr '1X:XX:XX:XX:XX:Xc'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config device
	option name 'wan'
	option macaddr '1X:XX:XX:XX:XX:Xc'

config interface 'wan'
	option device 'wan'
	option proto 'dhcp'

config interface 'wan6'
	option device 'wan'
	option proto 'dhcpv6'
```

castiel652, 2024-10-11:
try `ip -d link show dev wan`

ErwanMAS, 2024-10-11:
`ip -d link show dev wan  | sed 's|\([0-9a-f]\)\([0-9a-f]:[0-9a-f][0-9a-f]:[0-9a-f][0-9a-f]:[0-9a-f][0-9a-f]:[0-9a-f][0-9a-f]:[0-9a-f]\)\([0-9a-f]\)|\1X:XX:XX:XX:XX:X\3|'`

```
4: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether 1X:XX:XX:XX:XX:Xc brd ff:ff:ff:ff:ff:ff promiscuity 0 allmulti 0 minmtu 68 maxmtu 9000
    dsa conduit eth0 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 gso_ipv4_max_size 65536 gro_ipv4_max_size 65536 portname p1 switchid 00000000 parentbus mdio_bus parentdev 37000000.mdio-mii:10
```

ErwanMAS, 2024-10-23:
same as https://github.com/openwrt/openwrt/issues/15585

ErwanMAS, 2024-12-17:
@Ansuel    another report of the same bug on EA8500  see #17157

commit eba2fbf6 (2025-02-03): ipq806x: fix broken WAN on Linksys EAX and Asrock G10

Fix broken WAN on Linksys EAX and Asrock G10 by incrementing the WAN
interface MAC address + 1. This caused conflicting entry in the FDB
table and caused the WAN port to malfunction with the DSA conversion.

Fixes: #17157
Fixes: #15585
Fixes: #16604
Link: https://github.com/openwrt/openwrt/pull/17839
commit cd747cb3 (2025-02-03): ipq806x: fix broken WAN on Linksys EAX and Asrock G10

Fix broken WAN on Linksys EAX and Asrock G10 by incrementing the WAN
interface MAC address + 1. This caused conflicting entry in the FDB
table and caused the WAN port to malfunction with the DSA conversion.

Fixes: #17157
Fixes: #15585
Fixes: #16604
Link: https://github.com/openwrt/openwrt/pull/17839

(cherry picked from commit eba2fbf63814bbcf8c99c98177818476fca808a7)
closed 2025-02-03 by openwrt-bot