openwrt/openwrt #16608: mvebu: build error iproute2

anomeome, 2024-10-05:
### Describe the bug

Error:

`make[5]: Entering directory '/home/kc/wrtpac/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/iproute2-ipfull/iproute2-6.11.0/lib'
ccache arm-openwrt-linux-muslgnueabi-gcc -Wall -Wstrict-prototypes  -Wmissing-prototypes -Wmissing-declarations -Wold-style-definition -Wformat=2 -O2 -pipe -I../include -I../include/uapi -DRESOLVE_HOSTNAMES -DLIBDIR=\"/usr/lib\" -DNO_SHARED_LIBS -DCONF_USR_DIR=\"/usr/share/iproute2\" -DCONF_ETC_DIR=\"/etc/iproute2\" -DNETNS_RUN_DIR=\"/var/run/netns\" -DNETNS_ETC_DIR=\"/etc/netns\" -DARPDDIR=\"/var/lib/arpd\" -DCONF_COLOR=COLOR_OPT_NEVER -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -ffile-prefix-map=/home/kc/wrtpac/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/iproute2-ipfull/iproute2-6.11.0=iproute2-6.11.0 -ffunction-sections -fdata-sections -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -DPIC -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/home/kc/wrtpac/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-14.2.0_musl_eabi/usr/include -I/home/kc/wrtpac/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-14.2.0_musl_eabi/include -I/home/kc/wrtpac/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-14.2.0_musl_eabi/include/fortify   -I/home/kc/wrtpac/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/libnl-tiny  -DHAVE_SETNS -DHAVE_HANDLE_AT -DHAVE_LIBBPF  -I/home/kc/wrtpac/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -DLIBBPF_VERSION=\"1.4.5\" -DLIBBPF_DYNAMIC -DHAVE_LIBBPF_SECTION_NAME  -DHAVE_ELF -I/home/kc/wrtpac/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -DHAVE_SETNS -DHAVE_HANDLE_AT -DHAVE_LIBBPF  -I/home/kc/wrtpac/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -DLIBBPF_VERSION=\"1.4.5\" -DLIBBPF_DYNAMIC -DHAVE_LIBBPF_SECTION_NAME  -DHAVE_ELF -I/home/kc/wrtpac/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -DPIC -fpic   -c -o libgenl.o libgenl.c
In file included from ../include/libgenl.h:5,
                 from libgenl.c:12:
../include/libnetlink.h: In function 'rta_getattr_be64':
../include/libnetlink.h:280:16: error: implicit declaration of function 'htobe64' [-Wimplicit-function-declaration]
  280 |         return htobe64(rta_getattr_u64(rta));
      |                ^~~~~~~
make[5]: *** [../config.mk:43: libgenl.o] Error 1
make[5]: Leaving directory '/home/kc/wrtpac/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/iproute2-ipfull/iproute2-6.11.0/lib'
make[4]: *** [Makefile:81: all] Error 2
make[4]: Leaving directory '/home/kc/wrtpac/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/iproute2-ipfull/iproute2-6.11.0'
make[3]: *** [Makefile:261: /home/kc/wrtpac/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/iproute2-ipfull/iproute2-6.11.0/.built] Error 2
make[3]: Leaving directory '/home/kc/wrtpac/source/package/network/utils/iproute2'
time: package/network/utils/iproute2/ipfull/compile#0.20#0.05#0.24
    ERROR: package/network/utils/iproute2 failed to build (build variant: ipfull).
`

### OpenWrt version

master snapshot

### OpenWrt release

master snapshot

### OpenWrt target/subtarget

mvebu

### Device

build

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-10-05:
Format your post and add device release tags and diffconfig.

lededev, 2024-10-05:
Compile error on GCC14
```
make[5]: Entering directory '/build/r/build_dir/target-x86_64_musl/linux-x86_64/iproute2-ipfull/ipr
oute2-6.11.0/lib'
x86_64-openwrt-linux-musl-gcc -Wall -Wstrict-prototypes  -Wmissing-prototypes -Wmissing-declaration
s -Wold-style-definition -Wformat=2 -O2 -pipe -I../include -I../include/uapi -DRESOLVE_HOSTNAMES -D
LIBDIR=\"/usr/lib\" -DNO_SHARED_LIBS -DCONF_USR_DIR=\"/usr/share/iproute2\" -DCONF_ETC_DIR=\"/etc/i
proute2\" -DNETNS_RUN_DIR=\"/var/run/netns\" -DNETNS_ETC_DIR=\"/etc/netns\" -DARPDDIR=\"/var/lib/ar
pd\" -DCONF_COLOR=COLOR_OPT_NEVER -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGE
FILE64_SOURCE -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/build/r/build
_dir/target-x86_64_musl/linux-x86_64/iproute2-ipfull/iproute2-6.11.0=iproute2-6.11.0 -ffunction-sec
tions -fdata-sections -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-prot
ector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/build/r/staging_dir/toolchain-x86_64_gcc-14
.2.0_musl/usr/include -I/build/r/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/include -I/build/r/st
aging_dir/toolchain-x86_64_gcc-14.2.0_musl/include/fortify   -I/build/r/staging_dir/target-x86_64_m
usl/usr/include/libnl-tiny  -DHAVE_SETNS -DHAVE_HANDLE_AT -DHAVE_LIBBPF  -I/build/r/staging_dir/tar
get-x86_64_musl/usr/include -DLIBBPF_VERSION=\"1.4.5\" -DLIBBPF_DYNAMIC -DHAVE_LIBBPF_SECTION_NAME
 -DHAVE_ELF -I/build/r/staging_dir/target-x86_64_musl/usr/include -DHAVE_SETNS -DHAVE_HANDLE_AT -DH
AVE_LIBBPF  -I/build/r/staging_dir/target-x86_64_musl/usr/include -DLIBBPF_VERSION=\"1.4.5\" -DLIBB
PF_DYNAMIC -DHAVE_LIBBPF_SECTION_NAME  -DHAVE_ELF -I/build/r/staging_dir/target-x86_64_musl/usr/inc
lude -DPIC -fpic   -c -o libgenl.o libgenl.c
In file included from ../include/libgenl.h:5,
                 from libgenl.c:12:
../include/libnetlink.h: In function 'rta_getattr_be64':
../include/libnetlink.h:280:16: error: implicit declaration of function 'htobe64' [-Wimplicit-funct
ion-declaration]
  280 |         return htobe64(rta_getattr_u64(rta));
      |                ^~~~~~~
make[5]: *** [../config.mk:43: libgenl.o] Error 1
make[5]: Leaving directory '/build/r/build_dir/target-x86_64_musl/linux-x86_64/iproute2-ipfull/ipro
ute2-6.11.0/lib'
```

anomeome, 2024-10-05:
Maybe issue is the _full version_

john-tho, 2024-10-06:
This should be fixed by 685fa051d5608bbb2888a04b6e4474faca093cca

anomeome, 2024-10-06:
resolved by [PR](https://github.com/openwrt/openwrt/pull/16609) push

closed 2024-10-06 by anomeome