openwrt/openwrt #16612: make: *** [/home/angel/openwrt/include/toplevel.mk:233: world] Error 2

Gloryandel, 2024-10-06:
### Describe the bug

make: *** [/home/angel/openwrt/include/toplevel.mk:233: world] Error 2
[image]

### OpenWrt version

openwrt-23.05.5

### OpenWrt release

openwrt-23.05.5

### OpenWrt target/subtarget

x86/x86

### Device

x86

### Image kind

Self-built image

### Steps to reproduce

first build,any build

### Actual behaviour

firtst build，Errors
run `make`
- ERROR: package/feeds/packages/ruby [host] failed to build.
run `make V=s`
- make: *** [/home/angel/openwrt/include/toplevel.mk:233: world] Error 2

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_DRIVER_11AC_SUPPORT=y
CONFIG_DRIVER_11AX_SUPPORT=y
# CONFIG_GRUB_IMAGES is not set
CONFIG_GRUB_TIMEOUT="0"
CONFIG_KERNEL_PAGE_POOL=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_LUCI_LANG_zh_Hans=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_OPTIMIZE_SPEED=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_MAC80211_DEBUGFS=y
CONFIG_PACKAGE_MAC80211_MESH=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_ca-certificates=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_chinadns-ng=y
CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_coreutils-nohup=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_dnsmasq-full=y
CONFIG_PACKAGE_dnsmasq_full_auth=y
CONFIG_PACKAGE_dnsmasq_full_conntrack=y
CONFIG_PACKAGE_dnsmasq_full_dhcp=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_dnsmasq_full_dnssec=y
CONFIG_PACKAGE_dnsmasq_full_nftset=y
CONFIG_PACKAGE_dnsmasq_full_noid=y
CONFIG_PACKAGE_dnsmasq_full_tftp=y
CONFIG_PACKAGE_etherwake=y
CONFIG_PACKAGE_hostapd=y
CONFIG_PACKAGE_hostapd-common=y
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_ipset=y
CONFIG_PACKAGE_iptables-mod-ipopt=y
CONFIG_PACKAGE_iptables-nft=y
CONFIG_PACKAGE_iw=y
CONFIG_PACKAGE_iwinfo=y
CONFIG_PACKAGE_kmod-cfg80211=y
CONFIG_PACKAGE_kmod-crypto-aead=y
CONFIG_PACKAGE_kmod-crypto-ccm=y
CONFIG_PACKAGE_kmod-crypto-cmac=y
CONFIG_PACKAGE_kmod-crypto-ctr=y
CONFIG_PACKAGE_kmod-crypto-gcm=y
CONFIG_PACKAGE_kmod-crypto-gf128=y
CONFIG_PACKAGE_kmod-crypto-ghash=y
CONFIG_PACKAGE_kmod-crypto-hmac=y
CONFIG_PACKAGE_kmod-crypto-manager=y
CONFIG_PACKAGE_kmod-crypto-null=y
CONFIG_PACKAGE_kmod-crypto-rng=y
CONFIG_PACKAGE_kmod-crypto-seqiv=y
CONFIG_PACKAGE_kmod-crypto-sha512=y
CONFIG_PACKAGE_kmod-ifb=y
CONFIG_PACKAGE_kmod-inet-diag=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-ipopt=y
CONFIG_PACKAGE_kmod-ipt-ipset=y
CONFIG_PACKAGE_kmod-mac80211=y
CONFIG_PACKAGE_kmod-mt76-connac=y
CONFIG_PACKAGE_kmod-mt76-core=y
CONFIG_PACKAGE_kmod-mt7921-common=y
CONFIG_PACKAGE_kmod-mt7921-firmware=y
CONFIG_PACKAGE_kmod-mt7921e=y
CONFIG_PACKAGE_kmod-mt792x-common=y
CONFIG_PACKAGE_kmod-nf-conntrack-netlink=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nft-bridge=y
CONFIG_PACKAGE_kmod-nft-compat=y
CONFIG_PACKAGE_kmod-nft-netdev=y
CONFIG_PACKAGE_kmod-sched-cake=y
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libcap-bin=y
CONFIG_PACKAGE_libcap-bin-capsh-shell="/bin/sh"
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libgmp=y
CONFIG_PACKAGE_libipset=y
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo=y
CONFIG_PACKAGE_libiwinfo-data=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnetfilter-conntrack=y
CONFIG_PACKAGE_libnettle=y
CONFIG_PACKAGE_libnfnetlink=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libruby=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_libyaml=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-adguardhome=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-homeproxy=y
CONFIG_PACKAGE_luci-app-nft-qos=y
CONFIG_PACKAGE_luci-app-openclash=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-app-smartdns=y
CONFIG_PACKAGE_luci-app-sqm=y
CONFIG_PACKAGE_luci-app-wol=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-i18n-base-zh-cn=y
CONFIG_PACKAGE_luci-i18n-firewall-zh-cn=y
CONFIG_PACKAGE_luci-i18n-homeproxy-zh-cn=y
CONFIG_PACKAGE_luci-i18n-nft-qos-zh-cn=y
CONFIG_PACKAGE_luci-i18n-opkg-zh-cn=y
CONFIG_PACKAGE_luci-i18n-smartdns-zh-cn=y
CONFIG_PACKAGE_luci-i18n-sqm-zh-cn=y
CONFIG_PACKAGE_luci-i18n-wol-zh-cn=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_nft-qos=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ruby=y
CONFIG_PACKAGE_ruby-bigdecimal=y
CONFIG_PACKAGE_ruby-date=y
CONFIG_PACKAGE_ruby-digest=y
CONFIG_PACKAGE_ruby-enc=y
CONFIG_PACKAGE_ruby-forwardable=y
CONFIG_PACKAGE_ruby-pstore=y
CONFIG_PACKAGE_ruby-psych=y
CONFIG_PACKAGE_ruby-stringio=y
CONFIG_PACKAGE_ruby-yaml=y
CONFIG_PACKAGE_sing-box=y
CONFIG_PACKAGE_smartdns=y
CONFIG_PACKAGE_sqm-scripts=y
CONFIG_PACKAGE_tc-tiny=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_ucode-mod-nl80211=y
CONFIG_PACKAGE_ucode-mod-rtnl=y
CONFIG_PACKAGE_ucode-mod-uloop=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_v2ray-geoip=y
CONFIG_PACKAGE_v2ray-geosite=y
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_wpa-supplicant=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_SINGBOX_WITH_CLASH_API=y
CONFIG_SINGBOX_WITH_ECH=y
CONFIG_SINGBOX_WITH_GVISOR=y
CONFIG_SINGBOX_WITH_QUIC=y
CONFIG_SINGBOX_WITH_REALITY_SERVER=y
CONFIG_SINGBOX_WITH_UTLS=y
CONFIG_SINGBOX_WITH_WIREGUARD=y
CONFIG_TARGET_ROOTFS_PARTSIZE=1024
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_WPA_MBO_SUPPORT=y
CONFIG_WPA_MSG_MIN_PRIORITY=3
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Gloryandel, 2024-10-06:
Linux Debian 6.1.0-25-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.106-3 (2024-08-26) x86_64 GNU/Linux

robimarko, 2024-10-06:
Please post the actual full build error

Gloryandel, 2024-10-07:
> Please post the actual full build error

make V=s
```
install ./apps/openssl.cnf -> /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-install/etc/ssl/openssl.cnf
install ./apps/ct_log_list.cnf -> /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-install/etc/ssl/ct_log_list.cnf.dist
install ./apps/ct_log_list.cnf -> /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-install/etc/ssl/ct_log_list.cnf
make[4]: Leaving directory '/home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15'
touch /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.built
rm -rf /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl.installed /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl
mkdir -p /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl
install -d -m0755 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl/etc/ssl/certs
install -d -m0755 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl/etc/ssl/private
chmod 0700 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl/etc/ssl/private
install -d -m0755 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl/usr/lib
install -m0644 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-install/usr/lib/libcrypto.so.* /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl/usr/lib/
install -m0644 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-install/usr/lib/libssl.so.* /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl/usr/lib/
install -d -m0755 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl/usr/lib/engines-3
touch /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl.installed
mkdir -p /home/angel/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
echo '3' | cmp -s - /home/angel/openwrt/staging_dir/target-x86_64_musl/pkginfo/libopenssl.version || { echo '3' > /home/angel/openwrt/staging_dir/target-x86_64_musl/pkginfo/libopenssl.version;  }
SHELL= flock /home/angel/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/.pkgdir/libopenssl/. /home/angel/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/angel/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libopenssl_installed
removed '/home/angel/openwrt/bin/packages/x86_64/base/libopenssl3_3.0.15-1_x86_64.ipk'
mkdir -p /home/angel/openwrt/bin/targets/x86/64/packages /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/CONTROL /home/angel/openwrt/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/etc/ssl/certs
install -d -m0755 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/etc/ssl/private
chmod 0700 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/etc/ssl/private
install -d -m0755 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/usr/lib
install -m0644 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-install/usr/lib/libcrypto.so.* /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/usr/lib/
install -m0644 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-install/usr/lib/libssl.so.* /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/usr/lib/
install -d -m0755 /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/usr/lib/engines-3
find /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/angel/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/angel/openwrt/scripts/strip-kmod.sh" PATCHELF="/home/angel/openwrt/staging_dir/host/bin/patchelf" /home/angel/openwrt/scripts/rstrip.sh /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl
rstrip.sh: /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/usr/lib/libcrypto.so.3: shared object
rstrip.sh: /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/usr/lib/libssl.so.3: shared object
(cd /home/angel/openwrt/build_dir/target-x86_64_musl/openssl-3.0.15/ipkg-x86_64/libopenssl/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/angel/openwrt/bin/packages/x86_64/base
[... cut ...]
 make[3] -C feeds/packages/utils/unzip compile
 make[3] -C package/firmware/linux-firmware compile
    ERROR: package/feeds/packages/ruby [host] failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/angel/openwrt/include/toplevel.mk:233: world] Error 1
```
I am just an ordinary user, and the repository source code is currently the latest without any further changes. Make V=s and the errors prompted by make seem to be inconsistent.

robimarko, 2024-10-07:
Sadly, I cannot reproduce this locally at all and the piece of log you posted with make V=s has no errors.
Can you run `make -j1 V=s` and log it to a file which you can then post?

Gloryandel, 2024-10-08:
> Sadly, I cannot reproduce this locally at all and the piece of log you posted with make V=s has no errors. Can you run `make -j1 V=s` and log it to a file which you can then post?

Preprocessing: 
```
make targetclean
make config-clean
git pull
./scripts/feeds update -a
./scripts/feeds install -a
make download
make menuconfig，and Saved
nohup make -j1 V=s &
```
There is [all logs](https://1drv.ms/u/s!AiTUztWg3wQq00Zk3W-rqMwpsnak?e=gbYhOt).

robimarko, 2024-10-08:
Well, the thing is that with your .config ruby host builds just fine for me.

What is the host distribution you are using?

Gloryandel, 2024-10-09:
Linux Debian 6.1.0-25-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.106-3 (2024-08-26) x86_64 GNU/Linux, i3 8100t

1715173329, 2024-10-09:
openwrt/packages#24450
openwrt/packages#25052

We did have some reports in the past, but I cannot reproduce it either.

robimarko, 2024-10-09:
@Gloryandel What is the Debian version, I will try to reproduce in a container but I am not hopefull.

Gloryandel, 2024-10-10:
@robimarko @1715173329 ：When I copied the wrong source code to Debain12 LXC and executed it, I still got an error. However, I ran make distclean and recompiled successfully. This unreproducible anomaly requires more error samples, and there are many contingencies in this problem. So that's it, thanks for all the help.

closed 2024-10-10 by Gloryandel