openwrt/openwrt #16614: [Build system] - Does not honor CONFIG_SECURITY=y

graysky2, 2024-10-06:
### Describe the bug

The buildsystem seems to disable `CONFIG_SECURITY=y` which is present in `target/linux/x86/config-6.6`

### OpenWrt version

r27677

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

HC Technology.,Ltd. HCAR5000-MI

### Image kind

Self-built image

### Steps to reproduce

1. Apply the following patch
2. Build the kernel or build an image

```
--- a/target/linux/x86/config-6.6
+++ b/target/linux/x86/config-6.6
@@ -1,3 +1,13 @@
+CONFIG_INIT_ON_ALLOC_DEFAULT_ON=y
+CONFIG_LSM="landlock,lockdown,yama,loadpin,safesetid,integrity"
+CONFIG_SECURITY=y
+CONFIG_SECURITY_LANDLOCK=y
+# CONFIG_SECURITY_NETWORK is not set
+CONFIG_SECURITY_PATH=y
+CONFIG_CRASH_HOTPLUG=y
+CONFIG_CRASH_MAX_MEMORY_RANGES=8192
```

### Actual behaviour

Built kernel has `# CONFIG_SECURITY is not set` rather than `CONFIG_SECURITY=y`

### Expected behaviour

The kernel should have support for the settings in the config file

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_CUSTOM=y
# CONFIG_BINUTILS_USE_VERSION_2_42 is not set
CONFIG_BINUTILS_USE_VERSION_2_43=y
CONFIG_BINUTILS_VERSION="2.43.1"
CONFIG_BINUTILS_VERSION_2_43=y
# CONFIG_GCC_USE_VERSION_13 is not set
CONFIG_GCC_USE_VERSION_14=y
CONFIG_GCC_VERSION="14.2.0"
CONFIG_GCC_VERSION_14=y
# CONFIG_KERNEL_WERROR is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

graysky2, 2024-10-06:
Edited version to reflect a current attempt.

robimarko, 2024-10-06:
It is because OpenWrt exposes that option via our kernel config options:
https://github.com/openwrt/openwrt/blob/b62e6f5beb2a574c67ba5159f0049e54cdf25d69/config/Config-kernel.in#L1379

So, you should basically drop it from the kernel config

graysky2, 2024-10-06:
@robimarko - Many thanks!  I am now able to build with the options I expected.

closed 2024-10-06 by graysky2