openwrt/openwrt #16642: No wireless network statistics

castillofrancodamian, 2024-10-08:
### Describe the bug

Wireless network statistics and status are not working, neither in LuCI nor in terminal tools. No matter the operating mode, client or access point, there is no activity information.

### OpenWrt version

r27700-170ecbecfc

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

MERCUSYS MR70X v1

### Image kind

Official downloaded image

### Additional info

AP mode

```
root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option band '2g'
        option channel '1'
        option htmode 'HE20'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
        option band '5g'
        option channel '36'
        option htmode 'HE80'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

root@OpenWrt:~# iw dev
phy#1
        Interface phy1-ap0
                ifindex 19
                wdev 0x100000001
                addr 00:eb:d8:bc:33:2a
                ssid OpenWrt
                type AP
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 20.00 dBm
                multicast TXQ:
[... 621 lines cut ...]
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Tue Oct  8 20:35:22 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Tue Oct  8 20:35:23 2024 daemon.warn odhcpd[2600]: No default route present, overriding ra_lifetime to 0!
Tue Oct  8 20:35:34 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.110 40:8d:5c:ee:52:dd
Tue Oct  8 20:35:34 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.110 40:8d:5c:ee:52:dd pc
Tue Oct  8 20:35:35 2024 daemon.warn odhcpd[2600]: No default route present, overriding ra_lifetime to 0!
Tue Oct  8 20:37:05 2024 daemon.notice netifd: wwan (9672): udhcpc: sending renew to server 10.200.0.1
Tue Oct  8 20:37:05 2024 daemon.notice netifd: wwan (9672): udhcpc: lease of 10.204.166.22 obtained from 10.200.0.1, lease time 300
```

</p>
</details> 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dangowrt, 2024-10-08:
@nbd168 I'm also seeing this problem after the most recent mac80211 and mt76 updates.

castillofrancodamian, 2024-10-09:
> @nbd168 I'm also seeing this problem after the most recent mac80211 and mt76 updates.

I don't think it's an MT76 problem because I see the same results on an Archer C60. At least with the activity info.

```
root@archer-c60:~# cat /etc/config/wireless

config wifi-device '5g'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:00.0'
        option channel 'auto'
        option band '5g'
        option htmode 'VHT80'
        option cell_density '1'
        option country 'AR'
        option disabled '0'

config wifi-device '2g'
        option type 'mac80211'
        option path 'platform/ahb/18100000.wmac'
        option channel '11'
        option band '2g'
        option htmode 'HT20'
        option cell_density '1'
        option country 'AR'
        option disabled '0'

config wifi-iface '5g_wwan'
        option device '5g'
        option mode 'sta'
        option ssid 'TeleCentro Wifi'
        option encryption 'none'
        option ifname '5g_wwan'
        option macaddr '70:28:8B:C1:A4:43'
        option bssid '22:B8:2B:30:96:D9'
        option network 'wwan'
        option disabled '0'
        option time_advertisement '2'
        option time_zone '<-03>3'
        option wnm_sleep_mode '1'
        option wnm_sleep_mode_no_keys '1'
        option bss_transition '1'
        option proxy_arp '1'
        option ieee80211w '1'

root@archer-c60:~# iw dev
phy#0
        Interface 5g_wwan
                ifindex 16
                wdev 0x3
                addr 70:28:8b:c1:a4:43
                ssid TeleCentro Wifi
                type managed
                channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
                txpower 30.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
root@archer-c60:~# ifconfig 5g_wwan
5g_wwan   Link encap:Ethernet  HWaddr 70:28:8B:C1:A4:43  
          inet addr:10.204.85.207  Bcast:10.207.255.255  Mask:255.248.0.0
          inet6 addr: fe80::7228:8bff:fec1:a443/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

commit fa6bd065 (2024-10-09): kernel: backport upstream commit that allows leaving out .ndo_get_stats64

Fixes mac80211 backport device statistics.

Fixes: #16642
closed 2024-10-09 by openwrt-bot