openwrt/openwrt #16646: wireless: 5ghz dfs starts on a incorrect supported band?

xize, 2024-10-09:
### Describe the bug

Hello,

I noticed wifi having problems on my mt6000 after performing a upgrade.

please see the text below.

### OpenWrt version

r27786+105-2d2aba3088

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

aarch64_cortex-a53

### Device

GL.iNet GL-MT6000

### Image kind

Self-built image

### Steps to reproduce

1. upgrade firmware with a sysupgrade with the keep settings checkbox checked.
2. upon reboot I see this:

```
Wed Oct  9 12:59:44 2024 daemon.notice hostapd: phy1-ap0: DFS-CAC-COMPLETED success=1 freq=5180 ht_enabled=0 chan_offset=0 chan_width=5 cf1=5250 cf2=0 radar_detected=0
Wed Oct  9 12:59:44 2024 daemon.warn hostapd: Can't set DFS state for freq 5180 MHz
Wed Oct  9 12:59:44 2024 daemon.warn hostapd: Can't set DFS state for freq 5200 MHz
Wed Oct  9 12:59:44 2024 daemon.warn hostapd: Can't set DFS state for freq 5220 MHz
Wed Oct  9 12:59:44 2024 daemon.warn hostapd: Can't set DFS state for freq 5240 MHz
```

### Actual behaviour

5ghz does not get up, and luci shows wrong information here:

[image]

please look to the channel reported, this looks wrongly because 6.130 ghz isn't wifi 7?, I don't think mt7986 can support wifi7.

```
Interface phy1-ap0
        ifindex 36
        wdev 0x100000001
        addr ca:ad:9e:e8:e8:9b
        type AP
        wiphy 1
        txpower 23.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       0       0       0       0       0       0               0
root@MT6000:~# iw reg get
global
country NL: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
[... 90 lines cut ...]
        option iface 'default_radio1'
        option vid '90'
        option key '<snip>'

config wifi-station
        option vid '52'
        option iface 'default_radio0'
        option key '<snip>'

config wifi-vlan
        option vid '52'
        option name 'iot'
        option network 'iot'
        option iface 'default_radio0'
```

### Diffconfig

```text
CONFIG_TARGET_mediatek=y                                                                                                CONFIG_TARGET_mediatek_filogic=y                                                                                        CONFIG_TARGET_mediatek_filogic_DEVICE_glinet_gl-mt6000=y                                                                CONFIG_AIRCRACK_NG_OPENSSL=y                                                                                            CONFIG_BIND_ENABLE_DOH=y                                                                                                CONFIG_DEBUG=y                                                                                                          CONFIG_EXPERIMENTAL=y                                                                                                   # CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set                                                                           # CONFIG_KERNEL_CIFS_ACL is not set                                                                                     # CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set                                                                            # CONFIG_KERNEL_EXT4_FS_SECURITY is not set                                                                             # CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set                                                                            # CONFIG_KERNEL_F2FS_FS_SECURITY is not set                                                                             # CONFIG_KERNEL_FS_POSIX_ACL is not set                                                                                 # CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set                                                                         # CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set                                                                             # CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set                                                                           # CONFIG_KERNEL_JFFS2_FS_SECURITY is not set                                                                            # CONFIG_KERNEL_JFS_POSIX_ACL is not set                                                                                # CONFIG_KERNEL_NFS_ACL_SUPPORT is not set                                                                              # CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set                                                                          # CONFIG_KERNEL_TMPFS_POSIX_ACL is not set                                                                              # CONFIG_KERNEL_UBIFS_FS_SECURITY is not set                                                                            # CONFIG_KERNEL_WERROR is not set                                                                                       # CONFIG_KERNEL_XFS_POSIX_ACL is not set                                                                                CONFIG_LIBCURL_COOKIES=y                                                                                                CONFIG_LIBCURL_FILE=y                                                                                                   CONFIG_LIBCURL_FTP=y                                                                                                    CONFIG_LIBCURL_HTTP=y                                                                                                   CONFIG_LIBCURL_MBEDTLS=y                                                                                                CONFIG_LIBCURL_NGHTTP2=y                                                                                                CONFIG_LIBCURL_NO_SMB="!"                                                                                               CONFIG_LIBCURL_PROXY=y                                                                                                  CONFIG_LIBCURL_UNIX_SOCKETS=y                                                                                           CONFIG_OPENSSL_ENGINE=y                                                                                                 CONFIG_OPENSSL_WITH_ASM=y                                                                                               CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y                                                                                   CONFIG_OPENSSL_WITH_CMS=y                                                                                               CONFIG_OPENSSL_WITH_DEPRECATED=y                                                                                        CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y                                                                                    CONFIG_OPENSSL_WITH_IDEA=y                                                                                              CONFIG_OPENSSL_WITH_MDC2=y                                                                                              CONFIG_OPENSSL_WITH_PSK=y                                                                                               CONFIG_OPENSSL_WITH_SEED=y                                                                                              CONFIG_OPENSSL_WITH_SRP=y                                                                                               CONFIG_OPENSSL_WITH_TLS13=y                                                                                             CONFIG_OPENSSL_WITH_WHIRLPOOL=y                                                                                         CONFIG_PACKAGE_aircrack-ng=y                                                                                            CONFIG_PACKAGE_avahi-dbus-daemon=y                                                                                      CONFIG_PACKAGE_avahi-utils=y                                                                                            CONFIG_PACKAGE_banip=y                                                                                              [... cut ...]                      CONFIG_PACKAGE_xtables-nft=y                                                                                            CONFIG_PACKAGE_zlib=y                                                                                                   CONFIG_TARGET_ROOTFS_PARTSIZE=104                                                                                       # CONFIG_USE_FS_ACL_ATTR is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

xize, 2024-10-09:
> Invalid Version reported. `r27786+105-2d2aba3088` Is this from a clean repository?

not entirely clean, my repo is here https://github.com/xize/openwrt-flint2-testing only has a single patch to mt76 to enable 256 qam.

xize, 2024-10-09:
after the user qunvureze (https://forum.openwrt.org/t/gl-inet-flint-2-gl-mt6000-discussions/173524/1870) mentoided the same setting as i have ``list hostapd_options 'enable_background_radar=1'`` the wrong reported dfs seem particularly caused by this option.

after i removed this option when i rebooted the router it seem to work fine.

FZXizzbKv7jg, 2025-11-29:
Netgear RAX120 v2 driver here. Confirm, exactly the same issue. Background radar detection with a non-DFS channel selected.