openwrt/openwrt #16648: No usb port power - bcm53xx/generic, netgear R6250,R6300V2. Solved with GPIO0 -> set high.

russinnes, 2024-10-09:
### Describe the bug

USB ports have no power. Manually setting GPIO 0 high activates the power to the port and attached devices come to life. 

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

bcm53xx/generic

### Device

Netgear R6300 V2 (BCM4708), Netgear R6250 (BCM4708)

### Image kind

Official downloaded image

### Steps to reproduce

Stock config image leaves gpio 0 low. Must be activated high to power USB ports. 
Plug a non-powered device in - no bueno. Activate the GPIO output and it comes to life. 
This gpio controls power to all ports (for devices with >1 USB port). 
USB hubs (internal) are all powered and recognized on boot, this affects the physical port itself. 

### Actual behaviour

No usb power with base image

### Expected behaviour

Usb power. 

### Additional info

GPIO 0 low (stock):
`root@Lan-Router:~# lsusb`
`Bus 002 Device 001: ID 1d6b:0001 Linux 5.15.167 ohci_hcd Generic Platform OHCI controller`
`Bus 004 Device 001: ID 1d6b:0003 Linux 5.15.167 xhci-hcd xHCI Host Controller`
`Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.167 ehci_hcd EHCI Host Controller`
`Bus 003 Device 001: ID 1d6b:0002 Linux 5.15.167 xhci-hcd xHCI Host Controller`

GPIO 0 high:
`root@Lan-Router:/sys/class/gpio# echo 1 > gpio0/value`

`Wed Oct  9 11:35:59 2024 kern.info kernel: [283797.385844] usb-storage 1-2:1.0: USB Mass Storage device detected`
`Wed Oct  9 11:35:59 2024 kern.info kernel: [283797.393146] scsi host0: usb-storage 1-2:1.0`
`Wed Oct  9 11:36:00 2024 kern.notice kernel: [283798.441471] scsi 0:0:0:0: Direct-Access     Lexar    USB Flash Drive  PMAP PQ: 0 ANSI: 6`
`Wed Oct  9 11:36:01 2024 kern.notice kernel: [283799.187506] sd 0:0:0:0: [sda] 62554112 512-byte logical blocks: (32.0 GB/29.8 GiB)`
`Wed Oct  9 11:36:01 2024 kern.notice kernel: [283799.196490] sd 0:0:0:0: [sda] Write Protect is off`
`Wed Oct  9 11:36:01 2024 kern.debug kernel: [283799.201444] sd 0:0:0:0: [sda] Mode Sense: 23 00 00 00`
`Wed Oct  9 11:36:01 2024 kern.err kernel: [283799.208772] sd 0:0:0:0: [sda] No Caching mode page found`
`Wed Oct  9 11:36:01 2024 kern.err kernel: [283799.214254] sd 0:0:0:0: [sda] Assuming drive cache: write through`
`Wed Oct  9 11:36:01 2024 kern.info kernel: [283799.237775]  sda: sda1 sda2`
`Wed Oct  9 11:36:01 2024 kern.notice kernel: [283799.245653] sd 0:0:0:0: [sda] Attached SCSI removable disk`

`root@Lan-Router:~# lsusb'
'Bus 001 Device 006: ID 05dc:c75d Lexar USB Flash Drive'
'Bus 002 Device 001: ID 1d6b:0001 Linux 5.15.167 ohci_hcd Generic Platform OHCI controller'
'Bus 004 Device 001: ID 1d6b:0003 Linux 5.15.167 xhci-hcd xHCI Host Controller'
'Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.167 ehci_hcd EHCI Host Controller'
'Bus 003 Device 001: ID 1d6b:0002 Linux 5.15.167 xhci-hcd xHCI Host Controller`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2024-10-22:
@russinnes Did it work on previous versions?

Sounds like a device tree problem.
I see in the following in `linux-bcm53xx_generic/linux-*/arch/arm/boot/dts/broadcom/bcm4708-netgear-r6250.dts`:
```
&usb3 {
        vcc-gpio = <&chipcommon 0 GPIO_ACTIVE_HIGH>;
};
```
which should do the trick, but nothing like that in `bcm4708-netgear-r6300v2.dts`.

greg33331, 2025-01-14:
Hello @russinnes 
i have the same issue with "Netgear R6250 V1 (BCM4708)"
Can you describe the steps to solve the problem?
Thanks a lot!

russinnes, 2025-01-14:
> Hello @russinnes i have the same issue with "Netgear R6250 V1 (BCM4708)" Can you describe the steps to solve the problem? Thanks a lot!

Sure thing - I'm going to assume the GPIO base/addr is the same. Let me know if this doesnt work:  
In a shell:  
`cd /sys/class/gpio`  
`echo 0 > export`    
`echo out > gpio0/direction`     
`echo 1 > gpio0/value`    

if you run `dmesg` following that - the last few lines should show the usb port coming alive.

If that works, put it in a script (start with `#!/bin/sh` on the first line) and run it on boot

raenye, 2025-01-14:
This should be done automatically by the device tree. IIRC for `bcm53xx` the device trees are taken directly from the Linux kernel, so ideally this needs to be fixed upstream (but meanwhile can be done in a PR with a patchfile).

> ash: write error: Resource busy

This means you probably already have `/sys/class/gpio/gpio0`.

russinnes, 2025-01-14:
yes; missing from the upstream dts was the initial issue..