openwrt/openwrt #16661: Regression: sysupgrade fails to flash MR8300 in recent snapshots

ygi80, 2024-10-11:
### Describe the bug

Greetings,

I recently noticed that my Linksys mr8300 silently fails to flash any openwrt-sysupgrade.bin image. Be it from sysupgrade, owut, auc or Luci and whether I keep settings or not or even reset first, it doesn't matter. Sysupgrade behaves like it's flashing and reboots to the secondary system normally but did not actually flash anything.

I know the change that broke it happened late June or early Jully because the last good snapshot was running kernel 6.6.36 (sorry, I don't have a commit number to provide). Latest stable doesn't exhibit the issue either.

TL;DR: some change broke sysupgrade for the past ~3 months.

### OpenWrt version

r27716-fa6bd065dd

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys MR8300 (Dallas)

### Image kind

Official downloaded image

### Steps to reproduce

Running any snapshot version from the past 3 months, use any official mean to flash any sysupgrade.bin image.

owut upgrade | sysupgrade (-n) sysugrade-image.bin | Luci's manual flash | Luci attended-sysupgrade

All will fail silently

### Actual behaviour

MR8300 already running a snapshoot version from less than 3 months ago till current.
sysupgrade <openwrt-xxxxxxx-ipq40xx-generic-linksys_mr8300-squashfs-sysupgrade.bin>
will result in rebooting to same version it started with. No actual flashing will occur and no error will be displayed at any point. If sysupgrade -n is specified, rootfs/alt_rootfs reset will still happen even though the kernel/alt_kernel partition won't be flashed. 

### Expected behaviour

Flashing a -sysupgrade.bin file should result in running the new version after reboot.

### Additional info

sysupgrade -F <openwrt-xxxxxxx-ipq40xx-generic-linksys_mr8300-squashfs-**factory**.bin> does flash as intended.
All other official flashing methods fail silently. I suspect they all rely on the same broken function under the hood but I'm no developer.
Only snapshots from the last 3 months (after kernel 6.6.36) to current are affected. Stable 23.05.[1-5] aren't.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

badulesia, 2024-10-21:
Hi.
I have just sysupgraded from snapshots r27800 to r27836 using LuCI, keeping settings. Everything went fine.

ygi80, 2024-10-22:
Lately I had taken the habit to reboot to stable 23.05.5 on partition 1 and flash latest snapshot from there on partition 2. Last attempt I made to flash from snapshot to snapshot was from 27716, when I made the report.
Since you commented on a positive outcome, I just gave it another try, this time from snapshot 27836 to 27885. Well, I'm happy to say that it works.
My best guess would be mtd-utils: https://github.com/openwrt/openwrt/commits/main/tools/mtd-utils
It did get updated late June to 2.2.0 just around the time the issue started and yet again to 2.2.1 just after I made the bug report.

Well, I don't know how it broke or what fixed it but it is fixed nonetheless.

Cheers

closed 2024-10-22 by ygi80