openwrt/openwrt #16671: /bin/uclient-fetch cannot follow the HTTP 301 redirect

ptpt52, 2024-10-12:
### Describe the bug

```
:~# wget -O- http://x-wrt.dev/ip
Downloading 'http://x-wrt.dev/ip'
Connecting to 8.28.31.37:80
Redirected to /get/envs/REMOTE_ADDR on x-wrt.dev
Writing to stdout

Download completed (27 bytes)

:~# which wget
/usr/bin/wget
:~# readlink /usr/bin/wget 
/bin/uclient-fetch
```

### OpenWrt version

xx

### OpenWrt release

xx

### OpenWrt target/subtarget

xx

### Device

xx

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2024-10-12:
it is `Writing to stdout` but write nothing?

ptpt52, 2024-10-12:
@nbd168

ptpt52, 2024-10-12:
but `wget -O/tmp/out.txt http://x-wrt.dev/ip` works fine

ptpt52, 2024-10-12:
commit https://git.openwrt.org/?p=project/uclient.git;a=commit;h=007d945467499f43656b141171d31f5643b83a6c still works fine.

ptpt52, 2024-10-12:
normal wget works fine.
```
:~# wget -O- http://x-wrt.dev/get/envs/REMOTE_ADDR
Downloading 'http://x-wrt.dev/get/envs/REMOTE_ADDR'
Connecting to 8.28.31.17:80
Writing to stdout
REMOTE_ADDR=156.232.105.67

Download completed (27 bytes)
```

`wget -O-` to write to stdout not working if url is 301 redirect
```
:~# wget -O- http://x-wrt.dev/ip
Downloading 'http://x-wrt.dev/ip'
Connecting to 8.28.31.17:80
Redirected to /get/envs/REMOTE_ADDR on x-wrt.dev
Writing to stdout

Download completed (27 bytes)
```

brada4, 2024-10-12:
Works just fine 23.05.5
```
# opkg search `which uclient-fetch`
uclient-fetch - 2023-04-13-007d9454-1
# uclient-fetch -q -O- http://x-wrt.dev/ip
REMOTE_ADDR=123.24.67.89
```

ptpt52, 2024-10-15:
there is no issue before ea609fe486a99af6540dcc7a18f0827481d0c4b0

ptpt52, 2024-10-15:
version 2024-04-18 is good