openwrt/openwrt #16675: Ninja fails to build with Python 3.13

huaracheguarache, 2024-10-12:
### Describe the bug

Ninja (part of the toolchain) fails to build on Fedora 41 with the following error message:

```bash
make[3]: Entering directory '/var/home/michael/src/router_stuff/openwrt/tools/ninja'
cd /var/home/michael/src/router_stuff/openwrt/build_dir/host/ninja-1.11.1 && CXX="/var/home/michael/src/router_stuff/openwrt/staging_dir/host/bin/g++" CXXFLAGS="-O2 -I/var/home/michael/src/router_stuff/openwrt/staging_dir/host/include  -I/var/home/michael/src/router_stuff/openwrt/staging_dir/host/include " LDFLAGS="-L/var/home/michael/src/router_stuff/openwrt/staging_dir/host/lib " /var/home/michael/src/router_stuff/openwrt/staging_dir/host/bin/python3 configure.py --bootstrap --verbose
Traceback (most recent call last):
  File "/var/home/michael/src/router_stuff/openwrt/build_dir/host/ninja-1.11.1/configure.py", line 26, in <module>
    import pipes
ModuleNotFoundError: No module named 'pipes'
make[3]: *** [Makefile:40: /var/home/michael/src/router_stuff/openwrt/build_dir/host/ninja-1.11.1/.built] Error 1
make[3]: Leaving directory '/var/home/michael/src/router_stuff/openwrt/tools/ninja'
time: tools/ninja/compile#0.20#0.20#0.38
    ERROR: tools/ninja failed to build.
make[2]: *** [tools/Makefile:231: tools/ninja/compile] Error 1
make[2]: Leaving directory '/var/home/michael/src/router_stuff/openwrt'
make[1]: *** [tools/Makefile:227: /var/home/michael/src/router_stuff/openwrt/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/var/home/michael/src/router_stuff/openwrt'
make: *** [/var/home/michael/src/router_stuff/openwrt/include/toplevel.mk:248: world] Error 2
```

This happens because Fedora uses Python 3.13 which has [removed](https://docs.python.org/3/whatsnew/3.13.html) `pipes` as a standard library. The issue has been fixed in version 1.12.0 and above of Ninja: https://github.com/ninja-build/ninja/pull/2340

### OpenWrt version

r27146-2ce206f27d

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

Netgear Nighthawk X4S R7800

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Darin755, 2025-01-15:
I am still having this issue when I attempt to build OpenWRT v23.05. Is that just because the fixes are only in newer versions?

grosjo, 2025-01-26:
Any update on that problem ? It says closed, but this is a real issue in 23.05

Darin755, 2025-01-26:
> Any update on that problem ? It says closed, but this is a real issue in 23.05

You just need to use a older version of the package. I would just build it in a Debian container.

jeremywh7, 2025-01-30:
```
    import pipes
ModuleNotFoundError: No module named 'pipes'
```
I ran in to this building 23.05-snapshot in Arch just now, and based upon [this](https://github.com/ninja-build/ninja/pull/2340#issue-1932836910) I renamed 'pipes' to 'shlex' in both parts of:  `./build_dir/host/ninja-1.11.1/configure.py` ...and it built.  😎

brianredbeard, 2025-04-08:
FYI, this is the patch I'm using at `tools/ninja/patches/102-fix_pipes.patch`:

```
--- a/configure.py
+++ b/configure.py
@@ -23,7 +23,7 @@ from __future__ import print_function
 
 from optparse import OptionParser
 import os
-import pipes
+import shlex
 import string
 import subprocess
 import sys
@@ -264,7 +264,7 @@ n.variable('configure_args', ' '.join(co
 env_keys = set(['CXX', 'AR', 'CFLAGS', 'CXXFLAGS', 'LDFLAGS'])
 configure_env = dict((k, os.environ[k]) for k in os.environ if k in env_keys)
 if configure_env:
-    config_str = ' '.join([k + '=' + pipes.quote(configure_env[k])
+    config_str = ' '.join([k + '=' + shlex.quote(configure_env[k])
                            for k in configure_env])
     n.variable('configure_env', config_str + '$ ')
 n.newline()
```

This isn't needed once OpenWRT pulls in Ninja v1.12.1 (currently blocked in OpenWrt by opinions on JobServers).

PR #16693 (2024-10-14): tools/ninja: update to 1.12.1 with same patches
ping @hauke @robimarko 

Closes #16675 

In the interest of solving issue #16675 as simply and as fast as possible,
update ninja without changing which jobserver implementation is used.

That way, we have a cleaner history, and Rosen's PR #16424 can focus
on the jobserver patch only, giving us more time to make the right decision.

Also included are very simple and helpful Makefile edits and a patch that allows
Make to control the parallel building of Ninja itself.