openwrt/openwrt #16680: daemon.err hostapd: nl80211: kernel reports: integer out of range after att. sysupgrade

euklidoid, 2024-10-12:
### Describe the bug

After performing attendend sysupgrade to SNAPSHOT - r27722-e00b6ba0c8, I realized "daemon.err hostapd: nl80211: kernel reports: integer out of range" in systemlog. 

### OpenWrt version

r27722-e00b6ba0c8

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPi-R3 Mini

### Image kind

Official downloaded image

### Additional info

I thought hostapd were not installed, checked the packages and found hostapd not installed, installation adempt result in errors:
`Collected errors:
 * check_data_file_clashes: Package hostapd wants to install file /usr/sbin/hostapd
	But that file is already provided by package  * wpad-basic-mbedtls
 * check_data_file_clashes: Package hostapd wants to install file /usr/share/hostap/hostapd.uc
	But that file is already provided by package  * wpad-basic-mbedtls
 * opkg_install_cmd: Cannot install package hostapd.`
 so somekind of hostapd is available I guess.

`Sat Oct 12 10:14:01 2024 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=DE ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 stationary_ap=1 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=acs_survey vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-3][BF-ANTENNA-3][VHT160][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=acs_survey he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 h
`
Im not sure what  should provide, unfortunatly I dont now from witch snapshot version I was updating from.. Anything else I can provide I glad to help.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

euklidoid, 2024-10-13:
@romanovj is this the solution, really silencing an error. so error became meaningless, just my 2cent

cmonroe, 2024-10-14:
The patch @romanovj linked above was added when we had something out of sync between hostapd and mac80211 if I remember correctly. The error (however unhelpful) indicates a real issue of some type. When I last saw this, it would randomly start after some amount of uptime and fill the entire syslog in a few minutes making it hard to pinpoint the event which triggered it. The patch will suppress these messages after the first instance.

I checked all of my running nodes and have not observed this message in a long time during normal operation. I think you may need to add some debugs to see what the offending nl80211 message is.

pktpls, 2024-10-17:
I updated to latest snapshot a few days ago, since then I see this a lot. mt7621 device, 802.11s over 5 GHz, and access point as well.

Lorca-R, 2024-10-23:
OpenWrt SNAPSHOT r27788-a475171e07 can confirm have the same problem.

Imwithstupid007, 2024-10-30:
Strange. I flashed snapshot 27965 to both my router and an AP (both MT6000). The router doesn't show the issue. But the AP log is flooded with:
Wed Oct 30 17:56:15 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Wed Oct 30 17:57:16 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Wed Oct 30 17:58:16 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Wed Oct 30 17:59:16 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range

anonymous-one, 2024-11-05:
mt6000 running r28135, i am seeing this as well.

doing a `wifi down ; sleep 5 ; wifi up` resolves this. although someone above mentioned it comes back after some time, i just tried a wifi down / up so cannot confirm if it does.

```
Tue Nov  5 17:05:00 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Tue Nov  5 17:06:00 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Tue Nov  5 17:07:01 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Tue Nov  5 17:08:01 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Tue Nov  5 17:09:01 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Tue Nov  5 17:10:02 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Tue Nov  5 17:11:02 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Tue Nov  5 17:15:24 CET 2024
```

quarkysg, 2024-12-05:
I'm also seeing this shown in the logs of my GL-iNet MT6000, running my own master build, pulled from Github on 4 Dec 2024.

I've commented on my findings [here ](https://forum.openwrt.org/t/gl-inet-flint-2-gl-mt6000-discussions/173524/2134?u=quarky)in the OpenWrt forum.

hauke, 2024-12-08:
Could someone please post a wireless and network configuration which shows these `integer out of range` errors. They are happening when hostapd process configures some value over netlink, but the kernel rejects it. Please try to remove everything else from your wireless and network configuration which is different from the default and not needed to reproduce this problem.

quarkysg, 2024-12-09:
> Could someone please post a wireless and network configuration which shows these `integer out of range` errors. They are happening when hostapd process configures some value over netlink, but the [...]

I do not think it is related to wireless configurations as my MT6000 did not have this logs shown in syslog for earlier builds.  I only noticed this for recent builds, and I think it started around a build I did 16 October 2024.  I don't remember seeing these logs in the build I did on 13 September 2024, so likely something has changed in between those dates.

For my builds, I always do a pull from OpenWrt Git so it is a snapshot as of those build dates.

But if you think the config will help, I think my wireless config is as vanilla as it gets (with certain details replaced with 'xx'):

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi'
	option band '2g'
	option channel 'auto'
	list channels '6'
	list channels '1'
	list channels '11'
	option htmode 'HE20'
	option country 'xx'
	option cell_density '0'
	option he_bss_color '60'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'xx'
	option key 'xx'
	option encryption 'sae-mixed'
	option wpa_disable_eapol_key_retries '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi+1'
	option band '5g'
	option channel 'auto'
	list channels '100'
	list channels '149'
	list channels '36'
	option htmode 'HE80'
	option country 'xx'
	option cell_density '0'
	option he_bss_color '62'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'xx'
	option key 'xx'
	option encryption 'sae-mixed'
	option wpa_disable_eapol_key_retries '1'
```

f00b4r0, 2024-12-17:
Seeing this as well on mt7621/mt7915e (YunCore AX820) upgraded from 23.05.5. May or may not be related, also new are the `hostapd: phy0-ap0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: did not acknowledge authentication response` messages.

Wireless config (the file also contains sections with `option disabled '1'`, which are omitted for clarity):

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option band '2g'
	option channel '9'
	option htmode 'HE20'
	option disabled '0'
	option country 'AT'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
	option band '5g'
	option channel '44'
	option htmode 'HE20'
	option disabled '0'
	option country 'AT'

config wifi-iface 'radio0_hotspot'
	option device 'radio0'
	option mode 'ap'
	option network 'hotspot'
	option ssid 'Hotspot'
	option encryption 'none'

config wifi-iface 'radio1_hotspot'
	option device 'radio1'
	option mode 'ap'
	option network 'hotspot'
	option ssid 'Hotspot'
	option encryption 'none'
```

quarkysg, 2024-12-23:
I added debug logs in the Linux kernel netlink library and this is what I got from the logs:

```
Mon Dec 23 11:35:44 2024 daemon.err hostapd: nl80211: kernel reports: integer out of range
Mon Dec 23 11:35:44 2024 kern.warn kernel: [ 2117.240207] netlink: 'hostapd': unsigned value[1669796576] out of range[0, 20].
```

hostapd is trying to set a very large value (1669796576) via netlink while the range is only between 0 and 20.

Bug with hostapd?

quarkysg, 2024-12-31:
Another interesting observation is that on both my routers (E8450 and MT-6000) the logs are consistently shown every 60s.  When I looked at the generated hostapd configuration file, the only entry in it that has a 60s timer is the line shown below:

`bss_load_update_period=60`

which I take to be 60s interval.

I thought it was due to the bss colour collision check, and so on a whim, I changed both router's `he_bss_color` config to 68, which is greater than 63, thereby triggering a random bss colour value to be generated when I restarted the wireless interface via Luci.  I have no means of checking if bss colour is active tho.

It's been more than 22 hours since I restarted the WiFi interface, and for some reason, the `integer out of range` logs are completely gone now for both routers.

It could just a happy co-incidence.

Tho. if it is due to bss colour, it could be that recent version of hostapd bss colour code somehow is not agreeing with the mt76 driver when it detected bss colour collision and tries to switch to another colour value?

I hope the above could help narrow down the root cause identification of this issue.

quarkysg, 2024-12-31:
@romanovj , the default 128 value will get randomised by hostapd to a value between 1 - 63.

You can refer to the patch here:

https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/network/services/hostapd/patches/465-hostapd-config-support-random-BSS-color.patch;h=16962024456d49d4346e1a2150f70c63e15e706f;hb=HEAD

Without the patch I linked above, 128 will get truncated to 0 tho.  Not sure if 0 is a valid bss colour value?

quarkysg, 2024-12-31:
@romanovj , you are a genius!  The patch you linked to seems to be the solution.  I applied it and added debug messages to check.  Now it changes bss color without the "integer out of range" error.

Tho. now I see this logged, but only one line so far:

`daemon.err hostapd: Failed to set beacon parameters`

Looks like we have some progress for this issue!

dhewg, 2024-12-31:
That patch is included in #17423, so you could just test that

Neustradamus, 2024-12-31:
Hope a merging of the @rany2 PR soon:
- https://github.com/openwrt/openwrt/pull/17423

hauke, 2025-01-12:
We will not backport the hostapd update into 24.10, but I would like to backport a fix for this problem. Could someone please create a PR for this hostapd patch for the main branch.

rany2, 2025-01-12:
@hauke https://github.com/openwrt/openwrt/pull/17590

Neustradamus, 2025-01-13:
@hauke: Why it is not possible to backport in 24.10?
- https://github.com/openwrt/openwrt/pull/17423

closed 2025-01-13 by openwrt-bot

ynezz, 2025-01-13:
> @hauke: Why it is not possible to backport in 24.10?

Its too late to do such changes in -rc5 stage and having 24.10.0 around the corner.