openwrt/openwrt #16681: VLAN other than Vlan1 not being bridged/swtiched on realtek/rtl930x

plaurien, 2024-10-12:
### Describe the bug

When setting 2 ports into a different Vlan than Vlan1, (to isolate the 2 connected hosts from other switch ports), packets are not getting switched at all.
While setting OpenWRT to have an IP in that specific Vlan, OpenWRT can ping and reach each host connected to these non-Vlan1 interfaces. However, the connected hosts cannot reach each other; it seems OpenWRT is not bridging/switching packets.

Trying the exact same release on VirtualBOX is working fine while running it on a Zyxel XGS1250-12 (using realtek/rtl930x) has it failing.

Tcpdump on each port does show the arp requests from each host but OpenWRT isn't bridging them.
If the said ports are in Vlan1, then everything is working fine.

### OpenWrt version

r24106-10cc5fcd0

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

realtek/rtl930x

### Device

Zyxel XGS1250-12 Switch

### Image kind

Official downloaded image

### Steps to reproduce

Keep all ports into VLAN1 as default.
Create a Vlan30 for port2 and port8, untagged member in Vlan30 and excluded from Vlan1. The hosts connected to port2 and port8 cannot talk to each other.

### Actual behaviour

OpenWRT does not bridge the 2 ports. However, if you set an IP on OpenWRT into Vlan30 then OpenWRT can reach the hosts that are connected to port2 and port8. So OpenWRT understands very well that this Vlan30 exists and can communicate with them. You do see the MAC addresses associated to the right port # into brctl showmacs bridge. However, when doing a tcpdump on them ports and the hosts are trying to talk to each other, you just see arp requests without replies.

### Expected behaviour

OpenWRT should bridge (...or switch) the 2 ports as they are in the same Vlan and isolate them from the default Vlan1.

### Additional info

Testing the exact same configuration with OpenWRT on Virtualbox works fine.
In that case, I'm just doing      VMHost1------ LAN1  OpenWRT Vbox   LAN2 ------ VMHost2
If LAN1 and LAN2 are in a different VLAN, the VMHost1 and VMHost2 can still talk to each other.
(....I'm using 2 separated internal networks to test this, packets are really switched (or bridged) by OpenWRT in that case)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

plappermaul, 2024-10-14:
Thanks for posting that issue. I do not know if I can help. But you could do me a favour in testing this. 

Testcase 1: 
- Configure switch with a single VLAN 30
- All ports set to PVID (aka untagged).
- Does it forward traffic?

Testcase 2:
- Configure switch with a single VLAN 30
- All ports set to tagged
- Does it forward traffic?

This will help to identifiy if we have a problem with tagged/untagged or with multiple VLANs.

plaurien, 2024-10-14:
Thanks a lot for your help.

Port1 is connected to a switch in trunk mode for Vlan1 and 30; with native vlan1. (in other words, Vlan1 can send tagged *and* untagged traffic, Vlan30 tagged only). When I'm using the word "switch" in this email, I'll be talking about that hardware switch (a Cisco 3750 not to name it), not about the Zyxel embedded switch interface name coming from OpenWRT, sorry for the confusion. I can provide the IP schemes and so on if it helps clarifying things.

Testcase1:
- Replaced the Vlan1 by Vlan30 (...so not adding another vlan but just replacing 1 by 30 in the GUI) All ports untagged.
- connecting a laptop on Port8

Results: 
Laptop gets an IP from Vlan30 (which I do not expect; port1 is setup as untagged for Vlan30 but the connected switch uses Vlan1 as untagged traffic so the laptop should get an IP from the Switch's Vlan1, I'm not using any VTP server)
This said, all traffic is forwarded. When doing a tcpdump on Lan8, I see the very first arp request but then traffic is not reported. Looks like the traffic is not targetting the CPU (which I guess is expected with ASICs?).
Interestingly, the LAN8 Tx count remains flat at 0.0Kb. I did an ifconfig lan8 down and the laptop lost its traffic flow until I did an up again.

Test case1b:
- Added a Vlan1 as a new VLAN (by keeping 30 as top line in Testcase1). Basically, I'm inverting the scenario where you normally have Vlan1 as very top then add other Vlans underneath it. 
- Added Port1 as trunk
- Port8 still in Vlan30 Untagged, the laptop no longer talks to anything.
- Port8 moved to Vlan1 Untagged, the laptop gets an IP from the Swtich's Vlan1. Can't get any IP from Switch's Vlan30 anymore, no matter the Port1 settings I try.

Test case2:
- Go back to Testcase1 with only Vlan30 existing, setting all interfaces as Trunk including Lan8 towards the laptop
Result: laptop can't get any IP (....the laptop does not have trunking drivers installed, I did not verify the Lan driver's availability as it's running windows.... I can run it on Linux and try if you want me to).

I'm running a bit out of time for tonight but I can certainly run other tests if it helps. I also kept the network file from the test cases if needed (while I switch config from the GUI and use the serial to restore access as when switching VLANs, I sometimes need to flex to get IP connectivity back :) )

Thank you again for all your help!

plappermaul, 2024-10-15:
So you basically say:

- you can run the switch in a basic VLAN30 untagged configuration (with minor issues).
- switching works fine on this VLAN 30
- as soon as you configure an unused port to another untagged VLAN (in your case 1) the existing VLAN 30 no longer works

plaurien, 2024-10-15:
That would be correct except that switching Vlan30 seems problematic today; while I'm running the exact same config than yesterday.

Hooking a Linux laptop with tcpdump, it looks like despite all ports being configured as Untagged, the Zyzel is sending 802.1Q with vlan30.

If I now set the port as trunk -either on Port1 connecting to my Cisco switch or Port8 connecting to my laptop-, the laptop is seeing 802.1Q-inQ:
ethertype 802.1Q-QinQ (0x88a8), length 222: vlan 30, p 0, ethertype 802.1Q (0x8100), vlan 30, p 0, ethertype IPv4 (0x0800)

Hope this helps you identifying the issue.
If needed, I can restart everything from scratch again as I can't narrow down while it had seemed to work Yesterday while all was in untagged Vlan30. But I definitely permanently see 802.1Q or Q-in-Q while interface is set to untagged.

thank you in advance for your help.

Neustradamus, 2024-10-27:
@plaurien: Interesting bug

Neustradamus, 2024-11-26:
@oliv3r, @svanheule, @TobleMiner, @achterin, @plaurien, @chapichapo, @Biergartenparadoxon, @plappermaul, @SimonFischer04, @0xharshal, @mr-wewlad, @Borromini, @jonasjelonek, @lucize, @tristan-k, @pierrepinon: Can you look?