openwrt/openwrt #16684: kmodloader: jitterentropy_rng

peterwillcn, 2024-10-12:
### Describe the bug

[    9.523489] jitterentropy: Initialization failed with host not compliant with requirements: 9
[    9.661916] kmodloader: 1 module could not be probed
[    9.662377] kmodloader: - jitterentropy_rng - 0

### OpenWrt version

r27736-eb05baff7f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPC-T6

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-10-19:
Check kernel 6.6.57 (https://github.com/openwrt/openwrt/commit/51bbc8114be18d3c00ae047d297609c1dfd63855), the problem should be solved.

ynezz, 2024-10-20:
> the problem should be solved.

@namiltd do you by any chance know which kernel commit specifically fixes this?

namiltd, 2024-10-20:
I don't have that knowledge.

namiltd, 2024-11-10:
> Check kernel 6.6.57 ([51bbc81](https://github.com/openwrt/openwrt/commit/51bbc8114be18d3c00ae047d297609c1dfd63855)), the problem should be solved.

However, I see that it still occurs for kernel 6.6.59.

nieroster, 2025-01-30:
seems still to be still there on kernel 6.6.73

Djfe, 2025-02-02:
This is still happening on rc7 on several ramips-mt7621 boards.
(I checked 4 different devices)

teckglobal, 2025-02-06:
24.10.0 sable release I still see it happening

BlueMax, 2025-02-06:
Here it's gone since stable (or last RCs).
TP-Link TL-WR1043ND v1
AR9132, ath79/generic, mips

namiltd, 2025-02-06:
24.10.0 - ramips/mt7620/TP-Link Archer C5 v4
`[   15.598992] jitterentropy: Initialization failed with host not compliant with requirements: 9`

teckglobal, 2025-02-15:
> 24.10.0 sable release I still see it happening

You can see that it's not running by doing:
cat /proc/crypto | grep jitter

After checking I found is I run "modprobe -q jitterentropy_rng" <-- I needed to do this 2 times and it started

Now when I check it again:
cat /proc/crypto | grep jitter
name          : jitterentropy_rng
driver          : jitterentropy_rng
module       : jitterentropy_rng

BlueMax, 2025-03-03:
Anyone noticing any effects of this? I think my WiFi login times (4-way handshakes) get horrible after some time and only get better when logging into ssh and giving the machine something to do (resulting in filling the entropy pool I guess).

ridzzmd, 2025-03-03:
Not sure if it is related or not. The router (Linksys EA8100) having issue when trying to perform upgrade from 23.05 to 24.10. Which cause me to reset to original firmware and do upgrade from there. Even using that approach, the upgrade failed most of the times. The symptom that I see is that the nic for the lan port not properly functioning or maybe it hung or something. Only the left led is light up. When I power cycle it (by switching off/on the power switch multiple times) both led will work and then only I can connect back to the router. If not, it seems to be stuck in the initialization stage. The reason I suspect it is stuck is because it throwing a bogon network (169..254.xx.xxx) when I try to connect to the router when only the left led is light up on the nic port.

namiltd, 2025-03-03:
@ridzzmd off topic. Move your information to a separate issue.

reinerotto, 2025-03-13:
Bug confirmed on Linux version 6.6.73, openwrt 24.10.0, SoC Type: MediaTek MT7620A ver:2 eco:6, WE826

schuettecarsten, 2025-03-14:
I can confirm that this happens whenever I connect a MT76x0u based device in an ATH79 router - Alfa Tube UAC2 and R36AH.

```
[  394.330022] jitterentropy: Initialization failed with host not compliant with requirements: 9
[  394.437004] kmodloader: 1 module could not be probed
[  394.442633] kmodloader: - jitterentropy_rng - 0
[  531.095104] usb 1-1.1: USB disconnect, device number 3
[  534.565609] usb 1-1.1: new high-speed USB device number 4 using ehci-platform
[  535.045623] usb 1-1.1: reset high-speed USB device number 4 using ehci-platform
[  535.303963] mt76x0u 1-1.1:1.0: ASIC revision: 76100002 MAC revision: 76502000
[  536.350593] mt76x0u 1-1.1:1.0: EEPROM ver:02 fae:04
[  536.406106] ieee80211 phy2: Selected rate control algorithm 'minstrel_ht'
```

getreu, 2025-03-14:
I confirm this bug on _Zbtlink ZBT-WE826 (16M)_ on "OpenWrt"
"24.10.0". Here my `dmesg`

```
[    0.000000] Linux version 6.6.73 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28427-6df0e3d02a) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Mon Feb  3 23:09:37 2025
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
[    0.000000] MIPS: machine is Zbtlink ZBT-WE826 (16M)
....
[   20.905798] usbserial: USB Serial support registered for GSM modem (1-port)
[   21.008400] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 6352, rev 0500 detected
[   21.017489] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 7620 detected
[   21.026407] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   21.494707] jitterentropy: Initialization failed with host not compliant with requirements: 9
[   22.576305] kmodloader: 1 module could not be probed
[   22.581712] kmodloader: - jitterentropy_rng - 0
```

smanet80, 2025-04-01:
I am also seeing that in my old Nexx WT3020 device running 24.10 with a wifi6 usb adapter:

`[ 21.101467] usb 1-1: reset high-speed USB device number 2 using ehci-platform
[   23.015624] usbcore: registered new interface driver mt7921u
[   23.033570] mt7921u 1-1:1.0: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a
[   23.033570]
[   23.138762] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 6352, rev 0500 detected[   23.155388] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 7620 detected
[   23.170905] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   23.217949] mt7921u 1-1:1.0: WM Firmware Version: ____010000, Build Time: 20231109190959
[   23.693439] jitterentropy: Initialization failed with host not compliant with requirements: 9
[   24.819989] kmodloader: 1 module could not be probed
[   24.863230] kmodloader: - jitterentropy_rng - 0`

I am reverting back to 23.05.5 to see if that works there.

closed 2025-04-09 by openwrt-bot

Djfe, 2025-04-09:
I will test this at some point on my device and report back. I didn't find the time, yet.

nieroster, 2025-04-23:
```
[   24.242252] jitterentropy: Initialization failed with host not compliant with requirements: 9
...
[   27.536354] jitterentropy: Initialization failed with host not compliant with requirements: 9
[   27.775814] urngd: v1.0.2 started.
[   28.176832] kmodloader: 1 module could not be probed
[   28.182125] kmodloader: - jitterentropy_rng - 0
```
seen today on Netgear WNDR3800, OpenWrt 24.10.1 r28597-0425664679 / LuCI openwrt-24.10 branch 25.111.77565~1b4aa1d

opty77, 2025-05-16:
I just noticed today on

```
root@strong:~# ubus call system board
{
        "kernel": "6.6.86",
        "hostname": "strong",
        "system": "MediaTek MT7621 ver:1 eco:3",
        "model": "MTC Wireless Router WR1201",
        "board_name": "mtc,wr1201",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.1",
                "revision": "r28597-0425664679",
                "target": "ramips/mt7621",
                "description": "OpenWrt 24.10.1 r28597-0425664679",
                "builddate": "1744562312"
        }
}
```

(actually Strong 1200):

```
2025-05-06T18:22:21.345194+02:00 strong kernel - - - mtk-eip93 1e004000.crypto: EIP93 Crypto Engine Initialized.
2025-05-06T18:22:21.732450+02:00 strong kernel - - - jitterentropy: Initialization failed with host not compliant with requirements: 9
[...]
2025-05-06T18:22:22.319334+02:00 strong urngd - - - v1.0.2 started.
[...]
2025-05-06T18:22:24.346655+02:00 strong kernel - - - jitterentropy: Initialization failed with host not compliant with requirements: 9
2025-05-06T18:22:24.486585+02:00 strong kmodloader - - - 1 module could not be probed
2025-05-06T18:22:24.496812+02:00 strong kmodloader - - - - jitterentropy_rng - 0
```

opty77, 2025-05-16:
I see, eec11fb hasn't made it to 24.10 yet, sorry for the noise.

BTW,

> If you run "modprobe -q jitterentropy_rng" <-- I needed to do this 2 times and it started

doesn't work here.