openwrt/openwrt #16687: OOM after booting Linksys RE6500.

MalikuMane, 2024-10-12:
### Describe the bug

After flashing the official image, the device is inaccessible. 

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys RE6500

### Image kind

Official downloaded image

### Steps to reproduce

1. Flash the official image.
2.  Wait.

### Actual behaviour

The Luci login screen does not appear, the LED flashes continuously.

### Expected behaviour

I would expect to wait a few minutes and then see the luci login screen when i visit http://192.168.1.1

### Additional info

[re6500bad.txt](https://github.com/user-attachments/files/17351293/re6500bad.txt)
[re6500bad2.txt](https://github.com/user-attachments/files/17351294/re6500bad2.txt)
Added to logs with monitoring the serial console during boot. There seems to be an OOM condition but not sure how to troubleshoot it further. 

22.03.7 worked fine.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stevenengland, 2024-10-12:
Seems like the same behaviour I recognized (https://github.com/openwrt/openwrt/issues/16688). But I have no means to check the serial part. Did you manage to sideload another image via TFTP? I wasn't.

MalikuMane, 2024-10-12:
TFTP does work but it is quite annoying and I had to monitor it with Wireshark to see why it kept failing. Failsafe is eaiser, Press the WPS once right when the light starts blinking quickly to trigger it, for me it's about 27 seconds after powerup, `scp` the older image over and `sysupgrade`  it.  

> Seems like the same behaviour I recognized (#16688). But I have no means to check the serial part. Did you manage to sideload another image via TFTP? I wasn't.

stevenengland, 2024-10-13:
> TFTP does work but it is quite annoying and I had to monitor it with Wireshark to see why it kept failing. Failsafe is eaiser, Press the WPS once right when the light starts blinking quickly to trigger it, for me it's about 27 seconds after powerup, then you can ssh over, `scp` the older image over and `sysupgrade`  it.  
> 
> > Seems like the same behaviour I recognized (#16688). But I have no means to check the serial part. Did you manage to sideload another image via TFTP? I wasn't.
> 
> 

Thanks mate, I will try that out. Where did you get the information about failsafe? It is not documented. And just for completeness: What did you find out regarding why TFTP is not working?

Thanks again.

MalikuMane, 2024-10-13:
> Thanks mate, I will try that out. Where did you get the information about failsafe? It is not documented. And just for completeness: What did you find out regarding why TFTP is not working?
>
> Thanks [...]

Just a little trial and error while watching the console to figure out what it was doing to get into failsafe. It was a while ago but i think i just forgot to open my firewall, a reminder on the wiki might have been helpful. The rest of the instructions are accurate.

emilyastranova, 2026-09-03:
Just wanted to report that the wiki currently recommends flashing 25.12.2. I just tried doing so and ran into this same issue, doesn't even boot to initramfs. For anyone looking in the future, just use 22.03.7. The kernel in 5.15 is just too big. Need to get the wiki updated as well.

**22.03.7 Links**

- https://downloads.openwrt.org/releases/22.03.7/targets/ramips/mt7621/openwrt-22.03.7-ramips-mt7621-linksys_re6500-initramfs-kernel.bin
- https://downloads.openwrt.org/releases/22.03.7/targets/ramips/mt7621/openwrt-22.03.7-ramips-mt7621-linksys_re6500-squashfs-sysupgrade.bin