openwrt/openwrt #16697: uboot(2024.10)-mediatek failed to build (build variant: mt7986_netcore_n60)

libiunc, 2024-10-13:
### Describe the bug

u-boot-mt7986_netcore_n60 build error

u-boot-2024.10

### OpenWrt version

snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

mediatek mt7986 

### Device

Netcore N60

### Image kind

Self-built image

### Steps to reproduce

gcc14,clang(to enable bridger) 

### Additional info

aarch64-openwrt-linux-musl-gcc -Wp,-MD,cmd/.cache.o.d -nostdinc -isystem /media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/14.2.0/include -Iinclude   -I./arch/arm/include -include ./include/linux/kconfig.h -I./dts/upstream/include -D__KERNEL__ -D__UBOOT__ -Wall -Wstrict-prototypes -Wno-format-security -fno-builtin -ffreestanding -std=gnu11 -fshort-wchar -fno-strict-aliasing -fno-PIE -Os -fno-stack-protector -fno-delete-null-pointer-checks -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-maybe-uninitialized -fmacro-prefix-map=./= -gdwarf-4 -fstack-usage -Wno-format-nonliteral -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=date-time -Wno-packed-not-aligned -D__ARM__ -fno-pic -mstrict-align -ffunction-sections -fdata-sections -fno-common -ffixed-x18 -mgeneral-regs-only -mbranch-protection=none -pipe -march=armv8-a+crc -D__LINUX_ARM_ARCH__=8 -I./arch/arm/mach-mediatek/include    -DKBUILD_BASENAME='"cache"'  -DKBUILD_MODNAME='"cache"' -c -o cmd/cache.o cmd/cache.c
cmd/cache.c: In function 'do_dcache':
cmd/cache.c:57:25: error: implicit declaration of function 'noncached_set_region' [-Wimplicit-function-declaration]
   57 |                         noncached_set_region();
      |                         ^~~~~~~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:257: cmd/cache.o] Error 1
make[4]: *** [Makefile:1895: cmd] Error 2
make[4]: Leaving directory '/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/u-boot-mt7986_netcore_n60/u-boot-2024.10'
make[3]: *** [Makefile:888: /media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/u-boot-mt7986_netcore_n60/u-boot-2024.10/.built] Error 2
make[3]: Leaving directory '/media/lib/LuckyE/opn/openwrt/package/boot/uboot-mediatek'
time: package/boot/uboot-mediatek/mt7986_netcore_n60/compile#0.40#0.33#0.71
    ERROR: package/boot/uboot-mediatek failed to build (build variant: mt7986_netcore_n60).
make[2]: *** [package/Makefile:177: package/boot/uboot-mediatek/compile] Error 1
make[2]: Leaving directory '/media/lib/LuckyE/opn/openwrt'
make[1]: *** [package/Makefile:171: /media/lib/LuckyE/opn/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_netcore_n60=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
# CONFIG_BINUTILS_USE_VERSION_2_42 is not set
CONFIG_BINUTILS_USE_VERSION_2_43=y
CONFIG_BINUTILS_VERSION="2.43.1"
CONFIG_BINUTILS_VERSION_2_43=y
CONFIG_BPF_TOOLCHAIN_HOST=y
CONFIG_BPF_TOOLCHAIN_HOST_PATH=""
# CONFIG_BPF_TOOLCHAIN_NONE is not set
# CONFIG_GCC_USE_VERSION_13 is not set
CONFIG_GCC_USE_VERSION_14=y
CONFIG_GCC_VERSION="14.2.0"
CONFIG_GCC_VERSION_14=y
CONFIG_HAS_BPF_TOOLCHAIN=y
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
# CONFIG_LUCI_CSSTIDY is not set
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_kmod-hwmon-pwmfan=y
CONFIG_PACKAGE_kmod-mt7981-firmware=y
# CONFIG_PACKAGE_kmod-mt7986-firmware is not set
CONFIG_PACKAGE_kmod-nls-base=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-usb-xhci-hcd=y
CONFIG_PACKAGE_kmod-usb-xhci-mtk=y
CONFIG_PACKAGE_kmod-usb3=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_mt7981-wo-firmware=y
# CONFIG_PACKAGE_mt7986-wo-firmware is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_LLVM_HOST=y
CONFIG_KERNEL_WERROR=
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

libiunc, 2024-10-13:
Switch to gcc13, it is normal

ggg70, 2024-10-14:
I noticed the same error, too, when building for Banana Pi BPI-R3 Mini with gcc 14.

lister-wrt, 2024-10-14:
I have the same issue building BPI-R3/4 and OpenWrt One

jonasjelonek, 2024-10-15:
also just encountered this issue. Fixed this with a different solution, adding an ifdef-ed prototype in a header and including that header in `cmd/cache.c` since that looks more reasonable to me than adding the extern declaration.

```patch
--- a/arch/arm/include/asm/system.h
+++ b/arch/arm/include/asm/system.h
@@ -658,6 +658,7 @@ void mmu_set_region_dcache_behaviour(phy
  * Return: 0 if OK
  */
 int noncached_init(void);
+void noncached_set_region(void);

 phys_addr_t noncached_alloc(size_t size, size_t align);
 #endif /* CONFIG_SYS_NONCACHED_MEMORY */
--- a/cmd/cache.c
+++ b/cmd/cache.c
@@ -10,6 +10,7 @@
 #include <command.h>
 #include <cpu_func.h>
 #include <linux/compiler.h>
+#include <asm/system.h>

 static int parse_argv(const char *);

```

Any comments on this?

rmandrad, 2024-10-15:
thx @jonasjelonek your patch works great :)

rmandrad, 2024-10-15:
@jonasjelonek are you able to push the patch to openwrt ? ping @dangowrt

N-Storm, 2024-10-15:
> Please also send the patch to [u-boot@lists.denx.de](mailto:u-boot@lists.denx.de) as this is a bug in upstream U-Boot when being built with GCC-14.

They also have a GitLab where issues page are enabled. Probably will be more convenient to submit it there. I can do this (based on @jonasjelonek patch, since I've already registered and got confirmed on their GitLab (they require additional manual email for registration validation).

jonasjelonek, 2024-10-15:
@N-Storm thanks anyway for the offer, I'll post it to the mailing list then

N-Storm, 2024-10-15:
@jonasjelonek I've posted it on the issues page as I've already prepared the text & reproduced the error with the 'vanilla' u-boot.: https://source.denx.de/u-boot/u-boot/-/issues/42
But still sending a report to a mailing list sounds reasonable for me. Feel free to copy/modify/quote my report in any ways if you wish so.

CharlesMengCA, 2024-10-15:
How to build and upgrade u-boot? 
My router is Linksys E8450, I built from the latest snapshot but when I tried `strings /dev/mtd1|grep -i u-boot` and still get `U-Boot 2024.07-OpenWrt-r27563-7850e19767 (Sep 23 2024 - 08:26:46 +0000)`

trini, 2024-10-15:
> @jonasjelonek I've posted it on the issues page as I've already prepared the text & reproduced the error with the 'vanilla' u-boot.: https://source.denx.de/u-boot/u-boot/-/issues/42 But still [...]

Thanks all. The patch sent to the mailing list (either via `git send-email` or `git format-patch` and inlining) are preferred.

commit 92ca322d (2024-10-16): uboot-mediatek: add noncached_set_region prototype to fix build

Building uboot-mediatek fails with GCC-14, uboot v2024.10 and
CONFIG_SYS_NONCACHED_MEMORY defined with error:
cmd/cache.c: In function 'do_dcache':
cmd/cache.c:57:25: error: implicit declaration of function
	'noncached_set_region' [-Wimplicit-function-declaration]

This is caused by upstream commit 7d6cee2cd0e2e2507aca1e3a6fe0e2cb241a116e
("cmd: cache: Remove weak functions") as this removes weak functions in
favor of arch-specific definitions.

This patch adds the function prototype for `noncached_set_region` to
arch-specific he [... cut]
closed 2024-10-16 by openwrt-bot